raspberrypi/firmware #955: dt-blob.bin not working for RaspberryPi 3+ ?

jens-maus, 2018-03-16:
I am maintaining an own IoT distribution for the RaspberryPi line of SBCs. To properly setup all GPIOs for my application I require to ensure that some GPIOs are actually input GPIOs with pull_up enabled per default (within the boot loader already). Previously and with a Pi2 or Pi3 this is working fine by providing an own patched version of `dt-blob.bin` in the `/boot` filesystem where I set the desired GPIOs to input+pull_up GPIOs.

I am doing this by patching the existing `dt-blob.dts` file in the raspberrypi/firmware repository and then running `dtc` to create the corresponding `dt-blob.bin` file.

See here fore the patch:
https://github.com/jens-maus/RaspberryMatic/blob/recovery-mode/buildroot-external/patches/rpi-firmware/0001-dt-blob.patch

Now with the release of the Pi3+ I recognised that the corresponding `pins_3b3 { ... }` section was recently added. Thus, I revised my patch to add the following line to `dt-blob.dts`:

```
pins_3b3 { // XXX pins_3bplus Pi 3 Model B+ rev 1.0
  pin_config {
    ...
    pin@p12 { function = "input";  termination = "pull_up"; }; // GPIO12 button
    ...
  }; // pin_defines
}; // pins
```

However, this doesn't seem to work for the Pi3+ while the same line within the `pins_3b2 { ... }` perfectly works with a plain Pi3.

Any explanation why this `dt-blob.bin` file doesn't seem to work with a Pi3+?

pelwell, 2018-03-16:
The uploaded dt-blob.dts is actually out-of-date already - the comment shows the correct name for the 3b+ pins section - `pins_3bplus`. Modify your .dts file accordingly, recompile it, and verify that it now works.

jens-maus, 2018-03-17:
Thanks for your comment and indeed replacing `pins_3b3` with `pins_3bplus` fixes my issues with using a modified `dt-blob.bin`.

However, I hope it is ok if I might use the opportunity to ask here if it wouldn't be a good idea to directly provide a possibility to define such simple "input" and "pullup" changes of the GPIO within the `config.txt` file with something like

```
gpio_setup=12,input,pull_up
```

which might set GPIO12 to an input pin with enabled pull_up. This would then make it easier to define the default state of GPIOs instead of having to provide a patched `dt-blob.bin` like I did in my project. Do you think this is something reasonable that might have a chance to be integrated in a future firmware version?

pelwell, 2018-03-17:
Having a simpler, more granular way of setting gpio state at boot time is something which has been on the wish list for a long time but never tackled. We'll give it some thought.

pelwell, 2018-03-21:
The next firmware build will include a new `gpio` config command. The commit message say:
```
    The 'gpio' command allows GPIOs to be configured and set to arbitrary
    functions and values at boot time. It is an alternative to a custom
    dt-blob for situations where the user isn't trying to add or change
    a named pin.

    The syntax is based on that of raspi-gpio. Each line applies the same
    settings (or at least makes the same changes) to a set of pins, either
    a single pin, a range (3-4), or a comma-separated list of either.
    The pins are followed by an '=', then one or more comma-separated
    attributes from the list:

      ip - Input
      op - Output
      a0-a5 - Alt0-Alt5
      dh - Driving high (for outputs)
      dl - Driving low (for outputs)
      pu - Pull up
      pd - Pull down
      np - No pull

    Attributes are applied in order, as are gpio commands.

    Examples:

      # Select Alt2 for GPIOs 0 to 27
      gpio=0-27=a2

      # Set GPIO12 to be an output set to 1
      gpio=12=op,dh

      # Change the pull on (input) pins 18 and 20
      gpio=18,20=pu

      # Make 17 to 21 inputs
      gpio=17-21=ip
```

jens-maus, 2018-03-21:
Wow, thank your very much, this is really superb. Can't wait to see the new firmware appear here on GitHub (ETA?). This really will make my life easier...

pelwell, 2018-03-21:
Try now.

jens-maus, 2018-03-22:
I just updated to the latest firmware and can verify that the new `gpio` command works as expected. Great, now I can get rid of patching dt-blob.bin. Thank you!

closed 2018-03-22 by jens-maus