raspberrypi/firmware #957: Raspberry Pi 3 B+: lan78xx does not work with upstream kernel?

stapelberg, 2018-03-19:
I tested the Debian image from https://wiki.debian.org/RaspberryPi3 (with updated `/boot/*.{elf,bin,dat}`) and the kernel image from https://github.com/gokrazy/kernel (4.15.11).

In both images, the ethernet interface is detected, can be set up, detects a link, but then never enters the RUNNING state.

I verified the device itself is okay because the ethernet interface does work when using the Raspbian image from https://www.raspberrypi.org/downloads/raspbian/.

I wonder if the lan78xx driver is not completely upstreamed yet? Or is an additional firmware file or kernel config option required?

Thanks for any hints,

pelwell, 2018-03-19:
If you look for my lan78xx commits in the rpi-4.14.y branch you will see the changes that were required. They include enabling auto-negotiation by default and settng the MAC address from properties in the Device Tree.

stapelberg, 2018-03-20:
Thanks for the pointer! For the benefit of other readers, here’s a direct link: https://github.com/raspberrypi/linux/commits/rpi-4.14.y/drivers/net/usb

Are these commits already being upstreamed into the Linux kernel?

pelwell, 2018-03-20:
Not yet, so far as I know. Some of them involve Device Tree changes, and the commits won't be accepted without changes to the bindings documentation (at least).

stapelberg, 2018-03-21:
I cherry-picked the following commits into a linux 4.15.4 kernel build:

* https://github.com/raspberrypi/linux/commit/2056147441c525a2fbe68ed52668b3be51d7c988.patch
* https://github.com/raspberrypi/linux/commit/6865c1e7b3926a9377777dcf7d71ca678913d884.patch
* https://github.com/raspberrypi/linux/commit/7c34b5c7793c40265af49880caea0a4a0caf202b.patch
* https://github.com/raspberrypi/linux/commit/8f3579bf03ec281d03ccfd71535f843bc291c200.patch
* https://github.com/raspberrypi/linux/commit/33c743fb8effa075fe777d09c18dda4dc1dac930.patch
* https://github.com/raspberrypi/linux/commit/c6b70aaae00d7fbf3931c17b5ed1c54cb52011fd.patch

I’m using the firmware files from https://github.com/raspberrypi/firmware/commit/25cf637ccc90d7d2fa37277c807ab33b655bd0f4

Unfortunately, the network interface still doesn’t enter the RUNNING state. Notably, I do see incoming packages (IPv6 router advertisments take effect), but cannot send outgoing packages.

Are there any other changes I need? Perhaps to the device tree? Elsewhere, you wrote that the firmware should be able to automatically use bcm2837-rpi-3-b.dtb, but perhaps that doesn’t cover networking?

Thanks!

pelwell, 2018-03-21:
Compare the DTs from live systems (DTC -I fs /proc/device-tree) to look for omissions or changes.

stapelberg, 2018-03-21:
Turns out the RUNNING attribute was a red herring: only `ifconfig(8)` shows RUNNING, `ip(8)` never does.

In the meantime, I got working boots with the 4.14 and 4.15 kernel. Not sure what’s different, or if the problem is intermittent. Will do some more experimentation later.

For the record, I dumped the device trees and attached them:

* [dt-rpi3-debian.txt](https://github.com/raspberrypi/firmware/files/1832512/dt-rpi3-debian.txt)
* [dt-rpi3b-debian.txt](https://github.com/raspberrypi/firmware/files/1832513/dt-rpi3b-debian.txt)
* [dt-rpi3b-raspbian.txt](https://github.com/raspberrypi/firmware/files/1832514/dt-rpi3b-raspbian.txt)

pelwell, 2018-03-21:
Your Raspbian image will be running a 4.9 kernel - the switch to 4.14 will be in a week or two. If you rpi-update it you'll get a 4.14 kernel, which will make the DT comparison more useful.

lategoodbye, 2018-03-23:
@stapelberg 

The Raspberry Pi 3+ has been released after the pull request for 4.17 has been submitted. So we will need to wait for 4.18 ...

In case you need a working kernel 4.16, you could try my branch which mostly base on Phil's work:
https://github.com/lategoodbye/rpi-zero/tree/bcm2837-rpi3-plus-4.16

JamesH65, 2019-01-09:
@lategoodbye Can this be closed, I presume all the required stuff is now upstreamed?

lategoodbye, 2019-01-09:
The handling for DT property microchip,eee-enabled hasn't been upstreamed yet. But i don't consider this as required.

JamesH65, 2019-01-10:
So you think this can be closed? Or leave open as a flag for the DT property stuff?

lategoodbye, 2019-01-10:
@pelwell @6by9 Do you have any plans to submit this eee-enabled stuff?

pelwell, 2019-01-10:
Rob Herring said (of the `eee-enabled` property):
```
I see we have some flags for broken EEE, but nothing already defined to 
enable EEE. Seems like this should either be a user option (therefore 
not in DT) or we should use the broken EEE properties if this is h/w 
dependent.
```
I think this is a user option to say that EEE won't work on the LAN7515 in their environment, for reasons unknown, so it doesn't belong in upstream DT - that's why I removed it from my upstreamed patch set.

lategoodbye, 2019-01-10:
Okay, thanks for clarify this. Is it possible to enable EEE via ethtool?

pelwell, 2019-01-10:
Yes:
```
$ sudo ethtool --set-eee eth0 eee on
```
But the reason for adding the `eee-enable` property was actually to disable EEE (it is enabled by default if supported by the switch), which requires:
```
$ sudo ethtool --set-eee eth0 eee off
```

lategoodbye, 2019-01-10:
Okay. I think we can close this.

closed 2019-01-10 by pelwell