raspberrypi/firmware #958: Can't netboot on Raspberry Pi Compute Module 3 Lite with the lastest firmware

jpduell, 2018-03-23:
I have a many CM3L boards network booting happily with an old bootcode.bin (unsure of the version, but quite old).

I tried the newest available for both the Master and Stable branches and the CM3L boards seem to hang after grabbing bootcode.bin from the TFTP server. Rolling back to a [slightly older](https://github.com/raspberrypi/firmware/commit/80e1fbeb78f9df06701d28c0ed3a3060a3f557ef) bootcode.bin allows them to boot successfully.

I must mention that I'm using LAN9512s instead of the usual LAN9514s found on the Raspberry Pi 3 B boards.

popcornmix, 2018-03-23:
Can you identify the exact bootcode.bin update which caused this?

jpduell, 2018-03-23:
Sure thing.

Working: [80e1fbeb78](https://github.com/raspberrypi/firmware/tree/80e1fbeb78f9df06701d28c0ed3a3060a3f557ef)
Not Working: [58395c1ea3](https://github.com/raspberrypi/firmware/tree/58395c1ea3f9ae129885aa7ff7d8b92c7be267db)

I tried one known-working CM3L five times with bootcode.bin files from each of those commits.  The 80e1fbeb78 bootcode.bin always netbooted successfully.  The 58395c1ea3 bootcode.bin would never request past bootcode.bin during boot.

popcornmix, 2018-03-23:
@ghollingworth looks like "Insert delay between disconnect and reconnect for USB device booting" is to blame. Any ideas?