raspberrypi/firmware #959: What are the cd, db, and x versions of fixup.dat and start.elf?

kvu787, 2018-03-25:


popcornmix, 2018-03-26:
I think you'll have to be more specific.
You can find the files here: https://github.com/raspberrypi/firmware/tree/master/boot
And they should be present on the first partition of sdcard (although that may be distribution dependent).

popcornmix, 2018-03-26:
start_x is the full firmware that includes camera and software codecs
It is enabled when ``start_x=1`` is in config.txt

start_cd is the cutdown firmware that has no support for firmware hardware blocks like codecs and 3d
It is enabled when ``gpu_mem=16`` is in config.txt

JamesH65, 2018-03-26:
We should probably put this is the documentation somewhere, if it isn't
there already. I'll have a dig around.

popcornmix, 2018-03-26:
There is a mention of start_x and start_db here: https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md
but not much explanation. start_cd is not mentioned.

JamesH65, 2018-03-26:
Thanks @popcornmix, I'll update the docs.

JamesH65, 2018-03-26:
Created an issue https://github.com/raspberrypi/documentation/issues/869, will get to it when I have the time.

ghollingworth, 2018-04-15:
> May I also ask some info about the difference between recovery.elf and
> start.elf? Any difference related to the sdcard i/o maybe? Thanks.

The only difference between recovery.elf and start.elf is the name.  It is
used by the firmware to identify itself as a NOOBS recovery partition

carlonluca, 2018-04-15:
When booting noobs I see that only around 230MB of ram are detected (on Pi3). Also I see other differences. Can I conclude this is not related to recovery.elf? I read in the wiki that recovery.elf "switches the firmware into "NOOBS mode"": does it only mean that it loads files with a different name by default (recovery.cmdline, recovery.img) and that it sets recovery.rfs as the root file system? Thanks.

ghollingworth, 2018-04-15:
Yes that is correct

lurch, 2018-07-01:
I guess https://github.com/raspberrypi/documentation/pull/883 means that this issue can be closed?

JamesH65, 2018-07-02:
Yup!

closed 2018-07-02 by JamesH65