raspberrypi/firmware #961: Please issue a new v1.2018xxxx release (for RPi3B+)

sakaki-, 2018-03-26:
Hi,

I maintain a [64-bit Gentoo image for the RPi3](https://github.com/sakaki-/gentoo-on-rpi3-64bit) and am updating this for the new RPi3 B+.

This needs updated firmware to boot, but without a tagged release tarball, there's no simple way to fetch this (into an ebuild I mean, without having to clone the entire master branch first, which is huge).

Accordingly, could you please release a v1.2018xxxx? Many thanks!

popcornmix, 2018-03-26:
It's actually the head of stable branch you want.
I believe @XECDesign was talking about adding a tag today - it should appear soon.

XECDesign, 2018-03-27:
Tagged
https://github.com/raspberrypi/firmware/releases/tag/1.20180313

sakaki-, 2018-03-27:
Perfect, thank you.

closed 2018-03-27 by sakaki-