raspberrypi/firmware #962: Wrong PARTUUID= in cmdline.txt in 2018-03-13-raspbian-stretch-lite.img

swarren, 2018-03-27:
In 2018-03-13-raspbian-stretch-lite.img, /boot/cmdline.txt contains:
root=PARTUUID=a8fe70f4-02
However, fdisk shows me that the disk ID is b2b56aed not a8fe70f4. A similar problem exists in /etc/fstab. The system will boot once both of those are corrected.

XECDesign, 2018-03-27:
Looks okay from here:
```bash
shift@serge-pc ~ $ fdisk -l 2018-03-13-raspbian-stretch-lite.img
Disk 2018-03-13-raspbian-stretch-lite.img: 1.7 GiB, 1858076672 bytes, 3629056 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xa8fe70f4

Device                                Boot Start     End Sectors  Size Id Type
2018-03-13-raspbian-stretch-lite.img1       8192   93802   85611 41.8M  c W95 FAT32 (LBA)
2018-03-13-raspbian-stretch-lite.img2      98304 3629055 3530752  1.7G 83 Linux
$ sudo mount -o loop,offset=$((8192 * 512)) 2018-03-13-raspbian-stretch-lite.img tmp
$ cat tmp/cmdline.txt 
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=a8fe70f4-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
```

Perhaps something went wrong with the resize in your case?

swarren, 2018-03-27:
Indeed I just downloaded the image at work and it looks OK there, even when written to a USB stick. I'll see if I can repro the issue on a Pi tonight at home.

Does the rootfs resize happen:
a) Automatically on first boot, or
b) Only once I've run raspi-config and requested it?

I was originally attempting to use the serial console (which I guess must be disabled by default?), so booted the system without HDMI/USB attached a few times without it working, and hence did some forced power cycles. If the system was resizing at the time then maybe that caused the issue. Still, nothing else was corrupted...

XECDesign, 2018-03-27:
Automatically on first boot: `init=/usr/lib/raspi-config/init_resize.sh`
https://github.com/RPi-Distro/raspi-config/blob/master/init_resize.sh

That script removes itself as init, adjusts the partition table and replaces the PARTUUID in cmdline.txt and fstab with the new value. Then when the pi reboots, it resizes the actual filesystem and continues normally.

It sounds like the partition the DISKID was altered by the script, but the new value wasn't written to the appropriate files for whatever reason.

swarren, 2018-03-27:
Ah, then that definitely explains the issue; I must have power-cycled the Pi at the wrong moment. I'll close the bug.

closed 2018-03-27 by swarren