raspberrypi/firmware #963: PXE netboot from PiServer fails on reboot with Pi 3 B+

Anonymousdog, 2018-03-27:
RPi 3B+ assigned R-Lite on PiServer won't reboot: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=208692.  This looks like a firmware issue to me.  The RPi stops the reboot process after pulling bootcode.bin (and sometimes requesting start.elf from a location where it won't exist).  When it stops, it gives a four-flash green light (as if it can't load start.elf that it has never downloaded).

This is pretty much a show-stopper for implementing PiServer with the new RPi 3B+.

Cold boot works fine.  Everything is up-to-date in the chroot, firmware, etc.

Tried some slightly older bootcode.bin files from repo with no success...they just broke PXE altogether.

Move to a more appropriate section is this is not a firmware issue.

air-ii, 2018-04-10:
I have the same issue, which led me here, After reboot, PXE fails. But there is also an issue where Pi would stop the process after pulling fixup.dat, even at cold boot. If I should open another issue on this one, let me know.

jmccoy555, 2018-04-23:
I can't even get it to show up in the PiServer Add Client list. 

Testing with bootcode.bin on a SD card, checked working with Pi 3.

rprr, 2018-04-25:
Just wanted to confirm that the 3b+ that I have exhibits the same issue. i.e. hard power cycle works but 'reboot' does not.

the6hu8b, 2018-06-27:
@ghollingworth this problem is biting me really hard. I assume there's no point in offering a bounty to fix it ASAP, but just in case there is, please let me know. I also happen to have ~100 Pi 3 B+'s (...and ~200 earlier versions) available for any kind of testing you may need. We were counting on a full-fledged PXE implementation this time, and our automated tests didn't catch this error b/c we opted for kinda rapidly power cycling it in order to subject it to the harshest treatment we could imagine (this wasn't the strong suit of earlier Pis). What a blunder. 
Nice job with this version BTW, if you could actually fix this issue, the 3 B+ might be the first Raspberry Pi I would recommend to practically anyone without reservations.

pelwell, 2018-06-27:
We've just merged a fix into our internal software repo, and it will be in the next firmware release.

mkreisl, 2018-06-28:
@pelwell Is this already available for download?

ghollingworth, 2018-06-29:
rpi-update and retest...

closed 2018-06-29 by ghollingworth

ghollingworth, 2018-06-30:
How did you update, did you update the server or the client?

You need to go to piserver, click on software and then shell to create a chroot into the client filesystem and then do rpi-update there.  Then check the vcgencmd version on the client

Gordon

mkreisl, 2018-06-30:
@ramprao I can confirm: it's working. Tested many times yesterday

ramprao, 2018-07-01:
It is working now. I think I was expecting the 'reboot' to work after the first reboot immediately after the update. Once it has been power cycled the first time after the rpi-update, subsequent reboot commands worked fine. I tested it on two more Raspberry Pi 3B+ models. It works great. Many, many thanks for this update.

grantemsley, 2020-12-15:
I'm still experiencing this on an up to date PI 3B+.  PXE works 100% of the time from cold boot. But on reboot, it sometimes fails. Not always. Maybe 1 in every 5 reboots.  I see it request dhcp information and dnsmasq sends the boot information. Then it retries again about 5 seconds later. It does that about 7 times, and gets responses, but never tries TFTP.

vcgencmd version says:
```
Feb 12 2020 12:38:08
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start)
```
Which klooks like I'm in an up to date firmware?

I've tried with no SD card, and with a FAT32 formatted sd card with just the latest bootcode.bin on it (a forum post somewhere said that had solved it for some people). Neither way works reliably.

So is this actually fixed? Did I miss a step? Is it a regression in newer firmware?