raspberrypi/firmware #965: i2c stopped working on 4.14.31+

manio, 2018-03-31:
Hi.
I am using Raspberry Pi Model B Rev 2.
After upgrade from official kernel 4.4.11+ to 4.14.31+ my i2c stopped working.
The bcm2708_i2c module is loaded.
The line:
`dtparam=i2c_arm=on`
is still in config.txt
In the old kernel I have in dmesg:
`[   29.245973] bcm2708_i2c 20804000.i2c: BSC1 Controller at 0x20804000 (irq 77) (baudrate 100000)`
In the new kernel - no simillar line.

The problem starts when I want to load the module:
`echo ds2482 0x18 > /sys/bus/i2c/devices/i2c-1/new_device`
In the old kernel - it was working, now in 4.14 there is no such file or directory.

I do not have the overlays subdirectory, but only the "bcm2708-rpi-b.dtb". This was sufficient for the old kernel and i2c worked fine. I hope it is now not a problem too?

pelwell, 2018-03-31:
bcm2708_i2c is an old module - we have now moved over to using bcm2835_i2c. Even if it is loaded successfully you won't see any entries in the kernel log because the Linux kernel devs prefer drivers to be quiet unless there is a problem. However, it will appear in the `lsmod` output.

The Device Tree files provided with the 4.14 kernel should cause the bcm2835_i2c driver to be loaded. If you are see bcm2708_i2c instead then something else must be loading it. Do you have bcm2708_i2c in /etc/modules? If so, please remove it because it will be interfering with the newer driver.

While you are in /etc/modules, make sure that i2c-dev is present.

If you've worked through those instructions (and rebooted) and you don't see `/dev/i2c-1` then post the contents of your config.txt.

manio, 2018-04-01:
Indeed, this was the problem !!! :)
I have the raspbian cutted down to more embedded solution, so I am explicitly loading all modules which I am using (I was not aware of the new one). After change from bcm2708 to bcm2835 the i2c is working back again.
In fact i don't even need the i2c-dev (and I did not need it before) - it is only necessary when doing eg. i2cdetect, but in my above scenario (I'm using the ds2482 as the i2c / 1wire bridge) it's needless.

Thank you very much for you help, @pelwell !
Closing.

closed 2018-04-01 by manio