raspberrypi/firmware #966: MMAL related application freezes

tvm1, 2018-04-06:
I'm running custom compiled mpv media player on RPI 3B. My application continuously supplies media for playback to mpv via its JSON IPC. Mpv is executed with standard `--vo=rpi` and `--hwdec=rpi` options.

The actual issue is that mpv will rather unpredictably freeze and stop responding. This issue occurs only with enabled `--hwdec=rpi` (hardware decoding).

I've discussed this with mpv devs here: mpv-player/mpv#5683

We narrowed the reason of the hang down to this:
```
#1  do_futex_wait (sem=sem@entry=0x769283bc <client+36>, abstime=0x0) at sem_waitcommon.c:115
#2  0x75d6d158 in __new_sem_wait_slow (sem=0x769283bc <client+36>, abstime=0x0) at sem_waitcommon.c:282
#3  0x76912d30 in mmal_vc_sendwait_message () from /opt/vc/lib/libmmal_vc_client.so
#4  0x76915a20 in mmal_vc_port_disable () from /opt/vc/lib/libmmal_vc_client.so
#5  0x7694f930 in mmal_port_disable_internal () from /opt/vc/lib/libmmal_core.so
#6  0x769504dc in mmal_port_disable () from /opt/vc/lib/libmmal_core.so
```

So this is clearly happens during the `unit video` operation that MPV does after finishing playback of individual media. The issue happens 100% of the time, sometimes it happens after few videos, sometimes it might play 150 shorter videos and do this.

Due to overall complexity of the environment where this issue happens i'm not really able to provide 
something that would easily reproduce this issue, however I can upload whatever would be helpful in determining the cause of this.

tvm1, 2018-04-06:
Here you go.

[asserts.txt](https://github.com/raspberrypi/firmware/files/1884210/asserts.txt)
[msg.txt](https://github.com/raspberrypi/firmware/files/1884211/msg.txt)

6by9, 2018-04-07:
Nothing unusual there.
It may be linked to #457, but it's tough to call. If it is the same then potentially the easiest option is to switch to zero copy (requires the buffers to be allocated from mmal_port_pool_create, but otherwise just ```mmal_port_parameter_set_bool(port, MMAL_PARAMETER_ZERO_COPY, MMAL_TRUE)```). That saves the copy from GPU to ARM memory too.

6by9, 2018-04-19:
Thank you, and sorry for not replying sooner. Nothing untoward in there.

I'm building mpv for other reasons, so I'll see if there is something obvious. Having kicked off a mmal_vc_sendwait_message it'll be waiting for the VPU to reply, so it's unclear why that reply isn't forthcoming.

zmartell, 2018-04-20:
@6by9 How would I go about switching to zero copy? Would that solve it/better? I am having the same issue. 

Should I use hwdec=rpi or hwdec=rpi-copy or some other option to make things better?

6by9, 2018-04-20:
I'm not familiar with the code base.
It appears to be using ffmpeg to do the decoding, and then mpv code to do the rendering. That means ffmpeg would need to support zero copy too and I'm not convinced that you can handle the required manipulations across the API.
Opaque buffers (which appears to be ```hwdec=rpi```) should avoid the issue mentioned in #457 anyway as those don't require the bulk transfer.

zmartell, 2018-04-20:
@6by9 oh sorry about that. These issues go back and forth between raspberry-firmware and not. 

The issue creator mentioned that it happens with hwdec=rpi but you mention that it would solve the issue?

xqtm, 2018-04-20:
I'm experiencing the same issue with JSON IPC and have already tested --hwdec=rpi 
However the issue is still present. 
Without rpi hardware decoder it works, however the Raspberry CPU cannot handle video rendering, so we have a lot of frame drop.  with --hwdec=rpi we have random freezes.

zmartell, 2018-04-20:
@xqtm do you mind sharing what you would use instead of "rpi hardware decoder" , or is the frame drops too unbearable? I am using JSON IPC as well. 

Would there another hardware decoder option that may be suitable on the Pi? 

I have a series of like ~40 raspberry Pis that freezing in occurring every undetermined amount of time, but in some cases as often as every 30 minutes. I have tried hwdec=rpi, hwdec=rpi-copy(to my understanding copy makes it copy to system ram) and it doesn't seem to make a difference. 

Im up for trying anything at this point.

6by9, 2018-04-20:
If there were a reliable mechanism to replicate this then it might be worth investigating what's going on. Without that I'm a little stuck. As it's only been reported by users of MPV it sounds like it is something slightly off with the use of MMAL, but that's a potential needle in a haystack.

A potential option that is currently under development is to use the decoder via a V4L2 M2M device. That should integrate nicely with DRM. I've yet to find out how well MPV works with V4L2 and DRM, but that's the reason I was building it.
...
And as a first test it's just happily decoded 720p Big Buck Bunny via V4L2 and rendered via the standard Pi overlay mechanism. It has hit an issue on EOS/teardown, but I know that still needs some attention.

tvm1, 2018-04-21:
I have full blown proprietary closed source application that can replicate the issue, i tried to make simplified example that would trigger the bug, but i couldn't replicate it.

@zmartell @xqtm If you guys have more simple way to trigger this freeze, please provide more details.

Just run mpv with something like this and provide the log, so we are sure that it freezes at same point:
`mpv --idle --input-ipc-server /tmp/mpvsocket --log-file/tmp/mpvlog --msg-level=a=debug`

Gdb backtrace could also be useful mpv has short writeup how to do that [here](https://mpv.io/bug-reports/) in crashes/segmentation faults section.

zmartell, 2018-04-27:
@tvm1 @xqtm , 

I have solved the issue for myself by simply reverting to an older version of mpv and ffmpeg I still had installed on one Pi. I am not sure if its suitable enough for you but I thought id mention it, but i have it deployed to ~40 Pis that play videos and send ipc commands constantly and well and I haven't had a single freeze since. For myslf I am just going to stick with it as it doesn't affect my environment being a bit outdated. I know its not a solution and more of a workaround. 

mpv --version result: 
mpv 0.28.0 (C) 2000-2017 mpv/MPlayer/mplayer2 projects
 built on Fri Dec 29 05:50:04 UTC 2017
ffmpeg library versions:
   libavutil       56.7.100
   libavcodec      58.9.100
   libavformat     58.3.100
   libswscale      5.0.101
   libavfilter     7.8.100
   libswresample   3.0.101
ffmpeg version: N-89651-gd01eeef4a2

Hope it helps you two.

JamesH65, 2019-01-09:
This appears to be an issue with a specific version of MPV, rarther than anything Pi specific, so can it be closed?

tvm1, 2019-01-09:
Yep, closing this.

closed 2019-01-09 by tvm1