raspberrypi/firmware #967: Can't change UART clock via mailbox interface

Phlamethrower, 2018-04-07:
It looks like with the March 7th firmware and newer, it's impossible to change the UART clock via the mailbox SET_CLOCK_RATE message. The new value is saved by the firmware and can be read back via GET_CLOCK_RATE, but the actual clock rate (as measured by vcgencmd measure_clock, or by the new GET_CLOCK_MEASURED, or by checking UART data transfer) seems unchanged. This is breaking UART support in RISC OS since we assume we can set the UART clock back to its original value of 3MHz via SET_CLOCK_RATE (we don't have any bluetooth support yet, so even on 3B/3B+ we still use the PL011 @ 3MHz as our main UART)

Specifying an init_uart_clock value in config.txt works, so it looks like it's just runtime modification via the mailbox that's broken.

I'm aware of #951, #953 and #956 which talk of similar clocking issues, but they seem to be mostly related to reading values rather than writing them. In any case, this issue is still present on the latest firmware, and seems to affect both new (3B/3B+) and old (1B) Pis.

popcornmix, 2018-04-09:
Can you test with latest firmware?

Phlamethrower, 2018-04-09:
The new firmware seems to work fine. Thanks!

closed 2018-04-09 by Phlamethrower