raspberrypi/firmware #968: Reboot from USB stopped working on 3B+ now Pi 3B+ flashes 4 times after reboot

hoopsurfer, 2018-04-09:
Boot works fine always (both from SD and USB), reboot never works from USB, but works fine from SD.

Using fresh Raspian images with apt update/apt full-upgrade/rpi-update applied. 
Current uname -a output:
Linux raspberrypi 4.14.32-v7+ #1107 SMP Fri Apr 6 16:36:46 BST 2018 armv71 GNU/Linux

On reboot system fails to reboot, the green LED blinks:  "short short short long " over and over

Fails both with and without blank SD card installed
Fails on multiple power supplies, multiple USB devices, multiple Pi 3B+s 
Full power-off/on always boots and same Pi 3B+ devices rebooted with original Pi-day bits

ghollingworth, 2018-04-09:
In reception

pelwell, 2018-04-10:
Your error code blink sequence looks odd - the known codes only have 0, 2 or 4 long flashes. Can you double check the pattern? Link to a short video if you aren't sure.

hoopsurfer, 2018-04-11:
Interesting AFTER shutdown I see 10 flashes consistently.  But here are more specifics.

I acquired another Pi 3B+ and it WAS able to reboot we'll need to see if it can maintain that ability.  We'll call this the GoodPi.  I have several Pi-Day Pi 3B+ that exhibit consistent reboot failure using the exact same USB devices of different types - we'll call those BadPi.

Both GoodPi and BadPi show repeated four rapid flashes when booting from USB _without an SD_ where the flashes start at OS boot, but _with an a blank SD_ card both do not flash.  What are the constant 4 rapid flashes when booting from USB when the boot works fine.

Using only a clean 2018-03-13-raspbian-stretch image etched on a USB _with no SD card_, Both GoodPi and BadPi 3B+ reboot successfully the first time to expand storage, then once they fully boot subsequent reboots fail on the BadPI and it flashes the Green LED 4 times short-short-short-long (more regular flashes not the rapid flashes above).  On the GoodPi reboot works as expected.  What are the 4 regular flashes when reboot fails.

On shutdown both GoodPi and BadPi turn off the Red LED after shutdown then back on and then flash the Green LED 10 times.  What are the 10 rapid flashes after shutdown?

burtyb, 2018-04-11:
Not sure if this is exactly the same as I see 4 very fast flashes (https://www.youtube.com/watch?v=dqm8poGcTJw 4x slow motion) but I've seen occasional problems rebooting from USB whilst testing other things. 

I setup the RPi3+ booting off USB with Raspbian lite to auto login and then reboot from bashrc. Sometimes I get a good run of over 1000 reboots before it fails but it can also be every reboot.

Good reboot (serial output)

> [    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.30-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1102 SMP Mon Mar 26 16:45:49 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @ba34a000 s38720 r8192 d22720 u69632
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=4ea08631-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 940260K/970752K available (7168K kernel code, 550K rwdata, 2048K rodata, 1024K init, 706K bss, 22300K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
[    0.000000]       .data : 0x80c00000 - 0x80c8986c   ( 551 kB)
[    0.000000]        .bss : 0x80c90f40 - 0x80d417f4   ( 707 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 24918 entries in 74 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000276] Console: colour dummy device 80x30
[    0.000819] console [tty1] enabled
[    0.000861] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000901] pid_max: default: 32768 minimum: 301
[    0.001230] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001263] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002215] Disabling memory control group subsystem
[    0.002314] CPU: Testing write buffer coherency: ok
[    0.002726] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003116] Setting up static identity map for 0x100000 - 0x10003c
[    0.003248] Hierarchical SRCU implementation.
[    0.003896] smp: Bringing up secondary CPUs ...
[    0.004646] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005446] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006228] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006332] smp: Brought up 1 node, 4 CPUs
[    0.006402] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.006423] CPU: All CPU(s) started in HYP mode.
[    0.006441] CPU: Virtualization extensions available.
[    0.007309] devtmpfs: initialized
[    0.017773] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.018022] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.018065] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.018642] pinctrl core: initialized pinctrl subsystem
[    0.019401] NET: Registered protocol family 16
[    0.022018] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.026798] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.026831] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.027053] Serial: AMBA PL011 UART driver
[    0.028699] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[... 359 lines cut ...]
[    1.952021] sdhost: log_buf @ bad13000 (fad13000)
[    1.991476] Indeed it is in host mode hprt0 = 00021501
[    2.063140] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    2.072750] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.081401] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    2.090229] of_cfs_init
[    2.095030] of_cfs_init: OK
[... cut ...]
[    2.971609] usb 1-1.1: New USB device found, idVendor=0424, idProduct=2514
[    2.980728] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.990830] hub 1-1.1:1.0: USB hub found
[    2.997017] hub 1-1.1:1.0: 3 ports detected
[  203.752220] random: crng init done

So it gets as far as fetching and booting the kernel from USB flash but then the USB flash isn't detected by the kernel. I tried removing and replugging the USB flash (before the "random: crng init done" shows up) but that didn't help.

pelwell, 2018-04-11:
Four fast flashes means "Start not found", as reported by bootcode.bin when the load fails for any reason.

The 10 flashes on shutdown is just a visual indication that shutdown has completed successfully.

Comparing the successful and unsuccessful boots, after removing the timestamps and a certain amount of reordering you get:
```diff
--- good.txt       2018-04-11 14:32:21.130087600 +0100
+++ bad.txt       2018-04-11 14:32:18.506254500 +0100
@@ -200,41 +200,4 @@
 usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
 hub 1-1.1:1.0: USB hub found
 hub 1-1.1:1.0: 3 ports detected
-dwc_otg_handle_wakeup_detected_intr lxstate = 2
-usb 1-1.1.1: new high-speed USB device number 4 using dwc_otg
-usb 1-1.1.1: New USB device found, idVendor=0424, idProduct=7800
-usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
-libphy: lan78xx-mdiobus: probed
-usb 1-1.1.3: new high-speed USB device number 5 using dwc_otg
-usb 1-1.1.3: New USB device found, idVendor=0781, idProduct=5583
-usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
-usb 1-1.1.3: Product: Ultra Fit
-usb 1-1.1.3: Manufacturer: SanDisk
-usb 1-1.1.3: SerialNumber: 4C530001220114110455
-usb-storage 1-1.1.3:1.0: USB Mass Storage device detected
-scsi host0: usb-storage 1-1.1.3:1.0
-scsi 0:0:0:0: Direct-Access SanDisk Ultra Fit 1.00 PQ: 0 ANSI: 6
-sd 0:0:0:0: [sda] 240254976 512-byte logical blocks: (123 GB/115 GiB)
-sd 0:0:0:0: [sda] Write Protect is off
-sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
-sda: sda1 sda2
-sd 0:0:0:0: [sda] Attached SCSI removable disk
-EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: (null)
-VFS: Mounted root (ext4 filesystem) readonly on device 8:2.
-devtmpfs: mounted
-Freeing unused kernel memory: 1024K
-systemd[1]: System time before build time, advancing clock.
-NET: Registered protocol family 10
-Segment Routing with IPv6
-ip_tables: (C) 2000-2006 Netfilter Core Team
-systemd[1]: systemd 232 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN)
-systemd[1]: Detected architecture arm.
-systemd[1]: Set hostname to .
-systemd[1]: Listening on udev Kernel Socket.
-systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
-systemd[1]: Listening on Syslog Socket.
-systemd[1]: Created slice User and Session Slice.
-systemd[1]: Started Forward Password Requests to Wall Directory Watch.
-systemd[1]: Reached target Swap.
-systemd[1]: Listening on /dev/initctl Compatibility Named Pipe.
 random: crng init done
```
In other words, the kernel just isn't seeing any USB devices, including the built-in Ethernet controller - just the hubs.

JamesH65, 2018-04-11:
Probably not related, but thought I would mention it, if I fail to update the modules to match the kernel when testing a new kernel, lack of USB is the major symptom. That would probably be indicated in the logs though.

pelwell, 2018-04-11:
Good point, but I don't think it's relevant in this case because in @burtyb's case the failure is intermittent and for @hoopsurfer it is board- (not model-) specific.

@hoopsurfer Try adding the following to config.txt:
```
sdram_freq=450
arm_freq=1200
```

hoopsurfer, 2018-04-12:
Swapping out a GoodPi with a BadPi (everything attached/inserted is identical) fails to reboot with 
 sdram_freq=450
 arm_freq=1200
it also fails with
 sdram_freq=400
 arm_freq=1000
I diffed logs (dmesg --notime) and they are identical with just a few minor ordering differences.  At this point I'm thinking I send the BadPi back as defective and call it a day.

BTW, the four flashes with no SD card are very rapid compared with the 4 flashes on reboot failures, still not clear what those mean since the happen on both GoodPi and BadPi.

Nilpo, 2018-04-12:
I have 2 Pi 3 B+ that are exhibiting the exact same 4 flash behavior when using any firmware from 2018. When booting 2017 versions, I can boot successfully to the GPU test screen with a lightning bolt indicating outdated firmware. I can successfully boot into older releases of NOOBS and Raspbian, but again only those on 2017 firmware. I have returned and replaced both Pis already and they have all exhibited the exact same behavior. All of them were sourced from the same supplier.

hoopsurfer, 2018-04-15:
After sitting idle for the better part of 3 days, GoodPi failed an attempted reboot, after a power cycle it did boot and then reboot and seems to consistently reboot successfully **again** (tried 5 times). Something is not stable about 3B+ reboot behavior at least when booting from a USB device.  I am using a tested 2.5A power adapter.  Maybe if I let BadPi sit idle for 3 days it will start to reboot again. : )

ghollingworth, 2018-04-15:
@hoopsurfer Just looking into this a little, but starting by adding a whole tonne of changes to get some logging out...

To confirm your problems:

When you say failed to reboot, can you be more specific:  How far does it get?  Can you get a video of the Pi rebooting and failing showing both the LEDs and the HDMI output?  Just want to make sure it is failing at the same point.

hoopsurfer, 2018-04-16:
Fail to reboot consistently on one of my 3B+ BadPi means:

1) Issue "reboot" command in terminal
2) The BadPi goes through shutdown/reboot process via systemd
3) Red LED goes out briefly then turns back on
4) The Green LED flashes 4 times, the 4th being a longer flash.  

I'll capture a video Monday.

hoopsurfer, 2018-04-17:
When preping to create a video, it appears GoodPi has now become a BadPi much to my disappointment.  Reboot fails with or without a blank SD now.  IF we assume inverted LEDs causes the long flash.  What is going to cause failure to find start.elf when it consistently IS found on boot from power-off?

Nilpo, 2018-04-17:
It's worth mentioning that USB booting is still officially considered an experimental feature, isn't it?

Nilpo, 2018-04-17:
Also, bad start.elf is short-short-short-short-long-long-long-long, is it not? Doesn't it do the same number of shorts and longs. It has in every instance for me.

hoopsurfer, 2018-04-18:
After leaving that system off for the past day and restarting it, it now reboots consistently including multiple USB devices.  Now we're back to GoodPi.  I cannot explain the change.

hoopsurfer, 2018-04-19:
@Nilpo we're talking Pi 3B+, which boots from USB by default, no wonky bit setting required.  For me when reboot fails it is short short short long (repeating).  When no SD card is preset boot works but I see blink blink blink blink (repeating flashes). When shutdown it is short short short short short short short short short short (once).  Same on multiple boards however I still have two BadPi (will not reboot) and one GoodPi (mostly reboots).  I did reboot testing for some benchmarks and GoodPi did 30-40 in a row no problems (with and without SD cards and with multiple USB mSATA SSDs).

rprr, 2018-05-02:
I suppose that the reboot issues when netbooting could be related. I too get the repeated sequence of 4 flashes -- 3 short and 1 long. /var/log/sysconfig says that start.elf is not found.

Nilpo, 2018-05-02:
My issues turned out to be related to HDMI.

hoopsurfer, 2018-06-22:
I have now spent some more time looking at the reboot issue using two 3B+ devices. From hours of reboot testing I would say that with a USB mSATA attached I get random results when rebooting - sometimes it works sometimes it fails, once it works it seems to work for a while, once it fails it fails for a while.  I cannot predict when it will work or fail or find any series of steps that causes a consistent result.

In my case both 3B+ devices ALWAYS boot correctly (from power off), but often reboot can fail with the four flashing LEDs.  Reboot failure means the system effectively hangs, power needs to be cycled to reboot.  Forcing fsck doesn't generally make a difference, running with or without a blank SD card seems to make no difference.  

I need help debugging reboot failures.

pelwell, 2018-06-23:
I was looking at this before leaving yesterday, and I think the problem must be that the VPU code to restore the internal LAN_RUN GPIO (LAN_RUN_BOOT) is clearing the (default) pull down setting, causing it to float. Sometimes this works, and sometimes it doesn't.

I'll build and link to a modified start.elf for testing.

pelwell, 2018-06-26:
The change I mentioned above didn't work, so I didn't bother uploading it. We're puzzled, but actively investigating the possible differences between a first-time boot and a reboot.

ghollingworth, 2018-06-29:
Fixed on latest build

closed 2018-06-29 by ghollingworth

hoopsurfer, 2018-06-30:
Reconfirmed the original failure on current public firmware from 2018-06-07 15:30 running Linux 4.14.50-v7+ 1122 SMP Tue Jun 19 12:26:26 BST 2018.

Tested after rpi-update to latest build firmware from 2018-06-29 13:12 running Linux 4.14.52-v7+ 1123 SMP Wed Jun 27 17:35:49 BST 2018 for 4 hours of continuous reboots with 100% success.

I am no longer able to reproduce the reboot failure - consider #968 tested fixed.   Thanks!

polishcode, 2021-07-05:
Hi all,

Although 3 years have passed I am currently experiencing the same issue.
My setup:
- pi 3b+
- all latest updates installed (pi os)
- sd card with (latest) bootcode.bin + timeout empty file
- 1 * usb bootable pendrive with pi os
- 3 * usb 2.0<->sata adapters each connected to 1 TB drive creating raid array (1 logical disk)
- external power supply powering all: pi + disks
- ethernet cable connected to router
- wifi + bt disabled in config files

Cold boot always succeeds.
Restart is unreliable: sometimes succeeds, sometimes fails. When it fails there is no HDMI output, green LED flashes 4 short + 1 long (if I count correctly). Power off/on fixes issue. However, I need that in a remote location, so no easy way to do cold reboot when it fails.

Setup utilized as a low-power backup NAS in secondary location.

Any help or debugging guidance appreciated.

[image]

polishcode, 2021-07-05:
Also, in case reboot fails:
- power LED in pendrive stays off
- data LEDs (blue) in usb<->sata adapters are off
It looks as if during such a reboot USB subsystem failed to somehow initialize.