raspberrypi/firmware #969: setting program_usb_timeout don't work when booting from USB

bukem, 2018-04-09:
Wanted to exclude using SD to switch to only USB device and it worked well by setting OTP bit with `program_usb_boot_mode=1`. But in case I connect more than one USB device, Pi don't start.
Creating an SD with _bootcode.bin_ and _timeout_ solved the problem, but it forces me to use SD. And here i have a few questions:
 - will `program_usb_timeout=1` set OTP bit to wait for 5 secs?
 - I've tried to set the bit for `program_usb_timeout=1` by modifying /boot/config.txt on the USB, not SD, but seems like didn't succeed. Is there any restriction, that forces me to use only SD when setting the bits in OTP?
 - how can I check if the bit for `program_usb_timeout=1` is set? You know, the same way as it done for _program_usb_boot_mode_ by executing `$ vcgencmd otp_dump | grep 17`.

I'm using Pi 3B+.

Thanks, Mike

ghollingworth, 2018-04-10:
Look at the last line of vcgencmd otp_dump, if it has bit 24 set then the
timeout is enabled.

Gordon

joergze, 2018-04-29:
> Look at the last line of vcgencmd otp_dump, if it has bit 24
so its `$ vcgencmd otp_dump | grep 66` and the result should be `66:0**1**000000`
Where have you got this information from? Maybe there is more usefull info?

What to do if `program_usb_timeout=1` did not work?

joergze, 2018-04-30:
Just for all who search for the same thing: 
`program_usb_timeout=1` is wrong.
It must be `program_usb_boot_timeout=1`
After setting, you can verify with `$ vcgencmd otp_dump | grep 66` the result should be a 1 at bit 24 in row 66: `66:01000000`

bukem, 2018-07-07:
Thank you @joergze , that is exactly what i needed!
And to answer to one of my questions: setting program_usb_boot_timeout=1 can be applied both from SD and from USB device.

closed 2018-07-07 by bukem