raspberrypi/firmware #970: New Raspberrypi 3 b+ crashes around every 24 hours

pvanderlinden, 2018-04-09:
I have the raspberrypi running for 3 days now, and it crashed completely twice already. The setup is basicly the same as my old 2 b, except that I updated to the newer raspbian. This seems to be a firmware issue though, the first time I couldn't find any trace. This time I had a lot of these messages in my syslog:

```
Apr  9 10:42:21 hive kernel: [94202.568497] Unable to handle kernel paging request at virtual address a1d7ecc0
Apr  9 10:42:21 hive kernel: [94202.570018] pgd = 81178000
Apr  9 10:42:21 hive kernel: [94202.571455] [a1d7ecc0] *pgd=21c1141e(bad)
Apr  9 10:42:21 hive kernel: [94202.572874] Internal error: Oops: 8000000d [#2193] SMP ARM
Apr  9 10:42:21 hive kernel: [94202.574278] Modules linked in: cmac bnep hci_uart bluetooth ecdh_generic evdev joydev cdc_acm brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdr
v_genirq uio fixed ip_tables x_tables ipv6
Apr  9 10:42:21 hive kernel: [94202.578525] CPU: 2 PID: 2102 Comm: lsb_release Tainted: G      D  C      4.14.30-v7+ #1102
Apr  9 10:42:21 hive kernel: [94202.581250] Hardware name: BCM2835
Apr  9 10:42:21 hive kernel: [94202.582564] task: 81170000 task.stack: 8116a000
Apr  9 10:42:21 hive kernel: [94202.583857] PC is at 0xa1d7ecc0
Apr  9 10:42:21 hive kernel: [94202.585099] LR is at vfs_getattr_nosec+0x6c/0x84
Apr  9 10:42:21 hive kernel: [94202.586313] pc : [<a1d7ecc0>]    lr : [<8028f3dc>]    psr: a0000013
Apr  9 10:42:21 hive kernel: [94202.587524] sp : 8116beb8  ip : 000007ff  fp : 8116bedc
Apr  9 10:42:21 hive kernel: [94202.588706] r10: 00000000  r9 : 8116a000  r8 : af04def0
Apr  9 10:42:21 hive kernel: [94202.589855] r7 : 8116bee8  r6 : 000007ff  r5 : a1d7ecc0  r4 : 8116bf28
Apr  9 10:42:21 hive kernel: [94202.591006] r3 : 00000000  r2 : 000007ff  r1 : 8116bf28  r0 : 8116bee8
Apr  9 10:42:21 hive kernel: [94202.592115] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Apr  9 10:42:21 hive kernel: [94202.593232] Control: 10c5383d  Table: 0117806a  DAC: 00000055
Apr  9 10:42:21 hive kernel: [94202.594343] Process lsb_release (pid: 2102, stack limit = 0x8116a210)
Apr  9 10:42:21 hive kernel: [94202.595450] Stack: (0x8116beb8 to 0x8116c000)
Apr  9 10:42:21 hive kernel: [94202.596531] bea0:                                                       00000800 00000001
Apr  9 10:42:21 hive kernel: [94202.598679] bec0: 00000800 ffffff9c 76a37120 8116bf28 8116bf1c 8116bee0 8028f510 8028f37c
Apr  9 10:42:21 hive kernel: [94202.600818] bee0: 00000000 36b76eb0 b9525210 af026880 80101850 7eebf760 00000000 00378000
Apr  9 10:42:21 hive kernel: [94202.602959] bf00: 000000c3 80108224 8116a000 00000000 8116bfa4 8116bf20 8028fe94 8028f498
Apr  9 10:42:21 hive kernel: [94202.605171] bf20: 000007ff 80c02080 000007ff 00000000 00000000 00000000 00000000 00000000
Apr  9 10:42:21 hive kernel: [94202.607526] bf40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Apr  9 10:42:21 hive kernel: [94202.610015] bf60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Apr  9 10:42:21 hive kernel: [94202.612597] bf80: 00000000 00000000 00000000 00000000 00000000 0000001f 00000000 8116bfa8
Apr  9 10:42:21 hive kernel: [94202.615310] bfa0: 80108080 8028fe68 0000001f 00000000 76a37120 7eebf760 7eebf760 00000000
Apr  9 10:42:21 hive kernel: [94202.618089] bfc0: 0000001f 00000000 00378000 000000c3 00000000 00000001 013acb48 ffffff9c
Apr  9 10:42:21 hive kernel: [94202.620932] bfe0: 003784ac 7eebf708 00191684 76e0a5b8 60000010 76a37120 a37ddb20 4058262e
Apr  9 10:42:21 hive kernel: [94202.623906] [<8028f3dc>] (vfs_getattr_nosec) from [<8028f510>] (vfs_statx+0x84/0xc8)
Apr  9 10:42:21 hive kernel: [94202.626915] [<8028f510>] (vfs_statx) from [<8028fe94>] (SyS_stat64+0x38/0x54)
Apr  9 10:42:21 hive kernel: [94202.628461] [<8028fe94>] (SyS_stat64) from [<80108080>] (ret_fast_syscall+0x0/0x28)
Apr  9 10:42:21 hive kernel: [94202.631436] Code: a1d7ecac 00000000 00000000 00000000 (00000029) 
Apr  9 10:42:21 hive kernel: [94202.632942] ---[ end trace e99b57fc6604ed4e ]---
```

pvanderlinden, 2018-04-10:
It looks like a firmware (kernel/driver) error to me, but is it? Or is this an hardware error (which means I need to send it back soon, only arrived a few days ago), or more likely a software error?

pvanderlinden, 2018-04-10:
It also happens more then once a day, basicly the raspberrypi crashes in less then 24 hours with a lot of these things in the syslog.

JamesH65, 2018-04-10:
Can you try adding the following to config.txt?

sdram_freq=450

If that doesn't work, also add

arm_freq=1200

pvanderlinden, 2018-04-13:
Thanks, It has been stable for 48+ hours (with just the sdram_freq). Does this mean the sdram is now under clocked, or should this have been the default in the first place?

JamesH65, 2018-04-13:
TBH, changing the SDRAM freq does not make much of a difference to overall performance. We have found some outlier SoC's that seem to struggle with the 500 setting, and are currently looking in to why and what can be done about it. They should be fine at 500, but in reality dropping it a bit really won't be particularly noticeable.

JamesH65, 2018-04-13:
ps. Glad it's sorted! Please keep us informed on here if you keep having problems.