raspberrypi/firmware #974: [Question] What are the other OTP Values?

jriwanek, 2018-04-16:
It's understood that 17 is where USB Host is programmed (3 instead of 1 in first column), 28 is the Serial, 30 is revision, and 32 is used atleast in part for the warrenty bit, but I was wondering what the other values are, and about any of the other numbers like say "x020000f" in 17.

ghollingworth, 2018-04-16:
17 is the bootmode register and 18 is a copy:
   Bit 1 sets the oscillator frequency to 19.2MHz
   Bit 3 enables pull ups on the SDIO pins
   Bit 19 enabled GPIO bootmode
   Bit 20 sets the bank to check for GPIO bootmode
   Bit 21 enabled booting from SD card
   Bit 22 sets the bank to boot from
   Bit 28 enables USB device booting
   Bit 29 enables USB host booting (ethernet and mass storage)

28/29 is the serial number and ~serial

30 is revision number

64/65: MAC Address

66 is advanced boot:
   bits 0-6: gpio for ETH CLK output pin
   bit 7: enable ETH_CLK output
   bits 8-14: gpio for LAN_RUN output pin
   bit 15: enable LAN_RUN output
   bit 24: Extend USB HUB timeout parameter
   bit 25: ETH_CLK frequency:
      0 - 25MHz
      1 - 24MHz

ghollingworth, 2018-04-16:
Probably...

:)

Thanks for volunteering James...

JamesH65, 2018-04-16:
Added a docs issue 

https://github.com/raspberrypi/documentation/issues/886

jriwanek, 2018-04-16:
16 appears to be used: "6c280000" On a 1B+, "00280000" on 2B 1.1 and later,
Bits 0 and 2 appear to be used from 17/18 on the 1B+, Any idea what they're for?
27 appears to be used: "00005050" on a 1B+ & 2B 1.1, "00007373" on a 2B 1.2, "00002727" on a 3B & "00001f1f" on a 3B+
31 Might also be used (Seems odd to have gap skipping to me)
64 and 65 seem to be 0 on all my Pi's, Is one the WIFI and one the Ethernet, and if so, which is which?

jriwanek, 2018-04-16:
Also, you say 29 is "~Serial" What exactly is that?

6by9, 2018-04-17:
```~``` is the bitwise complement operator in many programming languages. So ```~serial``` is the serial number with every bit flipped as a simple sanity check.

jriwanek, 2018-04-20:
@ghollingworth Considering you didn't answer the question about the warranty bit, and that's not "manufacturer and silicon specific", nor is the bit to prevent overvolting it would appear your claim there is incorrect, It'd be nice to get a straight answer at least for those, And while I appreciate you're likely bound by NDA, It'd be nice if you'd give straight answers rather than gaslighting, I'd rather not have to mess with the OTP on one of my PI's needlessly just to get the answer to those questions.

To recap what I'd like to know:
1) Which is Warrenty Bit
2) Which is Overvolt Prevention Bit
3) 17/18 Bits 0 and 2 (If not used for 1 & 2)
4) Why 64/65 are zeroed on every pi I've checked despite you saying they're for the MAC address, and Which is used for which MAC (Wifi/Ethernet)

pelwell, 2018-04-20:
> It'd be nice if you'd give straight answers rather than gaslighting

We are under no obligation to divulge any information about the OTP contents - Gordon has been more forthcoming than I would have expected. The answers have been very straight, so please adjust your attitude. Do you even know what the word "gaslighting" means?

jriwanek, 2018-04-20:
Yes, I do, And as I said, the issue is "x is all manufacturer and silicon specific" rather than "I'd rather not answer any further of your questions". Considering that said info *could* be gleaned from twidding settings in config.txt, clearly they're not.

E3V3A, 2018-04-22:
@jriwanek  From what I have found the *warranty bit* is pretty much not used anymore.
One place state that it is shown as the *high but* of the board revision (30) and found in `/proc/cpuinfo`, and it was defined by:  
```
(force_turbo || current_limit_override || temp_limit>85) && over_voltage>0)
```
I'm also curious to the MAC part, but each onboard chip already has a hard-coded MAC. So I guess that if you supply a new MAC in the *cmdline*, then perhaps it will show up there...

ghollingworth, 2018-04-22:
The system will use the MAC address in the OTP in preference to the automatically generated one from the serial number.  If you bought your own OUI you'd be able to program it in there.  Can't remember at the moment whether I created a config.txt parameter to program it...  try:

strings /boot/string | grep program

To see...

Gordon

jriwanek, 2018-04-22:
in current firmware, `strings /boot/start.elf` (I assume you typoed as there was no /boot/string), returns:
program_batch_num
program_board_rev
program_emmc_boot
program_gpio_bootmode
program_mac_address
program_serial_number
program_serial_random
_program_usb_boot_mode
program_usb_boot_timeout

jriwanek, 2018-04-22:
Okay, done some more testing, didn't set program_mac_address as I was using a pi zero and I don't want to really mess with that anyway, more questions:

1) Parameter format for program_mac_address, I assume just the straight mac address as format like: "ffeeddbbaa998877"?
2) As I couldn't test, is 64 ethernet, 65 wifi, the other way round or are they a mirror of each other and used for both?
3) never_over_voltage=1 sets 32 bit 0 correct?
4) overvolting sets the warrenty bit but the warrenty bit isn't stored in otp?
5) is program_emmc_boot actually implemented and if so is it only available on a pi 3 and later like program_usb_boot?
6) Is SPI and NAND boot implemented in current firmware and if not, could it be (and also documented on how to use) please
7) Use of 17/18 bits 0 and 2

ghollingworth, 2018-04-23:
> Parameter format for program_mac_address, I assume just the straight mac address as format like: "ffeeddbbaa998877"?

program_mac_address=ff:ee:dd:bb:aa:99

> As I couldn't test, is 64 ethernet, 65 wifi, the other way round or are they a mirror of each other and used for both?

WiFi is a modified version of the ethernet MAC address, uses the same OUI but rest is derived, same with the bluetooth MAC address

> never_over_voltage=1 sets 32 bit 0 correct?

@popcornmix 

> overvolting sets the warrenty bit but the warrenty bit isn't stored in otp?

@popcornmix 

> is program_emmc_boot actually implemented and if so is it only available on a pi 3 and later like program_usb_boot?

Yes, but it doesn't do anything useful for you...  It's used for CM3 just to help get around a bug in the bootrom

> Is SPI and NAND boot implemented in current firmware and if not, could it be (and also documented on how to use) please

Currently not implemented in bootcode.bin or start.elf.  Unlikely to be supported unless we've got a customer with a 100k CM3 order.

> Use of 17/18 bits 0 and 2

Not used

jriwanek, 2018-04-23:
Okay, and pushing my luck maybe, How about 16 and 27? (Thanks btw for prior answers) I'm now putting together a small Python program to list all this stuff in a pretty was as a front end for vcgencmd.

ghollingworth, 2018-04-23:
Sorry cannot comment on those

jriwanek, 2018-04-23:
Okay, so those are those mfr and silicon specific ones you mentioned, I won't ask further about them then.

Right, last questions barring the ones you've tagged popcorn to answer:
17/18 bits 4-18, 23-27, 30-31
66 bits 16-23, 26-31

and a side question on 17/18-0 and 2: I'm assuming the fact they're set on my 1B+ is just a programming error?

ghollingworth, 2018-04-23:
17/18 everything else is unused / unusable bootcode functionality
66 nothing else in there

Are you sure bits 0 and 2 are set, because I wouldn't expect it to boot if they were

E3V3A, 2018-04-23:
@jriwanek Juts because there are gaps in the bit fields, it doesn't have to mean that they are used. Now in the case for OTP 17 & 66, I guess they are, since this is an advanced field, and there are many other ways to boot, from serial, from JTAG debugger, form HEX files etc. (OTP 18 is a copy as already mentioned.)

jriwanek, 2018-04-23:
I'm very interested! I need to finish my small python program first though, and I'm sure @JamesH65 will be rather relieved at having it taken off his plate, So, as a thank you for your time, I'll endeavor to get that PR'd in the next week or so.
As a side point, I don't suppose Broadcom/the RPI foundation would be interested in releasing the docs on how to use the "Secondary Memory Interface" would they? (hey a girl can ask), failing that, maybe you could point me at someone at broadcom to ask about signing a NDA for personal use only to get at the spec?

Thanks again!
--
Jasmine

ghollingworth, 2018-04-23:
Jasmine,

I would suggest you check out the official driver for the SMI interface...

https://github.com/raspberrypi/linux/commit/7a95d43804a703f0f2df35879e42ef7755743140

And the example NAND driver

https://github.com/raspberrypi/linux/commit/8641eaa864ee2fe40f2eb2acec472ae833d0c8ab

It's possible there are additional bits of information you may need, you could always email me for more information.  

Gordon

jriwanek, 2018-04-23:
Oh wow, Apparently those commits slipped right past me, the last time I looked into it, the SMI was entirely undocumented and I remember seeing it said that it was NDA material, so thanks very much for those pointers.

closed 2018-06-29 by ghollingworth

ptesarik, 2019-04-25:
> 17/18 everything else is unused / unusable bootcode functionality

Are you sure it's unusable? From my experiments with a 3B, it seems that:
- bit 5 enables boot from NAND (SMI)
- bit 8 selects the SMI data bus width (0: 8-bit, 1: 16-bit)
- bits 9-10 change the SMI bus timings for SETUP, HOLD and PACE (0: 1 cycle, 1: 2 cycles, 2: 3 cycles, 3: 10 cycles); the STROBE is always twice as much, except it never takes more than 10 cycles
- bit 27 enables BSC slave boot on GPIO18-19
- bits 30-31 control SPI boot (00: disabled, 01: enabled on GPIO7-11, other values probably select different GPIOs for SPI, but those are unusable)

JamesH65, 2019-04-25:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/otpbits.md

ghollingworth, 2019-04-25:
Yes there is extra functionality that you can enable, but it is unusable because there is no software to support boot from it (it is possible to read bootcode.bin from NAND but bootcode.bin only knows how to read from SD card or USB)

ptesarik, 2019-04-25:
Yes, you're right, I could not boot beyond bootcode.bin. Let me also give out a warning to anyone who's dared to program these undocumented bits: DO NOT TRY TO SET BIT 0 EVER. It bricked my device – it still asks for bootcode.bin (and bootsig.bin) via DHCP, but the code therein is apparently never executed.

ptesarik, 2019-04-25:
Erm, I think I know what happens if you set bit 0. Although the BCM283x bootrom always asks for a `bootsig.bin`, it does not normally do anything with it. I believe bit 0 activates the signature check… Unfortunately, I haven't found any documentation for it.

ptesarik, 2019-05-04:
Let me make one last update on OTP registers 17/18. I noticed that my Raspberry will now try to load bootcode.bin from an exFAT partition. I believe this was not the case with its factory settings. If I'm right, then exFAT support is controlled by bit 6 of OTP register 17/18 (bootmode).

However, there is no support for exFAT in later stages of the boot loader, so this feature is also unusable. I'd say we can fully trust @ghollingworth that the Foundation is not trying to hide any useful features from us, although they could be a bit more verbose on what they are and why they're unusable.

Hash512, 2020-10-04:
> Yes, you're right, I could not boot beyond bootcode.bin. Let me also give out a warning to anyone who's dared to program these undocumented bits: **DO NOT TRY TO SET BIT 0 EVER.** It bricked my [...]

Hi, how did you to set this OTP bit?

JamesH65, 2020-10-04:
I think that publicising a possible way of bricking the Pi isn't the best idea (although not sure it is permanently bricked).