raspberrypi/firmware #975: Netboot not working on RPI 3B - support repeated dhcp discover

nopitoday, 2018-04-16:
I have my rpi set up for network booting. I connect it via ethernet directly to a mainboard which runs dnsmasq. The NIC is from intel. I can only boot when I put bootcode.bin on the sd card. It seems otherwise the intel nic does not come up fast enough since I don't see any dhcp request in wireshark.
Is there anything that could be changed in the firmware to repeat the request after a while (this is often done by other boot roms and also ipxe)  or on the mainboard so the interface is brought up faster?

I don't know if the usb timeout bit is set correctly, however.
https://github.com/raspberrypi/firmware/issues/969#issuecomment-380023631 says to look at the last line which looks like: 66:01009daa
the one starting with 24 (not sure if this is actually the 24th bit?) looks like: 24:ffffffff

I also tried an Asix 88772B usb ethernet nic on the mainboard. This does come up fast enough and fetches bootcode.bin but the following files fail. It seems the interface on the pi seems to go down for a short time after it fetched bootcode.bin. Wireshark sees the file request (to mac aa:aa:aa:aa:aa:aa) but the nic does not send them.

ghollingworth, 2018-04-17:
https://github.com/raspberrypi/firmware/issues/859

The aa:aa:aa:aa:aa:aa means the ARP failed...  It's probably because the switch or the bridge has filtered out the ARP request...

If it works with bootcode.bin on the SD card then stick with that, the problem you are seeing has been fixed in the bootcode but not in the bootrom (i.e. the bootrom is programmed in the silicon and can't be updated)

Gordon

rprr, 2018-04-22:
I have a similar issue as well on a number of pi3bs. The netboot only works if there is an SD card with bootcode.bin on it. The odd thing is that one out of 10 times, netboot works without the SD card.

JamesH65, 2019-01-09:
@ghollingworth Another booting/DHCP issue - are they all related/fixed with newer bootcode?

ghollingworth, 2019-01-17:
It's worth trying again with the latest code if you've got a gateway between them.  If not then you are likely to be suffering from the flushing bug in the bootrom (if you've only got two things in the network some responses can get stuck in the LAN9514 and don't get flushed) you can trigger it to flush through using an occasional ping.

Someone else suggested using nc to send a packet to the booting pi from the server occasionally to trigger the flush...

closed 2019-01-17 by ghollingworth