raspberrypi/firmware #979: Raspberry Pi userspace tools and libraries for armv7

pipiche38, 2018-04-23:
All in the subject. How can we get the userspace tools and librairies for armv7 ?

I currently have a Rpi3B+ installed with Fedora kernel 4.16.3-300.fc28.armv7hl
and would like to get the userspace tools and libraries .

Is there a way to get source code of it

pelwell, 2018-04-23:
"Raspberry Pi userspace tools and libraries" is a bit vague, but take a look at the [userland repo](https://github.com/raspberrypi/userland).

pelwell, 2018-04-24:
Yes, that is one of the utilities built from the code in the userland repo - the source is [gencmd.c](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/gencmd/gencmd.c). The buildme script in the root saves you having to understand CMake.

pipiche38, 2018-04-24:
Got it ! Thanks

closed 2018-04-24 by pipiche38