raspberrypi/firmware #980: RPI 3B+ fails to boot when dell 2408wfp is plugged in. Panic at 1.5s

beren12, 2018-04-25:
It also fails when a dell p2212hf is attached.
It works fine when my 24 or 65" TV is attached.

This is a raspbian install with latest packages installed as of today.

If I force 640x480 mode it boots.
If I force dtoverlay=vc4-kms-v3d it boots but seems to be 1080p not the native 1920x1200.
If I tell it to rotate the screen 90° it boots.

It fails so early the serial console isn't setup yet.

`/opt/vc/bin/edidparser edid.dat` shows:

Enabling fuzzy format match...
Parsing edid.dat...
HDMI:EDID version 1.3, 1 extensions, screen size 52x32 cm
HDMI:EDID features - videodef 0x80 standby suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor S/N descriptor tag 0xff
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is DELL_2408WFP
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 56-76 Hz, horizontal is 30-83 kHz, max pixel clock is 170 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found preferred DMT detail timing format: 1920x1200p @ 60 Hz (68)
HDMI:EDID established timing I/II bytes are A5 4B 00
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 6, 640x480p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 18, 1024x768p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 36, 1280x1024p @ 75 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0x8180 A940 714F 0101 0101 0101 0101 0101 
HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 0
HDMI:EDID found DMT format: code 51, 1600x1200p @ 60 Hz (4:3) in standard timing 1
HDMI:EDID found DMT format: code 21, 1152x864p @ 75 Hz (4:3) in standard timing 2
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:yes, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:1
HDMI:EDID found CEA detail timing format: 1920x1080p @ 60 Hz (16)
HDMI:EDID found CEA detail timing format: 1920x1080i @ 60 Hz (5)
HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz 
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz 
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz 
HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz 
HDMI:EDID found CEA format: code 7, 1440x480i @ 60Hz 
HDMI:EDID found CEA format: code 22, 1440x576i @ 50Hz 
HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz 
HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16|20|24 bits
HDMI:EDID found HDMI VSDB length 5
HDMI:EDID HDMI VSDB has physical address 1.0.0.0
HDMI:EDID HDMI VSDB has no extension fields
HDMI:EDID adding mandatory support for CEA (17) 720x576p @ 50Hz
HDMI:EDID adding mandatory support for CEA (18) 720x576p @ 50Hz
HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
HDMI:EDID best score mode initialised to DMT (4) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
HDMI:EDID best score mode is now CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 43432)
HDMI:EDID best score mode is now CEA (2) 720x480p @ 60 Hz with pixel clock 27 MHz (score 3045736)
HDMI:EDID CEA mode (3) 720x480p @ 60 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID best score mode is now CEA (4) 1280x720p @ 60 Hz with pixel clock 74 MHz (score 3580296)
HDMI:EDID DMT mode (4) 640x480p @ 60 Hz with pixel clock 25 MHz has a score of 36864
HDMI:EDID best score mode is now CEA (5) 1920x1080i @ 60 Hz with pixel clock 74 MHz (score 4149416)
HDMI:EDID DMT mode (6) 640x480p @ 75 Hz with pixel clock 31 MHz has a score of 46080
HDMI:EDID CEA mode (7) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 35368
HDMI:EDID DMT mode (9) 800x600p @ 60 Hz with pixel clock 40 MHz has a score of 57600
HDMI:EDID DMT mode (11) 800x600p @ 75 Hz with pixel clock 49 MHz has a score of 72000
HDMI:EDID best score mode is now CEA (16) 1920x1080p @ 60 Hz

 with pixel clock 148 MHz (score 4711624)
HDMI:EDID DMT mode (16) 1024x768p @ 60 Hz with pixel clock 65 MHz has a score of 94370
HDMI:EDID CEA mode (17) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID CEA mode (18) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID DMT mode (18) 1024x768p @ 75 Hz with pixel clock 78 MHz has a score of 117964
HDMI:EDID DMT mode (21) 1152x864p @ 75 Hz with pixel clock 108 MHz has a score of 174298
HDMI:EDID CEA mode (22) 1440x576i @ 50 Hz with pixel clock 27 MHz has a score of 35368
HDMI:EDID DMT mode (35) 1280x1024p @ 60 Hz with pixel clock 108 MHz has a score of 182286
HDMI:EDID DMT mode (36) 1280x1024p @ 75 Hz with pixel clock 135 MHz has a score of 196608
HDMI:EDID DMT mode (51) 1600x1200p @ 60 Hz with pixel clock 162 MHz has a score of 255400
HDMI:EDID best score mode is now DMT (68) 1920x1200p @ 60 Hz with pixel clock 154 MHz (score 5414720)
HDMI:EDID preferred mode remained as DMT (68) 1920x1200p @ 60 Hz with pixel clock 154 MHz
HDMI:EDID has HDMI support and audio support
/opt/vc/bin/edidparser exited with code 0

I am attaching my edid.dat and a high pic
[image]
[image]

 of the boot log.

beren12, 2018-04-25:
[edid-2408wfp.dat.zip](https://github.com/raspberrypi/firmware/files/1947712/edid-2408wfp.dat.zip)

popcornmix, 2018-04-25:
I booted with your edid (using ``hdmi_edid_file=1``) and it seemed happy:
```
pi@domnfs:~ $ tvservice -s
state 0x12000a [HDMI DMT (68) RGB full 16:10], 1920x1200 @ 60.00Hz, progressive
```
No panics. What does
``
vcgencmd version
uname -a
```
report for you?

beren12, 2018-04-25:
hdmi_ignore_edid=0xa5000080 made no difference it still froze.

I have no usb devices attached, and I used config.txt to force the things that got it to boot:

hdmi_group=2
hdmi_mode=4
or
display_hdmi_rotate=1

let it boot.

I'm also able to plug in the monitor after it's fully booted and do `tvservice -p` and it's fine, I can play emulation station without issue.

beren12, 2018-04-25:
pi@raspberrypi:~ $ vcgencmd version
Apr 16 2018 18:16:56 
Copyright (c) 2012 Broadcom
version af8084725947aa2c7314172068f79dad9be1c8b4 (clean) (release)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux

pelwell, 2018-04-25:
You say:
> If I force dtoverlay=vc4-kms-v3d it boots but seems to be 1080p not the native 1920x1200.

Are you normally using vc4-fkms-v3d or the default VPU-based graphics?

beren12, 2018-04-25:
I should note that I am using an initrd image with raspbian but the kernel and all are stock. I just have a root filesystem on zfs. This panic occurs before any of those modules are loaded however, and it boots fine connected to a tv or nothing at all.

beren12, 2018-04-25:
I'm using default graphics as I wanted this for retropie mainly. No gui installed it's mostly a plain -lite image

beren12, 2018-04-25:
config.txt is currently:
```
dtparam=audio=on
enable_uart=1
hdmi_drive=2
initramfs initrd.img-4.14.34-v7+ followkernel
```

and cmdline.txt is `dwc_otg.lpm_enable=0 console=serial0,115200 root=ZFS=rastank/raspbian-stretch rootfstype=zfs elevator=noop rootwait plymouth.enable=0 zfs.zfs_arc_max=134217728`

beren12, 2018-04-25:
Also, the 2408wfp fails to boot when connected to hdmi or hdmi→dvi, the other dell monitor only has dvi so it's connected hdmi→dvi.

beren12, 2018-04-25:
Think it's a hardware problem? strange that it's monitors not tv showing the problem...

popcornmix, 2018-04-25:
Can you copy the edid to /boot/edid.dat and add to config.txt
```
hdmi_edid_file=1
hdmi_force_hotplug=1
```
Does it still fail? Does it still fail if booted with monitor unplugged?

maxnet, 2018-04-25:
>I should note that I am using an initrd image with raspbian

Probably same issue as here then: https://github.com/raspberrypi/linux/issues/2450

beren12, 2018-04-25:
it works fine if the monitor is unplugged. it maybe related to raspberrypi/linux#2450 but it's not the same if I'm reading it correctly. mine only has a problem with plugged into the 2 monitors I tried, not with the 2 TVs I tried. serial console or not doesn't matter, and it happens every single time. if the pi is hot or cold. I gotta grab the monitor from work to play more since I won't be in the office the next 4 days.

maxnet, 2018-04-25:
Do you also still get the exact same kernel panic if you comment out the initramfs from config.txt for testing purposes?

beren12, 2018-04-25:
I'm not sure. I can't boot it without the image.

maxnet, 2018-04-25:
I know.
But then it should go on for some seconds longer and fail with an "unable to find init" or similar error, instead of the "Bad mode in data abort handler" you have now...

popcornmix, 2018-04-25:
> it works fine if the monitor is unplugged.

Can you clarify this? Is this with:
```
hdmi_edid_file=1
hdmi_force_hotplug=1
```
and are you saying it panics with monitor plugged in and doesn't panic with monitor unplugged?

beren12, 2018-04-25:
no, sorry. i mean w/o anything it works fine. with

```
hdmi_edid_file=1
hdmi_force_hotplug=1
```

It panics. I have to get the actual monitor to check if it doesn't panic w/o the initrd.

popcornmix, 2018-04-25:
Does it panic both with the monitor plugged in and without? (with the quoted config.txt settings).

beren12, 2018-04-25:
I just tried a fresh image, updated it, then added the 

```hdmi_edid_file=1
hdmi_force_hotplug=1```

and it didn't freeze, so maybe it is related to initrd. still, very odd.

beren12, 2018-04-26:
but it's also strange, rotating the video lets it boot, why would that matter?

maxnet, 2018-04-26:
>but it's also strange, rotating the video lets it boot, or 640x480, why would that matter?

Anything that can affect timing or memory layout can cause it to happen/not happen.
While I am only seeing it very occasionally myself (as reported in the linux repo issue), for others it indeed does happen almost every time (for example reported by lucky here: https://github.com/maxnet/berryboot/issues/488 who is using same files as me)

beren12, 2018-04-29:
is there anything that can be done to avoid it? maybe load the the initrd at a slightly different address or something? IIRC you can specify an address instead of just followkernel…

JamesH65, 2019-01-09:
@beren12  Did newer firmware fix this or is it still a problem?