raspberrypi/firmware #981: GPU FFT not working on Pi1 (Unable to enable V3D)

iUltimateLP, 2018-04-26:
Hey there, I tried to compile the hello_fft example on my Raspberry Pi 1 I'm using for driving a 64x64 LED Matrix screen, and I want to use GPU FFT for an audio visualizer on the display. The compilation went successful, but once I try to run the hello_fft.bin example, it returns:

`
Unable to enable V3D. Please check your firmware is up to date.
`

I ran it using
`
/opt/vc/src/hello_pi/hello_fft $ sudo ./hello_fft.bin 8 1
`

The firmware is already up-to-date, one thing to note though is that it's running Raspbian Lite on it, because my only access to it is via SSH so I don't need the whole xserver interface, but still the GPU access should work, right?

Thanks for help!

6by9, 2018-04-26:
Have you enabled the vc4-kms-v3d driver? With that the ARM takes control of the QPUs, therefore anything asking the VPU to submit QPU jobs will fail.

Eric has added a couple of commits to intercept QPU jobs in https://github.com/anholt/linux/tree/rpi-4.14.y-userqpu. Investigating it further is on my list of things to do, but a first pass check implies that it works (although is possibly slower than launching it via the VPU).

popcornmix, 2018-04-26:
Make sure you don't have ``gpu_mem=16`` - I think that disallows use of the 3D hardware.

iUltimateLP, 2018-04-26:
@6by9 I think so, yea. I added "dtoverlay=vc4-kms-v3d" in /boot/config.txt, and doing a lsmod also shows vc4 in the list.

@popcornmix it's set to 128, so that shouldn't be a problem too.

6by9, 2018-04-26:
The vc4 driver will be your problem then.
Your options are therefore to either:
- disable it and revert to the legacy driver (if you're headless then it makes little difference)
- cherry-pick the top couple of commits from Eric's branch, build your own kernel, and live very much on the bleeding-edge.

I will be looking at adding hooks so that with the vc4 driver the VPU will call back to the ARM to execute QPU jobs, but that's not even started yet.

iUltimateLP, 2018-04-26:
Huh, I did enable the vc4 driver, thought it was a good thing. I can disable it, but how do I enable the legacy driver then? AFAIK there wasn't a driver enabled in first place.

6by9, 2018-04-26:
vc4 is the ARM side OpenGL driver. Without that overlay the older VPU based OpenGLES driver is enabled.

If you use raspi-config, then under Advanced Option, GL driver, you have the options of 
- GL (Full KMS) - vc4-kms-v3d overlay
- GL (Fake KMS) - vc4-fkms-v3d overlay (ARM side OpenGL, but leaves driving of the actual display hardware with the firmware)
- Legacy - no overlay specified. This is the default.

iUltimateLP, 2018-04-26:
Thank you! I just switched back to default, and it works. Weird though, that it didn't work before where it was set to default already, I mean this was the reason I switched to the vc4 one. Anyways, thanks for your help, have a nice day :)

closed 2018-04-26 by iUltimateLP