raspberrypi/firmware #986: RPI3B+ bootloader  possibly doesn't handle dhcp correctly

mangodan2003, 2018-05-03:
When booting a bunch (8 in this case) of slave/client RPis simultaneously occasionally 2 or more will be offered the same IP. At first I thought this was an error on the part of the DHCP server, but apparently not [1].

When this happens all PIs offered the same IP fail to boot, where as in theory one should request and get the IP whilst the others will have to start the process again.

[1] http://lists.thekelleys.org.uk/pipermail/dnsmasq-discuss/2016q1/010353.html

We have worked around this for now but this might need investigating.

mangodan2003, 2018-05-03:
on top of this there seems to be a error with the timing of retries, after power on there is a delay of some seconds before the fiest attempt, when this fails the following 3 attempts all happen at near the same time  then a short while later is the final attempt :

`May  3 13:36:13 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:b3:ab:c2
May  3 13:36:13 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.84 b8:27:eb:b3:ab:c2
May  3 13:36:16 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:87:95:fa
May  3 13:36:16 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.76 b8:27:eb:87:95:fa
May  3 13:36:19 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:53:de:0a
May  3 13:36:19 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.50 b8:27:eb:53:de:0a
May  3 13:36:22 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:d9:59:00
May  3 13:36:22 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.83 b8:27:eb:d9:59:00
May  3 13:36:25 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:87:fb:d8
May  3 13:36:25 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.92 b8:27:eb:87:fb:d8
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:2e:94:16
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.73 b8:27:eb:2e:94:16
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:1d:7e:2a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.91 b8:27:eb:1d:7e:2a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:87:fb:d8
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.92 b8:27:eb:87:fb:d8
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:87:95:fa
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.76 b8:27:eb:87:95:fa
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:53:de:0a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.50 b8:27:eb:53:de:0a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:d9:59:00
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.83 b8:27:eb:d9:59:00
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:b3:ab:c2
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.84 b8:27:eb:b3:ab:c2
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:1d:7e:2a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.91 b8:27:eb:1d:7e:2a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:2e:94:16
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.73 b8:27:eb:2e:94:16
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:c3:3c:2e
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.74 b8:27:eb:c3:3c:2e
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:d9:59:00
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.83 b8:27:eb:d9:59:00
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:87:fb:d8
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.92 b8:27:eb:87:fb:d8
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:b3:ab:c2
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.84 b8:27:eb:b3:ab:c2
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:53:de:0a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.50 b8:27:eb:53:de:0a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:87:95:fa
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.76 b8:27:eb:87:95:fa
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:1d:7e:2a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.91 b8:27:eb:1d:7e:2a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:2e:94:16
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.73 b8:27:eb:2e:94:16
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:c3:3c:2e
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.74 b8:27:eb:c3:3c:2e
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:53:de:0a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.50 b8:27:eb:53:de:0a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:87:fb:d8
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.92 b8:27:eb:87:fb:d8
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:d9:59:00
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.83 b8:27:eb:d9:59:00
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:b3:ab:c2
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.84 b8:27:eb:b3:ab:c2
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:87:95:fa
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.76 b8:27:eb:87:95:fa
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:2e:94:16
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.73 b8:27:eb:2e:94:16
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:1d:7e:2a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.91 b8:27:eb:1d:7e:2a
May  3 13:36:28 router dnsmasq-dhcp[772]: DHCPREQUEST(enp1s0) 10.3.2.74 b8:27:eb:c3:3c:2e
[... cut ...]
May  3 13:36:32 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:d9:59:00
May  3 13:36:32 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.83 b8:27:eb:d9:59:00
May  3 13:36:33 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:2e:94:16
May  3 13:36:33 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.73 b8:27:eb:2e:94:16
May  3 13:36:33 router dnsmasq-dhcp[772]: DHCPDISCOVER(enp1s0) b8:27:eb:1d:7e:2a
May  3 13:36:33 router dnsmasq-dhcp[772]: DHCPOFFER(enp1s0) 10.3.2.91 b8:27:eb:1d:7e:2a
`

mangodan2003, 2018-05-03:
Ignore the last post, it seems to be some odd behaviour of syslog timestamping. tcpdump shows there are seconds between each.

JamesH65, 2019-01-09:
@ghollingworth I believe there have been quite a few changes in boocode, are they likely to have fixed this?