raspberrypi/firmware #987: Pi 3B+ won't netboot with bootcode.bin on SD card

booski, 2018-05-03:
This is probably related to #963 and possibly #941.

I have an environment running with ~65 pi 1B and ~10 pi 2B, all netbooted with bootcode.bin on the SD card. I am now working on introducing pi 3B+ into this environment, and am running into two issues that to me seem interconnected.

When booting without an SD card, everything works great, until I try rebooting the 3B+ (by running "reboot" via SSH). Upon reboot, all that happens is this:
 - The red power LED turns on
 - The green status LED starts blinking a continuous pattern of 3 short, 1 long. The long blink is roughly the same length as the 3 short blinks combined.
The connected switch gives a very brief activity blink on the relevant port and then falls silent, and this is what the daemon.log on the DHCP/TFTP server says (serial and MAC obfuscated):
```
May  3 11:35:42 schema-gw dnsmasq-dhcp[3539]: DHCPDISCOVER(eth0) aa:bb:cc:dd:ee:ff
May  3 11:35:42 schema-gw dnsmasq-dhcp[3539]: DHCPOFFER(eth0) 172.17.1.74 aa:bb:cc:dd:ee:ff
May  3 11:35:42 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/bootsig.bin not found
May  3 11:35:42 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/bootcode.bin to 172.17.1.74
May  3 11:35:42 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/xxxxxxxx/start.elf not found
```

When doing a cold boot, the same log says this:
```
May  3 11:40:26 schema-gw dnsmasq-dhcp[3539]: DHCPDISCOVER(eth0) aa:bb:cc:dd:ee:ff
May  3 11:40:26 schema-gw dnsmasq-dhcp[3539]: DHCPOFFER(eth0) 172.17.1.74 aa:bb:cc:dd:ee:ff
May  3 11:40:26 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/bootsig.bin not found
May  3 11:40:26 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/bootcode.bin to 172.17.1.74
May  3 11:40:26 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/xxxxxxxx/start.elf not found
May  3 11:40:26 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/autoboot.txt not found
May  3 11:40:26 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/config.txt to 172.17.1.74
May  3 11:40:26 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/recovery.elf not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/start.elf to 172.17.1.74
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/fixup.dat to 172.17.1.74
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/recovery.elf not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/config.txt to 172.17.1.74
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/dt-blob.bin not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/recovery.elf not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/config.txt to 172.17.1.74
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/bootcfg.txt not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/cmdline.txt to 172.17.1.74
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/recovery8.img not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/recovery8-32.img not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/recovery7.img not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/recovery.img not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/kernel8.img not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/kernel8-32.img not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/armstub8.bin not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: error 0 Early terminate received from 172.17.1.74
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: failed sending /srv/kernel/kernel7.img to 172.17.1.74
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/armstub8-32.bin not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/armstub7.bin not found
May  3 11:40:28 schema-gw dnsmasq-tftp[3539]: file /srv/kernel/armstub.bin not found
May  3 11:40:31 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/kernel7.img to 172.17.1.74
May  3 11:40:31 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/bcm2710-rpi-3-b-plus.dtb to 172.17.1.74
May  3 11:40:31 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/config.txt to 172.17.1.74
May  3 11:40:41 schema-gw dnsmasq-dhcp[3539]: DHCPDISCOVER(eth0) aa:bb:cc:dd:ee:ff
May  3 11:40:41 schema-gw dnsmasq-dhcp[3539]: DHCPOFFER(eth0) 172.17.1.74 aa:bb:cc:dd:ee:ff
May  3 11:40:41 schema-gw dnsmasq-dhcp[3539]: DHCPREQUEST(eth0) 172.17.1.74 aa:bb:cc:dd:ee:ff
May  3 11:40:41 schema-gw dnsmasq-dhcp[3539]: Ignoring domain local for DHCP host name act-lab
May  3 11:40:41 schema-gw dnsmasq-dhcp[3539]: DHCPACK(eth0) 172.17.1.74 aa:bb:cc:dd:ee:ff act-lab
```

Now, since this is basically the exact problem described in #963, I went to try booting with bootcode.bin on the SD card. However, this works even worse - I get the exact same blink pattern, but no activity on the network at all, neither on the switchport LEDs or in the server log.
I have tried with several different versions of bootcode.bin - one pulled directly from this repo's master branch (pulled yesterday), one pulled from the Raspbian Stretch Lite image (found here https://www.raspberrypi.org/downloads/raspbian/), one pulled from this repo's 'next' branch (pulled today), and one taken from an SD card that boots a pi 2B without problems (unsure about the age of that file). I also tried adding a timeout file to no avail. Setting ```program_usb_boot_mode``` also seems to do nothing.

As of yet I have reproduced this behavour on 2 different 3 B+ boards, but I have 64 more to try if necessary.

I would greatly appreciate any pointers as to what I'm doing wrong or how this can be fixed. Ideally I would like to be able to (re)boot completely without SD card, but since all the existing pis in my environment are running off SD cards with bootcode.bin, needing that on the new pis is no big deal.

mangodan2003, 2018-05-03:
just to add that have same problem with reboot  here using rpi3b+ built in bootloader.

3 blinks followed by a long on the green LED.

whilst booted correctly they all have the green led blinking rapidly 4 times followed by a pause. I've not yet found what that means.

tinue, 2018-05-04:
Serial output:
```
Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 1
Trying USB
USB Host enumeration failed, no device detected at root
```

booski, 2018-05-16:
Booting from SD card works reliably for me now (from commit 0f5f899). The reboot issue remains, however, with slightly different symptoms.

This is the log output on the server when rebooting without having an individual directory for the PI serial (in my case /srv/kernel/xxxxxx):
 ```
May 16 11:22:17 schema-gw dnsmasq-dhcp[13233]: DHCPDISCOVER(eth0) aa:bb:cc:dd:ee:ff
May 16 11:22:17 schema-gw dnsmasq-dhcp[13233]: DHCPOFFER(eth0) 172.17.1.75 aa:bb:cc:dd:ee:ff
May 16 11:22:17 schema-gw dnsmasq-tftp[13233]: file /srv/kernel/xxxxxxxx/start.elf not found
```
When I add such a directory (before or after initially booting the PI), things get weirder:
```
May 16 11:32:59 schema-gw dnsmasq-dhcp[13233]: DHCPDISCOVER(eth0) aa:bb:cc:dd:ee:ff
May 16 11:32:59 schema-gw dnsmasq-dhcp[13233]: DHCPOFFER(eth0) 172.17.1.75 aa:bb:cc:dd:ee:ff
```

So if there is no serial directory, the PI asks for one, but if one exists, the PI won't ask for it. Most strange.

booski, 2018-05-16:
No, it logs all TFTP file transfers. My first post contains the log ouptut for a successful boot, and contains lots of lines like this one:
```
May  3 11:40:26 schema-gw dnsmasq-tftp[3539]: sent /srv/kernel/bootcode.bin to 172.17.1.74
```

daggeisler, 2018-05-31:
Tested today with .43 and .44 Kernelversions, but the Problem still exist not reboot after netboot. same enviroment with a Pi3B no Problems with 3b+ the reboot did not work. Got some changes in behavior ... now the Raspi 3b+ will shut down but still no reboot. Any idea when this Problem is solved ?

thanks for your Service Dag

pelwell, 2018-05-31:
How have you updated? Did you include the firmware (start.elf) when updating? I would have expected the release of May 18, 2018 ("kernel: Bump to 4.14.41") to allow a 3B+ to reboot when netbooted, provided it is an ordinary, orderly reboot.

daggeisler, 2018-06-01:
I used the rpi-update on the "server" Raspi then I copied the boot dir to the tftp folder and copied the /lib/modules/ dirs to the nfs Server .....

daggeisler, 2018-06-01:
or what is the correct way for updating in the netboot enviroment ?

booski, 2018-06-26:
Closing this issue as it seems like the remaining part (reboot not working under netboot) is covered by #968.

closed 2018-06-26 by booski