raspberrypi/firmware #988: bootcode.bin does not support the Raspberry Pi 3 Plus model

tinue, 2018-05-04:
The `bootcode.bin` that is available on this repository does not support the Raspberry Pi 3 Plus model. This is the console output that I get:
```
Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 1
Trying USB
USB Host enumeration failed, no device detected at root
```

Compare this to the output from a Raspberry Pi 3 model:
```
Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 1
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 5
Device failed to respond to reset
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC b8:27:eb:6d:a4:07
Wait for Link up
```

Apparently the USB hub device or the host itself is not supported yet.

popcornmix, 2018-05-04:
Ping @ghollingworth

tinue, 2018-05-04:
Thanks for the quick support! I'll check back on Tuesday.

tinue, 2018-05-10:
Just wondering: I checked all three official locations of bootcode.bin and don't seem to find an updated version. Do you know if it is coming?

Which one of these is the best location to check?
* https://github.com/raspberrypi/firmware/blob/master/boot/bootcode.bin
* https://github.com/RPi-Distro/firmware/blob/debian/boot/bootcode.bin
* https://github.com/raspberrypi/firmware/files/1466884/bootcode.bin.gz

Dowster64, 2018-05-12:
Hi, I'm just wondering too.  I've got a piserver application for a number of machine monitors and I guess I'm counting on bootcode.bin working to boot and reboot the monitors,

Thanks in anticipation!

Steve

popcornmix, 2018-05-13:
An update has been pushed.

tinue, 2018-05-13:
Thank you, it works fine.

closed 2018-05-13 by tinue

lowlyocean, 2018-10-21:
@popcornmix @tinnue Please advise location of working bootcode.bin? I am trying from [next branch](https://github.com/raspberrypi/firmware/blob/next/boot/bootcode.bin) without sucess