raspberrypi/firmware #993: Documentation Mailbox Property Voltage

lategoodbye, 2018-05-11:
Hi,
i'm experiment with a hwmon driver as an upstream replacement for vcgencmd measure_volts / get_throttled. I looked at https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#get-voltage (RPI_FIRMWARE_GET_VOLTAGE) and noticed that the wiki entry doesn't match the returned value. The returned value seems like the voltage in microvolts (no offset and scale factor).

But even under this assumption the value for sdram_p (id = 0x000000003) is different to the output of vcgencmd measure_volts sdram_p.

So does `vcgencmd measure_volts` use a different firmware property tag (RPI_FIRMWARE_MEASURE_VOLTAGE)?

pelwell, 2018-05-11:
The GET_VOLTAGE documentation looks like it's been cut-and-pasted - it just returns microvolts, as you say.

The vcgencmd commands are interpreted by the VPU so they don't use mailbox properties, but measure_volts will end up using the same underlying function as RPI_FIRMWARE_MEASURE_VOLTAGE. However, there does appear to be a disconnect between the expected/requested voltage (as returned RPI_FIRMWARE_GET_VOLTAGE) and the actual voltage (as reported by vcgencmd measure_volts).

@popcornmix ?

lategoodbye, 2018-05-11:
Can you please provide value behind RPI_FIRMWARE_MEASURE_VOLTAGE, i cannot find it in the header file?

pelwell, 2018-05-11:
It doesn't exist - it's a hypothetical tag.

closed 2019-07-31 by lategoodbye

ptesarik, 2019-08-01:
Erm, I'm not sure it's right to close this issue.

IIUC the hwmon driver now gets the desired voltages (i.e. whatever value that was set using RPI_FIRMWARE_GET_VOLTAGE), not the actual voltages (as read from the SMPS and/or LDO).

It seems to me that the corresponding hardware registers (A2W_SMPS_xxx) are accessible by the ARM core, so hwmon might be able to read them out directly instead of asking the firmware, but without more information I can't make any sense of the values:

```
A2W_SMPS_LDO0: 0x00613277
A2W_SMPS_LDO1: 0x00327700
A2W_SMPS_A_MODE: 0x00000100
A2W_SMPS_B_STAT: 0x00613277
A2W_SMPS_C_CLK: 0x00613277
A2W_SMPS_L_SPV: 0x00613277
A2W_SMPS_A_VOLTS: 0x00000100
A2W_SMPS_C_CTL: 0x00613277
A2W_SMPS_L_SPA: 0x00613277
A2W_SMPS_A_GAIN: 0x00000000
A2W_SMPS_L_SCV: 0x00613277
A2W_SMPS_L_SCA: 0x00613277
A2W_SMPS_L_SIV: 0x00613277
A2W_SMPS_L_SIA: 0x00613277
```

lategoodbye, 2019-08-11:
In this case, please open a new issue (maybe kernel fits better). The original issue has been "resolved".