raspberrypi/firmware #994: Boot error, what does this mean?

tinue, 2018-05-13:
A Samsung 850 EVO boots in about 50% of the cases on my Raspberry Pi 3 Model B+.

I wanted to debug this, so I am using an SD card with ``bootcode.bin``, and and an empty file ``UART``. Later, I added an empty file ``TIMEOUT``. The extended timeout makes no difference whatsoever.

What I see on the console in case of failure is this:
```
...
Sending and retrying TEST_UNIT_READY
Failed to test volume, probably not ready
init_msd returned -2
Failed initialising mass storage device
```

If the usb boot succeeds this is shown:
```
...
Sending and retrying TEST_UNIT_READY
Sending SCSI_COMMAND_INQUIRY
Sending READ_CAPACITY
init_msd returned 0
...
```

The difference seems to be this ``init_msd`` command. 

What does the response code ``-2`` mean? Not enough power, or any other failure?

Thanks for any help / suggestions!

ghollingworth, 2018-05-14:
Just creating a build with a 10 second timeout to see if this fixes the problem permanently for you...

Gordon

ghollingworth, 2018-05-14:
This bootcode just has the timeout extended to 10 seconds....

[bootcode_extended.zip](https://github.com/raspberrypi/firmware/files/2000012/bootcode_extended.zip)

tinue, 2018-05-14:
Thanks, I‘ll test it tonight!