raspberrypi/firmware #997: RPI3 B+ config.txt gpio not working

liqdfire, 2018-05-16:
`pi@raspberrypi:~/test$ uname -a
Linux raspberrypi 4.14.39-v7+ #1112 SMP Sat May 5 12:01:33 BST 2018 armv7l GNU/Linux
`

`pi@raspberrypi:~/test$ /opt/vc/bin/vcgencmd version
May 11 2018 10:20:16
Copyright (c) 2012 Broadcom
version aab36c8af1b6215d1b2d704cef981303a2f8022b (clean) (release)
`

config.txt
`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
disable_splash=1

gpio=4=ip,pd
gpio=17=ip,pd
`

my expectation is that gpio4 and gpio17 would be exported under _/sys/class/gpio_ and set to input; however, they are not there.

`pi@raspberrypi:/boot$ ls /sys/class/gpio
export  gpiochip0  gpiochip128  unexport
`

what might I be missing?

pelwell, 2018-05-17:
You are missing the fact that that isn't the purpose of the gpio setting/directive/command - read the [announcement](https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=208748]announcement) for more information, but the key lines are:
>  It can be considered an easier alternative to a custom dt-blob.bin for applications that don't need to reassign the named pins (e.g. "LAN_RUN") or create custom clocks.

and

> Pins configured by this command are not reserved in any way, so they won't stop kernel drivers or the user-space sysfs GPIO interface from claiming and using them.

JamesH65, 2019-01-09:
Closing this issue as questions answered/issue resolved.

closed 2019-01-09 by JamesH65