raspberrypi/firmware #998: Invalid tag 1.20180417

takc923, 2018-05-25:
I downloaded raspbian Version:April 2018 from [here](https://www.raspberrypi.org/downloads/raspbian/).
In that image, I saw this changelog.

```
pi@raspberrypi:~ $ zcat /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz | head -5
raspberrypi-firmware (1.20180417-1) stretch; urgency=medium

  * firmware as of 5db8e4e1c63178e200d6fbea23ed4a9bf4656658

 -- Serge Schneider <serge@raspberrypi.org>  Tue, 17 Apr 2018 13:07:14 +0100
```

So, I thought the image's tag was https://github.com/raspberrypi/firmware/releases/tag/1.20180417
But the commit hash in the changelog is incompatible with tha tag.
The tag's commit hash is https://github.com/raspberrypi/firmware/commit/af994023ab491420598bfd5648b9dcab956f7e11
But the changelog's one is https://github.com/raspberrypi/firmware/commit/5db8e4e1c63178e200d6fbea23ed4a9bf4656658

I think the tag's commit hash should be changelog's one https://github.com/raspberrypi/firmware/commit/5db8e4e1c63178e200d6fbea23ed4a9bf4656658

Sorry if I'm wrong, but in that case could you explain why?

Thanks

pelwell, 2018-05-25:
Yes, I think you are correct. The top three commits in the stable branch are (newest first):

- https://github.com/raspberrypi/firmware/commit/5db8e4e1c63178e200d6fbea23ed4a9bf4656658 (Apr 17)
- https://github.com/raspberrypi/firmware/commit/ce8652e2c743f02f04cb29f23611cbf13765483b (Mar 28)
- https://github.com/raspberrypi/firmware/commit/af994023ab491420598bfd5648b9dcab956f7e11 (Mar 23)

But the top three tags are:
- 1.20180417 https://github.com/raspberrypi/firmware/commit/af994023ab491420598bfd5648b9dcab956f7e11
- 1.20180328 https://github.com/raspberrypi/firmware/commit/ce8652e2c743f02f04cb29f23611cbf13765483b
- 1.20180313 https://github.com/raspberrypi/firmware/commit/af994023ab491420598bfd5648b9dcab956f7e11

So that most recent tag looks like a reversion, which I don't think it is, but I'll let @XECDesign confirm.

XECDesign, 2018-05-30:
Well spotted, thanks. Should be okay now.

takc923, 2018-05-30:
Looks good. Thank you!

closed 2018-05-30 by takc923