raspberrypi/firmware #1002: Compute module delay probing for HDMI

rewolff, 2018-05-31:
People keep running into this. And I think the fix is easy.... 

Here are some of the others who have run into this.... 
https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=115095
https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=127503&p=852959#p852959
https://www.raspberrypi.org/forums/viewtopic.php?t=142434
I'll go and apply the config.txt fix to remove the second probe and I'll add the resistor to my PCB, but I think that fixing this in the firmware should be easy.... 

Options: 
* Enable the builtin pullup. For real I2C it is not strong enough, but with nothing attached it will define the level and get a quick NACK on the first attempt. 
* Do a GPIO check first, use the pin as a GPIO, check the level before configuring it as I2C and... when it's low, just skip the whole probe thing. 
* When the first read times out after 500ms simply consider that a fatal error, no further probing necessary? (Or does HDMI allow the device to be still booting and mandate retries in this case?)

In my design I decided I did not need any of the HDMI features, so I simply didn't instantiate the HDMI pins circuit part. So none of those pins were connected.

JamesH65, 2019-01-09:
Hmm, not sure what to do here. The current fix is pretty easy, an entry in config.txt, so not sure spending time on a less 'hacky' fix is worth it for the relatively limited benefit.

rewolff, 2019-01-14:
It bites people in the arse at unexpected times. On CM desgins it seems logical to leave unused modules unconnected. There is not a "big red warning" to hook up pullups on the unused HDMI port, right? 

What's the problem with enabling the (weak) pullups on the I2C pins? As I've explained above, it doesn't influence the operation when I2C IS wired, and it helps in the cases where people build CM boards and run into this unexpected delay. 

With external pullups the impedance with the extra internal pullup the impedance will drop from 1.8k to 1.73k. Should be fine. And the probe will quickly timeout when nothing is connected. 

Some people ALWAYS enable the internal pullup when they configure pins for I2C. 

A valid argument would be: But those pins don't have an internal pullup. But I don't think that this is the case.

ghollingworth, 2019-01-14:
Unfortuately the HDMI I2C pins are not normal 'GPIO' pins, they are in a separate block.  There is no internal pull-up available on these pins and they cannot be read to see if they are low.

Some HDMI monitors take quite some time to power up and respond correctly, which is why we don't just drop out in a short period. (Unfortunately HDMI monitors often don't meet the spec correctly, but we just have to do whatever workaround's we can.

I believe the datasheet has been updated to highlight this fact.

@jimbojr

popcornmix, 2019-01-14:
> When the first read times out after 500ms simply consider that a fatal error, no further probing necessary? (Or does HDMI allow the device to be still booting and mandate retries in this case?)

We actually try reading the EDID ten times before giving up. There are some displays that don't respond reliably initially but do succeed eventually.

rewolff, 2019-01-14:
Yeah, agreed, you don't want to suddenly stop supporting monitors that work with everything else. 

what goes wrong the first few times for devices  that "do not reply reliably initially"? My guess is that they will not respond at all. So you'll get a NACK on the first read. Not a "clock stretching timeout" which happens when the pin is "stuck at zero".  It is the "clock stretching timeout" that takes a significant amount of time, that gets amplified by the "retry ten times".

burtyb, 2019-05-29:
Could the datasheet be updated to make this clearer please?

I'd like to see the "If Unused" - "Leave Open" for HDMI_SCL/HDMI_SDA changing to say it requires a pull up as it's not clear at the moment the pull-up is needed when it's not used.

rewolff, 2019-05-30:
Seconded!

With many "gotchas" in electronics design the datasheet often had the proper information hidden somewhere, where you didn't look. And I find "Ah, its my fault, stupid me, skipped section 5.4 of the datasheet" much preferable to work with than this: "wasn't documented". How many other gotchas are lurking below the surface? Maybe they only matter when working in a warm climate? Only once every 7 days? In that case I can imagine releasing a product that ends up needing an expensive recall later on.... 

With commercial chips you sometimes see such gotchas cause revisions of a datasheet.