raspberrypi/firmware #1004: Possible OTP fail - wrong value

Nightwalker83, 2018-06-07:
Hi,

I have tried to set the One Time Programmable but the output I receive back is "17:3030000a" instead of "17:3020000a". Why would that happen and can it be fixed, if so how?

pelwell, 2018-06-07:
The difference between those two values is in bit 20, which controls the group of GPIOs used by the GPIO bootmode. The boot ROM supports two banks - bank 1 is GPIOs 22-26, and bank 2 is GPIOs 39-43.
There was a bug in the firmware, fixed mid-October 2017, that meant that attempting to enable GPIO bootmode using "program_gpio_bootmode=1" would end up setting bit 20 to select bank 2, but wouldn't set bit 19 to enable GPIO boot mode.

Because OTP stands for "One Time Programmable", this bit can never be cleared. If you were now to set bit 19 the device would use GPIOs 39-43, which would likely be fatal on a Pi other than a Compute Module because you don't have access to those GPIOs. For this reason, the fixed programming logic refuses to set bit 20 if bit 19 is already set unless the user has specified "program_gpio_bootmode=2".

If this isn't OTP bit you are interested in then you'll have to say what you are trying to do - the OTP controls lots of functions.

Nightwalker83, 2018-06-07:
Hi,

Thanks! Enabling the pi to boot from USB rather than sd card is what I am trying to do  So I should set both "program_usb_boot_mode=2" and "program_gpio_bootmode=2"?

pelwell, 2018-06-07:
No - don't use  "program_gpio_bootmode"! You will brick your Pi.

USB boot mode is enabled by bit 29, which is already set in 0x3030000a so you're good to go. Your OTP settings do mean that something has tried to enable GPIO boot mode with an older firmware and you won't ever be able to make use of GPIO boot mode, but I doubt that you would want to do that.

Nightwalker83, 2018-06-07:
Ah ok! Have put "program_usb_boot_mode=1" in /boot/cmdline.txt and rebooted but it did n't work. The pi failed to boot after I removed the sd card and had the hard drive plugged in. I am not sure if I updated the firmware or the  program_usb_boot_mode first.

pelwell, 2018-06-07:
It shouldn't matter - the bit is set now. Note that even with USB boot mode enabled the Pi will still try the SD card first, so you must boot without an SD card present if you want to boot from USB.

Nightwalker83, 2018-06-07:
My pi does not seem to like either GPIO boot mode "1" or "2" anyway! Not sure what else to try?

pelwell, 2018-06-07:
**You don't want GPIO boot mode!** It uses a number of GPIOs to control at boot time which boot modes to try.

Have you written an image to the USB hard drive? If you have and it still doesn't work, try adding `program_usb_timeout=1` to config.txt on an SD card, reboot with that card in, then remove the card and try again.

lurch, 2018-07-01:
Which model of Raspberry Pi are you trying this on? IIRC USB bootmode (with no SD card inserted) only works on the Pi 3B and Pi 3B+

Nightwalker83, 2018-07-02:
Raspberry Pi 3

JamesH65, 2019-01-09:
@Nightwalker83 Did you ever get this sorted out? Can this be closed?

Nightwalker83, 2019-01-10:
Nah, I never found or received an answer to the question although, I don't use Pis anymore so I am not sure if the thread sure be closed or not in case anyone else has the same problem?

closed 2019-01-10 by ghollingworth

pelwell, 2019-01-10:
Your question was answered, although you could be forgiven for missing it:

> Your OTP settings do mean that _something has tried to enable GPIO boot mode with an older firmware_ and you won't ever be able to make use of GPIO boot mode, but I doubt that you would want to do that.

There was a bug in the firmware that meant that trying to enable GPIO boot mode (using `program_gpio_bootmode`) would result in the wrong bits being set. Fortunately this was almost never useful on a Raspberry Pi board (more so on a Compute Module), so very few people should have been affected by it unless they were just trying all of the available options looking for something cool.

The bug was fixed in October 2017, so releases made after that were safe.

Nightwalker83, 2019-01-14:
Ah ok Pelwell, that explains it!