raspberrypi/firmware #1008: rpi-update partially breaks video

DennisT1, 2018-06-12:
Hi,
I ran this update and am now having a problem.  Hopefully there's a fix available.  I have a RPi3B set up as a kiosk device (Raspbian Jesse).  It feeds a VGA monitor via a HDMI to VGA adapter.  The application is a python script run full screen under xterm.
Startup is like:
```
# disable DPMS (Energy Star) features.
xset -dpms
# disable screen saver
xset s off
# don't blank the video device
xset s noblank
# disable mouse pointer
unclutter &
# run window manager
matchbox-window-manager -use_cursor no -use_titlebar no  &
#start x11vnc for remote access
x11vnc -forever -usepw -display :0 &
# run xterm
lxterminal --geometry=150x50 --command /usr/local/bin/TC.sh
```
abbreviated TC.sh is
`unbuffer $BASEPATH/TimeClock.py`

The application monitors activity and turns off the display by making an OS call:
tvservice -o
When activity occurs it then turns the display back on:
tvservice -p >/dev/null 2>&1
  fbset -depth 8;
  fbset -depth 16;

This has all been working fine for months.  I need to add some capability and upgrade some things.  That lead me to rpi-update.
rpi-update results:
```
root@DC0:/home/pi# rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 13403  100 13403    0     0   111k      0 --:--:-- --:--:-- --:--:--  111k
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 4.4.13-v7+
#############################################################
This update bumps to rpi-4.14.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=197689
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    744      0 --:--:-- --:--:-- --:--:--   746
100 55.8M  100 55.8M    0     0  4227k      0  0:00:13  0:00:13 --:--:-- 3808k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.14.48-v7+
 *** depmod 4.14.48+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 31eb6c5d0f7374ba8404104e60618d40f54f1e8c
 *** A reboot is needed to activate the new firmware
root@DC0:/home/pi# reboot
```

Finally, the issue I'm having.  Once I've run rpi-update the script to turn the display back on doesn't work properly.  The display comes back on, but the screen is scrambled.  Colors are messed up and it appears as if the data being displayed is at a different resolution:
[Scrambled screentshot](https://image.ibb.co/k2O75J/20180612_100455.jpg)
Running this
root@DC0:/home/pi# fbset

mode "800x600"
    geometry 800 600 800 600 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/16
endmode

shows the correct settings.

Any help would be appreciated.  Let me know what additional data you may need or tests/changes I can do.

popcornmix, 2018-06-12:
If you just boot to console and run:
```
tvservice -o
tvservice -p >/dev/null 2>&1
fbset -depth 8
fbset -depth 16
```
do you see the same corruption to console framebuffer?

DennisT1, 2018-06-12:
Ok, I did some googling around and found that I could use 
vcgencmd display_power 0 
and
vcgencmd display_power 1 
to turn the screen off and back on.  These work properly so it appears the root problem is with the tvservice command.    As for my project the problem is solved.  As this appears to be an issue with tvservice shall I close this?

popcornmix, 2018-06-13:
Okay I see the problem.
We used to default to a 16bpp framebuffer, but newer versions of raspbian use a 32bpp console.

You are removing the framebuffer console and replacing it with a 16bpp console without X knowing about this. When it redraws you see the corruption.

Changing the final fbset to ``fbset -depth 32`` should avoid this.

But as you've discovered ``vcgencmd display_power`` is a much simpler and better solution.

closed 2018-06-13 by DennisT1