raspberrypi/firmware #1009: vcgencmd get_mem reloc_free always returns 0M

Ealdwulf, 2018-06-13:
This command, which is useful to see if you are using too much GPU mem, no longer seems to work on recent firmware.
I have tried this on Apr 16 and the most recent firmware/kernel pair, and it still gives the same result.
Versions:

```
 $ /opt/vc/bin/vcgencmd version
Apr 16 2018 18:16:56 
Copyright (c) 2012 Broadcom
version af8084725947aa2c7314172068f79dad9be1c8b4 (clean) (release)
$ uname -a
Linux kano 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux
```

```
$ /opt/vc/bin/vcgencmd version
Jun  7 2018 15:31:38 
Copyright (c) 2012 Broadcom
version 4800f08a139d6ca1c5ecbee345ea6682e2160881 (clean) (release)
$ uname -a
Linux kano 4.14.48-v7+ #1118 SMP Thu Jun 7 16:40:27 BST 2018 armv7l GNU/Linux
```

pelwell, 2018-06-14:
I think you've mis-read or mis-remembered the parameters (not helped by the fact that unrecognised parameters return 0 rather than failing). You want the `reloc` and `malloc` options.

From https://www.raspberrypi.org/forums/viewtopic.php?t=158157#p1029332:

```
arm:          total memory assigned to arm
gpu:          total memory assigned to gpu
malloc_total: total memory assigned to gpu malloc heap
malloc:       free gpu memory in malloc heap
reloc_total:  total memory assigned to gpu relocatable heap
reloc:        free gpu memory in relocatable heap
```

Ealdwulf, 2018-06-14:
Oops. Quite right, sorry. Closing...

closed 2018-06-14 by Ealdwulf