raspberrypi/firmware #1010: Would love a dual boot option (moved from Hexxeh/rpi-firmware)

Osolemio, 2018-06-20:
Hello,

As we know, PasPi can boot and continue on or from microSD or from USB stick either. That's hardcoded in /boot/config.txt
I used this feature for developed systems on RasPi, RasPi2 & RasPi3. But some time ago I had gone to RasPi based systems with microSD cards only and RO linux.
The question:
May we have an option, that allows us to start from microSD, continue from the USB stick when it's available, otherwise falls down back to microSD rootfs?
Why? I have a line of embedded devices based on RasPi. Some more or less small updates I can distribute via simple updates downloadable via web-interface . But sometimes I would need a complete microSD re-burning ant this is an issue: for big system updates, FW updates, recover and so on. Customers have no physical access to sd card, but USB slots. I spent many time for different attempts of re-flashing microSD "on the fly", but either one wasn't fully acceptable and safe. So, finally, I decided to leave my request here, because can't see another possibility beside realizing this feature: we start booting from SD, continue on from USB, then flash SD. If we had such an option I believe it would be fine. Even though a customer has to watch the system messages on console, but it's almost safe, because 1st partition will be burned very fast and RasPi will stay bootable even with other partitions corrupted. Thank you very much.

wx4cb, 2018-06-20:
I may be wrong, but, the problem i can see is to swap the root FS from one device or another, you would have to trigger a reboot with a new rootfs line in the kernel command line. That would not be limited to pi's but any linux box afaik.

Osolemio, 2018-06-20:
Of course, but config.txt is read by start.elf (GPU). Then start.elf reads cmdline.txt and starts kernel.img with these parameters. The availability of USB stick is known on 1st boot stage AFAIK. So, kernel may have different boot options for rootfs depending on USB stick availability. I would love the GPU system provides 2 options for kernel rootfs: if USB is ok, then /sdaN, otherwise, say, mmcblk0p2

Osolemio, 2018-06-20:
pi3b+ is very good piece of news. Really. What shall I do with many hundreds installed devices starting from B+? :)

wx4cb, 2018-06-20:
@Osolemio  yes but my point was that if you wanted to swap file system while it's running then that's not going to work

Ferroin, 2018-06-20:
@wx4cb Actually no, you can do that online using the `pivot_root` command.  In fact, this is an integral part of the startup process when using an initramfs (although they use the `switch_root` command instead, for somewhat complicated reasons that are irrelevant for what you're talking about).  Doing this is complicated (you essentially have to bring userspace down to just the init system and nothing else), but it is possible.

Alternatively, if you're willing to use a custom initramfs, it's fully possible to do the type of auto-detection being suggested by @Osolemio (and that is arguably the correct place to do it for this situation, provided you can use the same kernel for both systems).

Osolemio, 2018-06-20:
@wx4cb actually no, I don't want to. I want to have just a boot priority.
If we're about RasPi3B+, I'd like to ask you beforehand to add the boot priority in it: USB, then microSD. It would avoid many issues for embedded devices. thank you.

Osolemio, 2018-06-28:
@wx4cb 
I've bought 2 RasPi 3 B+ from the UK. Have done some experiments with USB booting.
What I liked and disliked:
1. There's no microSD, one USB stick: good, no problem. Processed as expected.
2. There's a corrupted microSD, one USB stick (bootable). Won't boot at all. Stuck. So sad.
3. There's both microSD and USB bootable. It's booting from the microSD. I won't be able to re-burn the microSD in such a case.

So, the feature is very stunning, but is almost useless for me without boot priority. What I'd like to have for embedded devices:

1. If microSD is corrupted, it falls back to USB booting
2. If it's able to boot both from the USB and microSD, the USB would be the first.

Thank you very much for the great job you did!

Osolemio, 2018-06-29:
UPD: Tried with formatted microSD. Boots from USB, which is partially ok. Actually, /boot partition might be corrupted very very rarely. 
Anyway, the most important that I still cannot make it start from MSD before microSD. May we have some option in OTP which device is the first to start from?

ghollingworth, 2018-06-29:
Have you tried gpio_bootmode?

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow.md

Osolemio, 2018-06-29:
@ghollingworth 

Of course, I did. Moreover USB boot works well on RPi3B, not only 3B+
I meant that I need boot priority, when both MSD and microSD are bootable. In that case Rpi starts from microSD.
I found a solution: to re-burn a microSD, I rename bootcode.bin into .bak, then reboot, it starts from MSD, executes dd, then reboot. If everything is ok, Rpi starts from microSD. Bingo.

ghollingworth, 2018-06-29:
OK, clearly you didn't read about gpio_bootmode...  Try again...

Osolemio, 2018-06-29:
@ghollingworth 

Yep, going over there. Thank you

Osolemio, 2018-06-29:
@ghollingworth 
Ok, re-read that in 10th time.
Probably I don't understand something, but I see:

If enabled: check primary SD for bootcode.bin on GPIO 48-53
    Success - Boot
    Fail - timeout (five seconds)
If enabled: check secondary SD
    Success - Boot
    Fail - timeout (five seconds)
If enabled: check NAND
If enabled: check SPI
If enabled: check USB 

So I see that in any conditions USB is being checked always after microSD. I can only disable microSD.
How I can set the priority of an USB over a microSD? Sorry, but I can't understand that.

Osolemio, 2018-06-29:
Anyway, I found a workaround for 3B, 3B+, thanks 2 all. The issue may be closed.

closed 2018-06-29 by Osolemio

ghollingworth, 2018-06-29:
No, I'm pointing you at gpio_bootmode:

> The GPIO boot mode OTP bits can be programmed by adding  program_gpio_bootmode=n to config.txt, where n is 1 to select the low bank (22-26) or 2 to select the high bank (39-43). Once added, boot the device, then power-cycle it (rebooting is not sufficient). You should expect it to no longer boot (all boot modes will be disabled by default). Apply a pull-up to the required pin to enable the required boot mode. After programming, the config.txt setting can be removed.

Osolemio, 2018-06-29:
@ghollingworth 

If you want me to apply a pull-up to a pin, I would tell you that my embedded devices are already built-in. The customer has no access to them. The circuit board also has no GPIO interconnections. So, it's not the solution in my case. Thank you anyway. It may be very useful in some cases.