raspberrypi/firmware #1012: RPi3, Rpi3B+ network boot: config.txt ignored

mkreisl, 2018-06-27:
While testing a sd-card less network boot of Pi3 and Pi3B+ again, I noticed that `config.txt` seems to be ignored.

All files are loaded, but all entries in config.txt have not effect. 
For example,  `dtoverlay=gpio-ir` should load `gpio-ir-recv` module but it is not loaded

Here is a logfile of my tftp server:

```
Jun 27 23:16:58 kmcubie in.tftpd[30827]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:16:58 kmcubie tftpd[30828]: tftpd: trying to get file: bootcode.bin
Jun 27 23:16:58 kmcubie tftpd[30828]: tftpd: serving file from /srv/tftp
Jun 27 23:16:58 kmcubie in.tftpd[30829]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:16:58 kmcubie tftpd[30830]: tftpd: trying to get file: bootsig.bin
Jun 27 23:16:58 kmcubie tftpd[30830]: tftpd: serving file from /srv/tftp
Jun 27 23:16:58 kmcubie in.tftpd[30831]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:16:58 kmcubie tftpd[30833]: tftpd: trying to get file: 2caf9c61/start.elf
Jun 27 23:16:58 kmcubie tftpd[30833]: tftpd: serving file from /srv/tftp
Jun 27 23:16:58 kmcubie in.tftpd[30837]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:16:58 kmcubie tftpd[30841]: tftpd: trying to get file: 2caf9c61/autoboot.txt
Jun 27 23:16:58 kmcubie tftpd[30841]: tftpd: serving file from /srv/tftp
Jun 27 23:16:58 kmcubie in.tftpd[30842]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:16:58 kmcubie tftpd[30846]: tftpd: trying to get file: 2caf9c61/config.txt
Jun 27 23:16:58 kmcubie tftpd[30846]: tftpd: serving file from /srv/tftp
Jun 27 23:16:58 kmcubie in.tftpd[30850]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:16:58 kmcubie tftpd[30854]: tftpd: trying to get file: 2caf9c61/recovery.elf
Jun 27 23:16:58 kmcubie tftpd[30854]: tftpd: serving file from /srv/tftp
Jun 27 23:16:58 kmcubie in.tftpd[30855]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:16:58 kmcubie tftpd[30859]: tftpd: trying to get file: 2caf9c61/start.elf
Jun 27 23:16:58 kmcubie tftpd[30859]: tftpd: serving file from /srv/tftp
Jun 27 23:17:06 kmcubie in.tftpd[30878]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:06 kmcubie tftpd[30879]: tftpd: trying to get file: 2caf9c61/fixup.dat
Jun 27 23:17:06 kmcubie tftpd[30879]: tftpd: serving file from /srv/tftp
Jun 27 23:17:06 kmcubie in.tftpd[30880]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:06 kmcubie tftpd[30881]: tftpd: trying to get file: 2caf9c61/recovery.elf
Jun 27 23:17:06 kmcubie tftpd[30881]: tftpd: serving file from /srv/tftp
Jun 27 23:17:06 kmcubie in.tftpd[30882]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:06 kmcubie tftpd[30883]: tftpd: trying to get file: 2caf9c61/config.txt
Jun 27 23:17:06 kmcubie tftpd[30883]: tftpd: serving file from /srv/tftp
Jun 27 23:17:06 kmcubie in.tftpd[30884]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:06 kmcubie tftpd[30885]: tftpd: trying to get file: 2caf9c61/dt-blob.bin
Jun 27 23:17:06 kmcubie tftpd[30885]: tftpd: serving file from /srv/tftp
Jun 27 23:17:06 kmcubie in.tftpd[30886]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:06 kmcubie tftpd[30887]: tftpd: trying to get file: 2caf9c61/recovery.elf
Jun 27 23:17:06 kmcubie tftpd[30887]: tftpd: serving file from /srv/tftp
Jun 27 23:17:06 kmcubie in.tftpd[30888]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:06 kmcubie tftpd[30889]: tftpd: trying to get file: 2caf9c61/config.txt
Jun 27 23:17:06 kmcubie tftpd[30889]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30890]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30891]: tftpd: trying to get file: 2caf9c61/bootcfg.txt
Jun 27 23:17:07 kmcubie tftpd[30891]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30892]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30893]: tftpd: trying to get file: 2caf9c61/cmdline.txt
Jun 27 23:17:07 kmcubie tftpd[30893]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30894]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30895]: tftpd: trying to get file: 2caf9c61/recovery8.img
Jun 27 23:17:07 kmcubie tftpd[30895]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30896]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30897]: tftpd: trying to get file: 2caf9c61/recovery8-32.img
Jun 27 23:17:07 kmcubie tftpd[30897]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30898]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30899]: tftpd: trying to get file: 2caf9c61/recovery7.img
Jun 27 23:17:07 kmcubie tftpd[30899]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30900]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30901]: tftpd: trying to get file: 2caf9c61/recovery.img
Jun 27 23:17:07 kmcubie tftpd[30901]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30902]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30903]: tftpd: trying to get file: 2caf9c61/kernel8.img
Jun 27 23:17:07 kmcubie tftpd[30903]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30904]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30905]: tftpd: trying to get file: 2caf9c61/kernel8-32.img
Jun 27 23:17:07 kmcubie tftpd[30905]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30906]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30907]: tftpd: trying to get file: 2caf9c61/kernel7.img
Jun 27 23:17:07 kmcubie tftpd[30907]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30908]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30909]: tftpd: trying to get file: 2caf9c61/kernel.img
Jun 27 23:17:07 kmcubie tftpd[30909]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30910]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30911]: tftpd: trying to get file: 2caf9c61/armstub8.bin
Jun 27 23:17:07 kmcubie tftpd[30911]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30912]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30913]: tftpd: trying to get file: 2caf9c61/armstub8-32.bin
Jun 27 23:17:07 kmcubie tftpd[30913]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30914]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30915]: tftpd: trying to get file: 2caf9c61/armstub7.bin
Jun 27 23:17:07 kmcubie tftpd[30915]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30916]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30917]: tftpd: trying to get file: 2caf9c61/armstub.bin
Jun 27 23:17:07 kmcubie tftpd[30917]: tftpd: serving file from /srv/tftp
Jun 27 23:17:07 kmcubie in.tftpd[30918]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:07 kmcubie tftpd[30919]: tftpd: trying to get file: 2caf9c61/kernel.img
Jun 27 23:17:07 kmcubie tftpd[30919]: tftpd: serving file from /srv/tftp
Jun 27 23:17:18 kmcubie in.tftpd[30977]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:18 kmcubie tftpd[30978]: tftpd: trying to get file: 2caf9c61/initramfs.gz
Jun 27 23:17:18 kmcubie tftpd[30978]: tftpd: serving file from /srv/tftp
Jun 27 23:17:49 kmcubie in.tftpd[31076]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:49 kmcubie tftpd[31077]: tftpd: trying to get file: 2caf9c61/bcm2710-rpi-3-b.dtb
Jun 27 23:17:49 kmcubie tftpd[31077]: tftpd: serving file from /srv/tftp
Jun 27 23:17:49 kmcubie in.tftpd[31078]: connect from 192.168.1.66 (192.168.1.66)
Jun 27 23:17:49 kmcubie tftpd[31079]: tftpd: trying to get file: 2caf9c61/config.txt
Jun 27 23:17:49 kmcubie tftpd[31079]: tftpd: serving file from /srv/tftp

```

Btw, I'm wondering why some files are requested multiple times

boodcode.bin, start.elf etc pp are from June, 22.

bootcode.bin is in tftp root, all other files from /boot are located in subolder == serial-number

```
[23:30:10][root@kmcubie:~]$ ls -la /srv/tftp
insgesamt 68
drwxr-xr-x  4 root root  4096 Jun 27 22:35 .
drwxr-xr-x 13 root root  4096 Jun 17 15:04 ..
drwxr-xr-x  3 root root  4096 Jun 27 23:15 2caf9c61
-rwxr-xr-x  1 root root 52116 Jun 22 21:13 bootcode.bin
drwxr-xr-x  3 root root  4096 Jun 27 22:34 ef8356fd
```

mkreisl, 2018-06-27:
Hmmm, how does the Pi get access to the overlay (*.dtbo) files if no sd-card is present?

pelwell, 2018-06-28:
The Pi reads the overlays the same way it reads the kernel image and config.txt - there is a filing system abstraction that works over SD card, USB drive and TFTP. config.txt is parsed multiple times, and each of those will appear as a separate file transfer.

If you want a proper analysis of your log, try capturing just one boot session - it's not clear where one ends and the next begins (if indeed it does) in your posted log.

To narrow down your config.txt problem, try using tcpdump/tshark/wireshark to sniff the packets and put the capture somewhere I can download it.

mkreisl, 2018-06-28:
> If you want a proper analysis of your log, try capturing just one boot session - it's not clear where one ends and the next begins (if indeed it does) in your posted log.

The log I posted was **one** boot process

And I can not see any access of dtbo files in folder overlays. So, being unsure if tcpdump helps.

I'm not using Rasbian, I was trying to boot XBian without a sd-card. Maybe the difference is that I'm using initramfs.gz (it is required) and Raspbian does not

Is there a limitation in size of config.txt, because it seems that some settings working (for example disable rainbow square) and some not (setting arm_freq clock)

pelwell, 2018-06-28:
> The log I posted was one boot process

I thought it might be, but wanted to confirm.

> And I can not see any access of dtbo files in folder overlays. So, being unsure if tcpdump helps.

I want to see the contents of the config.txt being sent.

mkreisl, 2018-06-28:
@pelwell Ok, will see if I can generate a capture, will take some time

Thanks for your help

mkreisl, 2018-06-28:
> Is there a limitation in size of config.txt, because it seems that some settings working (for example disable rainbow square) and some not (setting arm_freq clock)

Yep, seems to be

Moved `dtparam=tx_lpi_timer=6000` at the beginning of the file and now it is set

Will do some more tests after removing all comments

pelwell, 2018-06-28:
There does seem to be a 4096 byte limit for config.txt, but I haven't worked out where that limit is coming from.

mkreisl, 2018-06-28:
Ah, ok, mine is bigger (6577 bytes). After removing all comments it works bettter, but `gpio-ir` still not working (overlay is loaded but module `gpio-ir-recv` not, putting module into initramfs does not help)

```
Jun 28 14:57:24 kmcubie tftpd[20116]: tftpd: trying to get file: ef8356fd/bcm2710-rpi-3-b-plus.dtb
Jun 28 14:57:24 kmcubie tftpd[20116]: tftpd: serving file from /srv/tftp
Jun 28 14:57:24 kmcubie in.tftpd[20117]: connect from 192.168.1.51 (192.168.1.51)
Jun 28 14:57:25 kmcubie tftpd[20118]: tftpd: trying to get file: ef8356fd/config.txt
Jun 28 14:57:25 kmcubie tftpd[20118]: tftpd: serving file from /srv/tftp
Jun 28 14:57:25 kmcubie in.tftpd[20119]: connect from 192.168.1.51 (192.168.1.51)
Jun 28 14:57:25 kmcubie tftpd[20120]: tftpd: trying to get file: ef8356fd/overlays/gpio-ir.dtbo
Jun 28 14:57:25 kmcubie tftpd[20120]: tftpd: serving file from /srv/tftp

```

So, what's the next limitation?

And reboot still does not work on Pi3B+, but this is already reported in another issue here

JamesH65, 2018-06-28:
There's does appear to be a 80 character line length limit in one bit of code I looked at, could that be an issue?

pelwell, 2018-06-28:
@mkreisl gpio_ir_recv does load with gpio-ir on my network-booted 3B+. If you have `vcdbg` available on your system, what does `sudo vcdbg log msg |& grep -vi hdmi` report?

mkreisl, 2018-06-28:
@pelwell Sure, it is already there

```
root@kmxbilr2 ~ # vcdbg log msg |& grep -vi hdmi
021804.939: brfs: File read: /mfs/sd/config.txt
021805.582: brfs: File read: 508 bytes
021969.696: brfs: File read: /mfs/sd/config.txt
022291.307: *** Restart logging
022291.372: brfs: File read: 508 bytes
022405.389: brfs: File read: /mfs/sd/cmdline.txt
022405.554: Read command line from file 'cmdline.txt':
022405.629: 'telnet zswap.enabled=1 zswap.compressor=lz4 console=tty1 cnet=192.168.1.24:192.168.1.6:192.168.1.10:::eth0:off root=iSCSI=iqn.2017-12.com.kmhome:kmxbilr2,192.168.1.6:3260,UUID=29e1091f-1092-4787-9748-271d9a670f41 rootflags=subvol=root/@,autodefrag,compre
022405.979: brfs: File read: 431 bytes
033597.485: brfs: File read: /mfs/sd/kernel.img
033597.549: Loading 'kernel.img' to 0x8000 size 0x5a7b58
033597.618: brfs: File read: 5929816 bytes
056878.389: brfs: File read: /mfs/sd/initramfs.gz
056878.450: Loading 'initramfs.gz' to 0x5afb58 size 0xc78075
059378.261: No kernel trailer - assuming DT-capable
059378.353: brfs: File read: 13074549 bytes
059493.367: brfs: File read: /mfs/sd/bcm2710-rpi-3-b-plus.dtb
059493.434: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x5afb58 size 0x6031
059635.577: brfs: File read: 24625 bytes
059672.190: brfs: File read: /mfs/sd/config.txt
059672.278: dtparam: tx_lpi_timer=6000
059688.205: dtparam: audio=on
059706.657: brfs: File read: 508 bytes
059772.709: dterror: not a valid FDT - err -9
059772.727: Failed to load overlay 'gpio-ir'
059772.794: dtparam: eee=on
059792.163: brfs: File read: /mfs/sd/overlays/gpio-ir.dtbo
060816.877: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
060930.641: Device tree loaded to 0x2eff9a00 (size 0x65c8)
064808.616: vchiq_core: vchiq_init_state: slot_zero = 0xef880000, is_master = 1
064818.748: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x0000001f
```

Does not look good:

```
059772.709: dterror: not a valid FDT - err -9
059772.727: Failed to load overlay 'gpio-ir'
```

Corresponding config line in config.txt is `dtoverlay=gpio-ir,gpio_pin=22,rc-map-name=rc-nec-terratec-cinergy-xs`

and command line is truncated

mkreisl, 2018-06-28:
@JamesH65 
> There's does appear to be a 80 character line length limit in one bit of code I looked at, could that be an issue?

Hmmm, most of comment lines having exact 80 character, but all others are significantly shorter

pelwell, 2018-06-28:
-9 is ERR_BADMAGIC - the file doesn't look like an FDT file. My gpio-ir.dtbo is 1212 bytes long, but the log above suggests yours is only 508 bytes long - does that match what you see on the the tftp server?

mkreisl, 2018-06-28:
@pelwell Here we go. Captured via tcpdump and investigated by Wireshark. Looks ok for me

http://www.mediafire.com/file/jki8yl6tr06k13n/tcpdump.log/file

pelwell, 2018-06-28:
Thanks, I agree - it does all look OK.

In the absence of much of a theory about what could be going wrong, you could try loading a few overlays - gpio-shutdown and sdtweak should be harmless, and see if any load and whether order is significant.

pelwell, 2018-06-28:
Just a thought: the use of initramfs is a differentiator - have you tried using "followkernel" as a load address instead of an explicit address?

mkreisl, 2018-06-28:
> Just a thought: the use of initramfs is a differentiator - have you tried using "followkernel" as a load address instead of an explicit address?

`initramfs initramfs.gz followkernel`: does not make any difference
`dtoverlay=sdtweak,poll_once=on` does not work

pelwell, 2018-06-28:
So with multiple overlays being loaded, all get the same `not a valid FDT - err -9` error?

mkreisl, 2018-06-28:
> So with multiple overlays being loaded, all get the same not a valid FDT - err -9 error?

Yes, and swapped order of dtoverlay, put them to beginning or to end, does not make any differenct

pelwell, 2018-06-28:
OK, thanks - very strange.

I'm a regular user of network booting (using dnsmasq) without any problems, so we "just" have to find the crucial difference...

mkreisl, 2018-06-28:
> OK, thanks - very strange.
> 
> I'm a regular user of network booting (using dnsmasq) without any problems, so we "just" have to find the crucial difference...

Yes, indeed.. And the broken reboot function on Pi3B+ makes me crazy. In the meantime, switched back to Pi3, but same overlay problem here, but that was expected.

Probably the use of initramfs makes the difference. Will try to create a configuration without the need of initramfs in the next days.

mkreisl, 2018-06-28:
Initramfs is not the difference. Booted without initramfs into nfsroot, same issue

```
027162.363: brfs: File read: /mfs/sd/config.txt
027164.041: brfs: File read: 3329 bytes
027295.131: brfs: File read: /mfs/sd/config.txt
027297.147: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
027494.629: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
027592.508: *** Restart logging
027592.571: brfs: File read: 3329 bytes
027714.595: brfs: File read: /mfs/sd/cmdline.txt
027714.725: Read command line from file 'cmdline.txt':
027714.796: 'zswap.enabled=1 zswap.compressor=lz4 console=tty1 root=/dev/nfs rootfstype=nfs ip=192.168.1.24:192.168.1.6:192.168.1.10:::eth0:off nfsroot=/srv/nfs/kmxbilr2,tcp,vers=3 logo.nologo noswap partswap startevent=mountall selinux=0 splash nohdparm net.ifnames=
027715.124: brfs: File read: 295 bytes
039006.479: brfs: File read: /mfs/sd/kernel.img
039006.544: Loading 'kernel.img' to 0x8000 size 0x58d3b8
039006.613: brfs: File read: 5821368 bytes
039043.589: Failed to load 'initramfs.gz' - initramfs disabled
039047.251: No kernel trailer - assuming DT-capable
039125.614: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
039125.680: Loading 'bcm2710-rpi-3-b.dtb' to 0x5953b8 size 0x5efe
039260.775: brfs: File read: 24318 bytes
039303.762: brfs: File read: /mfs/sd/config.txt
039304.455: dtparam: audio=on
039322.958: brfs: File read: 3329 bytes
039410.951: dterror: not a valid FDT - err -9
039410.972: Failed to load overlay 'gpio-ir'
039473.957: dterror: not a valid FDT - err -9
039473.975: Failed to load overlay 'sdtweak'
039475.412: brfs: File read: /mfs/sd/overlays/sdtweak.dtbo
040538.079: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
040644.260: Device tree loaded to 0x27fe9c00 (size 0x63b5)
043200.477: vchiq_core: vchiq_init_state: slot_zero = 0xe7580000, is_master = 1
043209.046: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x0000001f
```

maxnet, 2018-07-01:
Did was able to successfully use overlays while PXE booting in the past.

One thing in which your setup differs from mine is that it seems you are using serial number specific folders (ef8356fd/) though.
Have you also tried without those to rule out there is a bug in the code that handles that?
Note that as an alternative you can put "tftproot-unique-root=mac" in dnsmasq.conf and use folders based on dash separated MAC address (b8-27-eb-83-56-fd/) instead.

mkreisl, 2018-07-01:
> One thing in which your setup differs from mine is that it seems you are using serial number specific folders (ef8356fd/) though.
> Have you also tried without those to rule out there is a bug in the code that handles that?

No, but investigated network traffic and content was sent to client correctly

> Note that as an alternative you can put "tftproot-unique-root=mac" in dnsmasq.conf and use folders based on dash separated MAC address (b8-27-eb-83-56-fd/) instead.

Actually, I do not use dnsmasq as tftp server, I'm using tftp instead

maxnet, 2018-07-01:
>Actually, I do not use dnsmasq as tftp server, I'm using tftp instead

Ah, there indeed seems to be something up with that.
Gave your `dtoverlay=gpio-ir,gpio_pin=22,rc-map-name=rc-nec-terratec-cinergy-xs` line a try, and managed to reproduce your problem.

With dnsmasq as TFTP server, it does work:

```
016962.342: brfs: File read: /mfs/sd/config.txt
016963.129: brfs: File read: 1660 bytes
017016.350: brfs: File read: /mfs/sd/config.txt
017238.632: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
017353.950: *** Restart logging
017354.018: brfs: File read: 1660 bytes
017357.553: brfs: File read: /mfs/sd/cmdline.txt
017357.670: Read command line from file 'cmdline.txt':
017357.741: 'dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 elevator=deadline rootwait quiet splash plymouth.ignore-serial-consoles ip=dhcp root=/dev/nfs nfsroot=192.168.178.141:/var/lib/piserver/os/Raspbian-2017-11-29,v3'
017358.071: brfs: File read: 218 bytes
020387.951: brfs: File read: /mfs/sd/kernel7.img
020388.018: Loading 'kernel7.img' to 0x8000 size 0x4b31f0
020391.691: No kernel trailer - assuming DT-capable
020391.779: brfs: File read: 4927984 bytes
020406.058: brfs: File read: /mfs/sd/bcm2710-rpi-3-b-plus.dtb
020406.126: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x4bb1f0 size 0x602d
020548.689: brfs: File read: 24621 bytes
020551.329: brfs: File read: /mfs/sd/config.txt
020551.476: brfs: File read: 1660 bytes
020554.207: brfs: File read: /mfs/sd/overlays/gpio-ir.dtbo
020562.930: Loaded overlay 'gpio-ir'
020562.984: dtparam: gpio_pin=22
020564.581: dtparam: rc-map-name=rc-nec-terratec-cinergy-xs
020565.575: dtparam: audio=on
021689.897: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
021798.571: Device tree loaded to 0x2eff9a00 (size 0x65d2)
022983.203: vchiq_core: vchiq_init_state: slot_zero = 0xfad80000, is_master = 1
022993.423: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010
```

With tftpd it does not:

```
016892.833: brfs: File read: /mfs/sd/config.txt
016893.621: brfs: File read: 1660 bytes
016963.344: brfs: File read: /mfs/sd/config.txt
017185.642: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
017272.945: *** Restart logging
017273.013: brfs: File read: 1660 bytes
017281.168: brfs: File read: /mfs/sd/cmdline.txt
017281.285: Read command line from file 'cmdline.txt':
017281.355: 'dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 elevator=deadline rootwait quiet splash plymouth.ignore-serial-consoles ip=dhcp root=/dev/nfs nfsroot=192.168.178.141:/var/lib/piserver/os/Raspbian-2017-11-29,v3'
017281.686: brfs: File read: 218 bytes
020167.672: brfs: File read: /mfs/sd/kernel7.img
020167.738: Loading 'kernel7.img' to 0x8000 size 0x4b31f0
020171.416: No kernel trailer - assuming DT-capable
020171.506: brfs: File read: 4927984 bytes
020195.280: brfs: File read: /mfs/sd/bcm2710-rpi-3-b-plus.dtb
020195.348: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x4bb1f0 size 0x602d
020337.843: brfs: File read: 24621 bytes
020345.176: brfs: File read: /mfs/sd/config.txt
020345.326: brfs: File read: 1660 bytes
020347.368: dterror: not a valid FDT - err -9
020347.387: Failed to load overlay 'gpio-ir'
020347.966: dtparam: audio=on
020367.558: brfs: File read: /mfs/sd/overlays/gpio-ir.dtbo
021413.555: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
021520.778: Device tree loaded to 0x2eff9b00 (size 0x64a8)
022709.962: vchiq_core: vchiq_init_state: slot_zero = 0xfad80000, is_master = 1
022720.499: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010
```

Only obvious difference between the two TFTP sessions seems to be that tftpd does not support the tsize  tftp extension, while dnsmasq does...
Wonder if the client does depend on that? (but then why does it not have a problem fetching the other boot files?)

mkreisl, 2018-07-01:
Very interesting

> Only obvious difference between the two TFTP sessions seems to be that tftpd does not support the tsize tftp extension, while dnsmasq does...
> Wonder if the client does depend on that? (but then why does it not have a problem fetching the other boot files?)

Maybe because of different procedures loading those files from tftp server. Never had issues with tftpd, and I'm using it since many years.

But, now we know what the difference is. Maybe a look into the firmware sources can help to figure out what's going on -> @pelwell

mkreisl, 2018-07-01:
Hmmm, seems that tftp is not a good choice:

> Tftpd is not suitable for use with the PXE bootloader; for that, use atftpd or tftpd-hpa

Will try to use `atftpd` tomorrow

mkreisl, 2018-07-01:
Bingo, `atftpd` works, so I can resume:

1.  `tftpd` does not work properly
2.  there is a annoying size limitation for `config.txt`

KasperSkytte, 2018-10-29:
This worked for me: https://stackoverflow.com/questions/49452245/kworker-has-high-cpu-usage-on-idle-system

JamesH65, 2019-01-09:
Not sure what to do with this issue. Can it be closed? We probably need a new issue for the config.txt size issue (if it hasn't already been fixed elsewhere)

emsi, 2020-12-15:
Am having similar issue. At least some config.txt options are not seem to be honored (`dtparam=spi=on` and HDMI configuration).

The boot process looks like this:
```
Dec 15 17:23:19 ubuntu dhcpd[2436]: DHCPREQUEST for 192.168.8.35 (192.168.8.1) from dc:a6:32:7c:c3:d5 via enp0s8
Dec 15 17:23:19 ubuntu dhcpd[2436]: DHCPACK on 192.168.8.35 to dc:a6:32:7c:c3:d5 via enp0s8
Dec 15 17:23:19 ubuntu tftpd[132099]: tftpd: trying to get file: start4.elf
Dec 15 17:23:19 ubuntu tftpd[132099]: tftpd: serving file from /tftpboot
Dec 15 17:23:19 ubuntu tftpd[132101]: tftpd: trying to get file: config.txt
Dec 15 17:23:19 ubuntu tftpd[132101]: tftpd: serving file from /tftpboot
Dec 15 17:23:19 ubuntu tftpd[132103]: tftpd: trying to get file: recover4.elf
Dec 15 17:23:19 ubuntu tftpd[132103]: tftpd: serving file from /tftpboot
Dec 15 17:23:19 ubuntu tftpd[132105]: tftpd: trying to get file: fixup4rc.dat
Dec 15 17:23:19 ubuntu tftpd[132105]: tftpd: serving file from /tftpboot
Dec 15 17:23:20 ubuntu tftpd[132107]: tftpd: trying to get file: recovery.elf
Dec 15 17:23:20 ubuntu tftpd[132107]: tftpd: serving file from /tftpboot
Dec 15 17:23:20 ubuntu tftpd[132109]: tftpd: trying to get file: fixup_rc.dat
Dec 15 17:23:20 ubuntu tftpd[132109]: tftpd: serving file from /tftpboot
Dec 15 17:23:20 ubuntu tftpd[132111]: tftpd: trying to get file: start4.elf
Dec 15 17:23:20 ubuntu tftpd[132111]: tftpd: serving file from /tftpboot
Dec 15 17:23:24 ubuntu tftpd[132113]: tftpd: trying to get file: fixup4.dat
Dec 15 17:23:24 ubuntu tftpd[132113]: tftpd: serving file from /tftpboot
Dec 15 17:23:24 ubuntu tftpd[132116]: tftpd: trying to get file: recovery.elf
Dec 15 17:23:24 ubuntu tftpd[132116]: tftpd: serving file from /tftpboot
Dec 15 17:23:24 ubuntu tftpd[132118]: tftpd: trying to get file: config.txt
Dec 15 17:23:24 ubuntu tftpd[132118]: tftpd: serving file from /tftpboot
Dec 15 17:23:29 ubuntu tftpd[132120]: tftpd: trying to get file: dt-blob.bin
Dec 15 17:23:29 ubuntu tftpd[132120]: tftpd: serving file from /tftpboot
Dec 15 17:23:34 ubuntu tftpd[132123]: tftpd: trying to get file: recovery.elf
Dec 15 17:23:34 ubuntu tftpd[132123]: tftpd: serving file from /tftpboot
Dec 15 17:23:34 ubuntu tftpd[132125]: tftpd: trying to get file: config.txt
Dec 15 17:23:34 ubuntu tftpd[132125]: tftpd: serving file from /tftpboot
Dec 15 17:23:40 ubuntu tftpd[132127]: tftpd: trying to get file: bootcfg.txt
Dec 15 17:23:40 ubuntu tftpd[132127]: tftpd: serving file from /tftpboot
Dec 15 17:23:45 ubuntu tftpd[132129]: tftpd: trying to get file: recovery.rfs
Dec 15 17:23:45 ubuntu tftpd[132129]: tftpd: serving file from /tftpboot
Dec 15 17:23:50 ubuntu tftpd[132132]: tftpd: trying to get file: bcm2711-rpi-4-b.dtb
Dec 15 17:23:50 ubuntu tftpd[132132]: tftpd: serving file from /tftpboot
Dec 15 17:23:55 ubuntu tftpd[132135]: tftpd: trying to get file: overlays/overlay_map.dtb
Dec 15 17:23:55 ubuntu tftpd[132135]: tftpd: serving file from /tftpboot
Dec 15 17:23:55 ubuntu tftpd[132137]: tftpd: trying to get file: config.txt
Dec 15 17:23:55 ubuntu tftpd[132137]: tftpd: serving file from /tftpboot
Dec 15 17:24:00 ubuntu tftpd[132139]: tftpd: trying to get file: recovery.cmdline
Dec 15 17:24:00 ubuntu tftpd[132139]: tftpd: serving file from /tftpboot
Dec 15 17:24:06 ubuntu tftpd[132142]: tftpd: trying to get file: recovery8.img
Dec 15 17:24:06 ubuntu tftpd[132142]: tftpd: serving file from /tftpboot
Dec 15 17:24:11 ubuntu tftpd[132145]: tftpd: trying to get file: recovery8-32.img
Dec 15 17:24:11 ubuntu tftpd[132145]: tftpd: serving file from /tftpboot
Dec 15 17:24:16 ubuntu tftpd[132148]: tftpd: trying to get file: recovery7l.img
Dec 15 17:24:16 ubuntu tftpd[132148]: tftpd: serving file from /tftpboot
Dec 15 17:24:21 ubuntu tftpd[132151]: tftpd: trying to get file: recovery7.img
Dec 15 17:24:21 ubuntu tftpd[132151]: tftpd: serving file from /tftpboot
Dec 15 17:24:26 ubuntu tftpd[132154]: tftpd: trying to get file: recovery.img
Dec 15 17:24:26 ubuntu tftpd[132154]: tftpd: serving file from /tftpboot
Dec 15 17:24:31 ubuntu tftpd[132157]: tftpd: trying to get file: kernel8.img
Dec 15 17:24:31 ubuntu tftpd[132157]: tftpd: serving file from /tftpboot
Dec 15 17:24:31 ubuntu tftpd[132159]: tftpd: trying to get file: kernel8-32.img
Dec 15 17:24:31 ubuntu tftpd[132159]: tftpd: serving file from /tftpboot
Dec 15 17:24:36 ubuntu tftpd[132162]: tftpd: trying to get file: kernel7l.img
Dec 15 17:24:36 ubuntu tftpd[132162]: tftpd: serving file from /tftpboot
Dec 15 17:24:36 ubuntu tftpd[132164]: tftpd: trying to get file: armstub8-32-gic.bin
Dec 15 17:24:36 ubuntu tftpd[132164]: tftpd: serving file from /tftpboot
Dec 15 17:24:41 ubuntu tftpd[132167]: tftpd: trying to get file: kernel7l.img
Dec 15 17:24:41 ubuntu tftpd[132167]: tftpd: serving file from /tftpboot

```

The config.txt file is:

```
[... 62 lines cut ...]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
initramfs initrd.img
```

I'm working with Pi 4 with 2020-12-11 firmware (the 2020-09-03 had the same issues).

My bootcfx.txt is:

```
[all] 

BOOT_UART=0 
WAKE_ON_GPIO=1 
POWER_OFF_ON_HALT=0 
BOOT_ORDER=0xf412 
ENABLEE_SELF_UPDATE=0 
TFTP_PREFIX=1
```

The boot sequence clearly attempts all the recovery and flashing options checking each file. I had to touch files like `recover4.elf`, `fixup4rc.dat`, `recovery.elf` etc. as otherwise it tried like doesen of times to fetch them which delayed boot tremendously. 
It'd be very convenient to disable all that stuff in config.txt (something like `bootloader_update=0` but saying just boot straight, do not attempt any updates or recovery) as tftp is rather slow (especially with multiple retries).

emsi, 2020-12-20:
Also tried with tiny config.txt. No success, overscan is enabled:

```
bootloader_update=0
disable_overscan=1
framebuffer_width=1920
framebuffer_height=1080

```

ovz93br43v7, 2021-01-08:
Same here PiZero ignores everything in config.txt

pinchartl, 2021-05-07:
> There does seem to be a 4096 byte limit for config.txt, but I haven't worked out where that limit is coming from.

I've spent the evening debugging a RPi CM4 boot issue that was caused by exactly this. The same config.txt works when booting from eMMC, but not when booting from the network. Maybe this additional piece of information could help figuring out where the 4k limitation comes from.

ElvishJerricco, 2021-05-17:
My CM4 ignores config.txt even though it's only 211 bytes long.

HayleyDeckers, 2022-12-05:
Can confirm that the CM4/raspi-4 will ignore the entire contents of config.txt _if_ the TFTP server does not support the blksize extension / send an option-acknowledge.  If the TFTP Server does not acknowledge this option (as is allowed per the spec) the firmware will happily ACK the data-packets, but silently ignore the contents of `config.txt`. Strangely enough, booting a `kernel8.img` will still succeed even if the server does not issue an OptionAck. 

 This explains why different TFTP servers show different behaviour as mentioned [here](https://github.com/raspberrypi/firmware/issues/1012#issuecomment-401626769)

There is _also_ a file size limitation at 4096 bytes. Anything past the first 4096 will be trimmed by the firmware. I tested this with a config.txt file that contained a 4081 `#`, a signal `LF` new line, and `kernel=test.img`. The final `g` of the filename is the 4097'th byte of `config.txt` and as expected I see the firmware asking the TFTP server for the kernel file `test.im`.