raspberrypi/firmware #1013: supplied .pc files don't work properly

ali1234, 2018-06-28:
Ask pkg-config how to build a program for brcmegl:

    PKG_CONFIG_LIBDIR=/opt/vc/lib/pkgconfig pkg-config --libs-only-l brcmegl

Result:

    -lbrcmEGL -lbrcmGLESv2 -lbcm_host -lvchostif -lbcm_host -lvcos -lvchiq_arm

There are two problems with the above output. The first is that brcmEGL and brcmGLESv2 don't need to link bcm_host, vchostif, bcm_host, vcos, or vchiq_arm. The second problem is that if you do link vcoshostif then you must also link pthreads, and this .pc does not. As a result, any build using it will fail:

```
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function dispmanx_notify_func: error: undefined reference to 'sem_wait'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function dispmanx_client_callback: error: undefined reference to 'sem_getvalue'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function dispmanx_client_callback: error: undefined reference to 'sem_post'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function dispmanx_notify_callback: error: undefined reference to 'sem_getvalue'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function dispmanx_notify_callback: error: undefined reference to 'sem_post'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function dispmanx_send_command: error: undefined reference to 'sem_wait'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function dispmanx_get_handle: error: undefined reference to 'sem_wait'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function vc_vchi_dispmanx_init: error: undefined reference to 'sem_init'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function vc_vchi_dispmanx_init: error: undefined reference to 'sem_init'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function vc_vchi_dispmanx_init: error: undefined reference to 'sem_destroy'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function vc_vchi_dispmanx_init: error: undefined reference to 'sem_destroy'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function vc_dispmanx_stop: error: undefined reference to 'sem_getvalue'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function vc_dispmanx_stop: error: undefined reference to 'sem_post'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function vc_dispmanx_stop: error: undefined reference to 'sem_destroy'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function vc_dispmanx_stop: error: undefined reference to 'sem_destroy'
> /home/al/Source/rpi-ramdisk/packages/sysroot/opt/vc/lib/libvchostif.a(vc_vchi_dispmanx.c.o):vc_vchi_dispmanx.c:function vc_dispmanx_query_image_formats: error: undefined reference to 'sem_wait'
> collect2: error: ld returned 1 exit status
> Makefile:67: recipe for target 'egl' failed
> make[2]: *** [egl] Error 1
> make[2]: Leaving directory '/home/al/Source/rpi-ramdisk/packages/qt/qtbase/config.tests/egl'
```

JamesH65, 2019-01-09:
@XECDesign Is this one for you? Seems that the brcmegl.pc file has unnecessary dependencies, but I am no expert.

XECDesign, 2019-01-10:
IIRC, you have to call the init function from bcm_host before you can use them, so why should it be removed? Or is the idea that you should be calling pkg-config separately for bcm_host anyway?

ali1234, 2019-01-10:
Further investigation: the problem seems related to brcmegl.pc both requiring bcm_host and also explicitly linking bcm_host and vcoshostif.

brcmglesv2.pc and brcmvg.pc don't do this. I am not sure why.

ali1234, 2019-01-10:
The problem is specifically caused by `-lvcoshostif` in `brcmegl.pc`. This is the library that needs pthreads. It does still cause Qt build failures when configure tests fail. The different behaviour I saw today happened because Qt refactored some code into a submodule which I didn't test (specifically qtdeclarative).

So I think the following should be done:

1. `brcmegl.pc` shouldn't contain `-lvcoshostif` because it isn't needed.
2. `brcmegl.pc` shouldn't contain `-lbcm_host` because `Requires: bcm_host` makes it redundant, and the gles and vg pcs don't have it.

popcornmix, 2019-01-15:
@ali1234 can you submit a PR?

ali1234, 2019-01-15:
Which repository should it be made on? userland or firmware?

popcornmix, 2019-01-15:
Userland is best. It'll be cherry-picked to firmware tree on next firmware update.

ali1234, 2019-01-16:
I just found out something new about this.

EGL does use vchostif functions, but because it is static, they all get bundled into bcm_host. However, other people build it as a shared lib, and then it needs to be linked separately. That is why patches like this exist: https://patchwork.openembedded.org/patch/140309/ - and see the comments, they are building it as a shared lib and they want to upstream it.

So vchostif can be removed from the pc in firmware as it is now. But if you make it into a shared library like some people want then you'll have to put it back. And for userland, the content of the pc should depend on how you build it.

I still don't understand why bcm_host doesn't bring a dependency on pthreads if it contains all the vchostif functions, or how building vchostif as a shared library will affect configure tests when the pcs are actually used. Nobody but me seems to use them.

ali1234, 2019-01-17:
Still investigating this. I found some new things.

pkg-config says the libs should be:

`-lbrcmEGL -lbrcmGLESv2 -lbcm_host -lvchostif -lbcm_host -lvcos -lvchiq_arm`

Notice that bcm_host appears twice. If you link the test executable with those libs it works. However, something in the Qt build is de-duplicating the list of libs so it tries to link with:

`-lbrcmEGL -lbrcmGLESv2 -lvchostif -lbcm_host -lvcos -lvchiq_arm`

This fails because now vchostif appears before bcm_host.

If you swap those two libs around in the command line, the build succeeds. I assume this happens because bcm_host supplies all the needed libs, so the compiler ends up ignoring vchostif.

I will keep investigating why Qt does that de-duplication. I think it should take the first appearance of a library rather than the last in order to not change the meaning when duplicates are present.

I also tested shared build with pcs that don't link vchostif (ie the above patch) and it still works, so what I said before about having to put it back is not true. I suppose this is because egl.so does not directly need anything from vchostif.so.

So at this point I'm fairly happy that my proposed fix is correct and I will open a pull request.

ali1234, 2019-01-17:
Okay so I spoke too soon. While you dont need to link vchostif for egl, you do need to link it if it is shared and you want to call `vc_dispmanx_display_open` which egl programs generally do. In fact Qt has a separate test for this (egl-brcm) which fails when vchostif is shared and it isn't mentioned in any pc. In this case it should really be in a separate pc.

In fact all these pc files are over-linking things. EGL only needs brcmEGL.so at link time. You don't need to link bcm_host or brcmGLESv2 if they are dynamic.

Seems I need to think about this some more.