raspberrypi/firmware #1014: Pi3B+ throttling back at 60C instead of 70C

jahboater, 2018-07-01:
The previous Raspbian release 2018-04-18 throttled back when the temperature (vcgencmd measure_temp arm) reached 70C.  The speed dropped by 200MHz and Vcore went down.
With the latest release 2018-06-17, throttling now starts at 60C.  This lower temp is easily reached when all four ARM cores are in use - doing day to day stuff.
I appreciate this may be a fix for the lockups, however for the vast majority of Pi3+ users this is a drop in performance.  

Please could an option be provided in /boot/config.txt to restore the old 70C threshold?

MilhouseVH, 2018-07-01:
https://forum.kodi.tv/showthread.php?tid=298461&pid=2739998#pid2739998
```
One other change with new firmware is the temperature threshold when we switch from 1.4GHz to 1.2GHz has been reduced from 70'C to 60'C. This may produce some reduction in performance when chip is hot, but it makes some of the borderline boards reliable. It can be overridden with temp_soft_limit=70 to get back to previous behaviour.
```

jahboater, 2018-07-01:
"temp_soft_limit=70" works perfectly, many thanks!

closed 2018-07-01 by jahboater