raspberrypi/firmware #1015: initramfs loaded at the same place as model-specific dtb

hackerjimbo, 2018-07-02:
I've been working on getting a Pi with an encrypted disk. Using the latest Raspbian it almost works but often gets a kernel panic on boot.

Using the vcdbg command I find that the device-specific dtb (in my case bcm2710-rpi-3-b.dtb) and initramfs are both loaded at the same address. The reason it appears to sometimes work is that I think there is a race condition on decompressing the initramfs so that sometimes it gets done before the corruption occurs.

I don't have the specific Pi on me rigth now but I'll add the output for vcdgb log msg as soon as I can.

In the meantime, the initramfs command in config.txt allows various other addresses rather than "followkernel". A workaround may be to load it elsewhere. Please feel free to comment.

hackerjimbo, 2018-07-02:
It also appears that the screen resolution affects the chances of it working! The high
[vcdbg.log](https://github.com/raspberrypi/firmware/files/2155056/vcdbg.log)

er the resolution the lower the chance it works. However, this isn't the actual issue, merely part of the symptoms.

I've attached the output from vcdbg but the key lines are:

```
018669.753: Loading 'initramfs.gz' to 0x4bb0f8 size 0xc2a0a0
020263.721: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x4bb0f8 size 0x6019
```

With the issue being them loading to the same address.

pelwell, 2018-07-02:
You are only seeing half of the story. The "Loading ... to ..." messages are from the common file reading code. What they don't show is that the contents are subsequently copied elsewhere.

The file I/O abstraction layer doesn't have a method to query the file length before reading it, so the loader loads to the initramfs and dtb to the lowest unused address. Subsequently they are both copied to the top end of RAM (but not higher than 768MB to ensure that kernel virtual addresses can cover them). There initramfs relocation has no associated logging output, but you can infer it from the DTB output:
```
018669.753: Loading 'initramfs.gz' to 0x4bb0f8 size 0xc2a0a0
...
020263.721: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x4bb0f8 size 0x6019
...
022065.246: Device tree loaded to 0x2e3ceb00 (size 0x6500)
```
After relocation the DTB stops just before 0x2e3ceb00 + 0x6500 = 0x2e3d5000. Adding in the size of the initramfs takes us to 0x2e3d5000 + 0xc2a0a0 = 0x2efff0a0, i.e. just under the 768MB limit of 0x30000000. If we were to add some initramfs-specific logging it would say something like:
```
0287000.000: initramfs loaded to 0x2e3d5000 (size 0xc2a0a0)
```
This would make it clearer that there is no overlap.

hackerjimbo, 2018-07-02:
OK, that's interesting.

Do you have any idea what could be causing the kernel panics later on? The frequency of occurrence seems to be higher with slower machines and higher-resolution monitors. I've looked on the serial port to try and get some logging output (on a Pi 3) but haven't been able to capture anything as nothing seems to be output.

I'm more than happy to assist with this in any way I can. It appears that others in the past have had similar problems with initramfs but I haven't seen a specific answer.

Bearing in mind that the initramfs is compressed could it have to do with where the decompressed output ends up?

pelwell, 2018-07-02:
I don't know. My first thought with timing-related symptoms like this would be cacheing, but the code is (by my reading) careful to bypass the cache for both the loads and the copies (at the cost of some performance), the version seen by the ARMs should be coherent with the VPU's (uncached) view.

According to [Documentation/arm/Booting](https://github.com/raspberrypi/linux/blob/rpi-4.14.y/Documentation/arm/Booting) the boot sequence is:
```
1. Setup and initialise the RAM.
2. Initialise one serial port.
3. Detect the machine type.
4. Setup the kernel tagged list.
  4a. Setup the kernel tagged list
  4b. Setup the device tree
5. Load initramfs.
6. Call the kernel image.
```
There should be no overlap between the DTB and initramfs processing, so it's hard to see where a race condition could come from.

hackerjimbo, 2018-07-02:
I bumped into Eben at Raspberry Fields and apparently there's concurrency in the VC, but I'm unsure how that affects things.

I've had another thought though. I've got the initramfs command in config.txt with "followkernel" as the address. If the initramfs is decompressed immediately after the end of the kernel then that could collide with the device tree file being loaded there before being relocated.

I've seen an almost identical panic cause by the root file system not being there.

So, my current theory is that the _decompressed_ root image is being corrupted and then the kernel panics. The timing comes from the speed of the SD card or external disk.

The order you give above doesn't seem correct. We see the initramfs being loaded _before_ the device tree for example.

By the way, I don't see any output from the serial not. I suspect this is because the Pi 3 uses the hardware serial port to communicate to the Bluetooth and WiFi.

pelwell, 2018-07-02:
> I bumped into Eben at Raspberry Fields and apparently there's concurrency in the VC, but I'm unsure how that affects things.

It doesn't - all of this image preparation logic is single-threaded, although other things run in parallel.

> The order you give above doesn't seem correct. We see the initramfs being loaded before the device tree for example.

The boot sequence I showed above is _ARM Linux_'s initialisation - the loading and relocation of the DTB and initramfs are performed by the VPU, and the entire procedure is complete before any of the ARMs execute their first instruction.

If you want serial port output on a BT-equipped Pi you will need `enable_uart=1` in config.txt. This is the default for non-BT Pis.

hackerjimbo, 2018-07-02:
Thanks for that, very helpful.

I believe that the issue is in the VideoCore, this is why all the information comes from vcdbg rather than dmesg. I think the corruption is occurring before the kernel even starts which is why I reported this against the firmware rather than the kernel.

Do you know who decompresses the initramfs? From what you say it's loaded up high but who decompresses it, when and where to?

I get the message:

```
[    0.352266] Trying to unpack rootfs image as initramfs...
[    0.547791] Freeing initrd memory: 10788K
```

which implies that it's Linux that does the decompressing. I'm going to assume that Linux will decompress it to a sensible location (because it should know) but I think that the "followkernel" parameter to the initramfs command makes me sceptical. If it actually does that then the compressed data may end up somewhere untoward. I'll have to check on this.

Do you know how the kernel knows where the compressed image is?

maxnet, 2018-07-02:
>Do you have any idea what could be causing the kernel panics later on? The frequency of occurrence seems to be higher with slower machines and higher-resolution monitors.

Sounds familiar.

https://github.com/raspberrypi/linux/issues/2450
https://github.com/raspberrypi/firmware/issues/980

hackerjimbo, 2018-07-02:
Excellent catch!
That looks like an identical issue! maybe it isn't where the initramfs is loaded but some entirely unrelated issue. I'll have a look at the workarounds and see what works.

Many thanks to Phil for walking me through the bootloader process.