raspberrypi/firmware #1016: Power off on rpi3 leads to videocore overtemp warning

satmandu, 2018-07-06:
The Arm Trusted Firmware port to the Raspberry Pi 3 defaults to responding to being asked to shutdown the Raspberry Pi via wfi on one core. (I think this is the default firmware behavior?)

This appears to lead to an overtemp warning in Videocore. @andreiw mentions this in his patch working around this by just resetting when power off is requested: https://github.com/andreiw/raspberry-pi3-atf/commit/b1f5b0343b10562685dbe603585cba565a4bb54a

The upstream Arm Trusted Firmware port (which has RPI3 patches by @AntonioND ) doesn't implement system power off at all, as per https://github.com/ARM-software/arm-trusted-firmware/blob/master/plat/rpi3/rpi3_pm.c

(Reset on both of these ports is implemented similarly using the watchdog timer, so shutdown appears to either slowly overheat, or call the watchdog to reboot the device.)

Is there another/better way to put the Raspberry Pi in a powered down/low power mode after being told to shutdown or is the overtemp warning spurious? Is there a firmware workaround using newer firmware from the Foundation?

This is also mentioned here: https://github.com/andreiw/raspberry-pi3-atf/issues/1

AntonioND, 2018-07-06:
The only real option is to reboot, I guess. Another option is to set the frequency of the cores to the minimum. However, I never got around cleaning my patches to do it because Raspbian doesn't use PSCI.

pelwell, 2018-07-06:
Are interrupts disabled on the core before wfi() is called? If not, it would only take one interrupt to cause it to spin forever.

The usual way to power down the Pi, as demonstrated by the Linux [BCM2835 watchdog driver](https://elixir.bootlin.com/linux/latest/source/drivers/watchdog/bcm2835_wdt.c) is to write 0x555 (plus the usual passwor) to the RSTS register, then cause the watchdog to reset the device. The firmware will then spot the magic value and keep the Pi in a low power state, holding the ARMs in reset. Beyond this you would need to use external hardware to shut off the power.

satmandu, 2018-07-06:
This seems like a relatively minor modification to https://github.com/andreiw/raspberry-pi3-atf/blob/master/plat/rpi3/rpi3_pm.c which could be pushed upstream in atf.

satmandu, 2018-07-06:
@pelwell Am I correct in assuming that the watchdog driver power down method is dependent upon the Pi rebooting and loading bootcode.bin from SD card/USB/NET? 

And it is bootcode.bin loaded at boot off of SD/USB/NET which notices 63 in the RSTS register and tells the Pi to power down to a low power state, flash the led ten times, and then wait for a GPIO wake event?

Or will the firmware see the RSTS field and act upon it to power down even if bootcode.bin is not present?

(Just trying to elucidate from what's in the official boot flow documentation.) https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow.md

pelwell, 2018-07-07:
Yes, that is correct - it is bootcode.bin that interprets the RSTS register and either boots from the indicated partition or halts. The boot ROM always loads `bootcode.bin` from the first available medium.

satmandu, 2018-07-08:
Thanks all.

closed 2018-07-08 by satmandu

antonio-nino-diaz-arm, 2018-07-12:
I've created this PR to add support of PSCI_SYSTEM_OFF to the Trusted Firmware based on the comments in this one: https://github.com/ARM-software/arm-trusted-firmware/pull/1478