raspberrypi/firmware #1018: Add vga666 mode 6

vince87, 2018-07-13:
Hello, can you add the vga666 mode 6 overlay in the future release?
Thanks
Vincenzo

pelwell, 2018-07-16:
Can you explain what you mean by this? There is already a `vga666` overlay (as I think you know), but all it does is set the pin functions for GPIOs 2-21 to enable the DPI interface - the firmware manages everything else.

popcornmix, 2018-07-16:
Possibly the 666 format mentioned here: https://github.com/fenlogic/vga666/blob/master/documents/vga_manual.pdf which uses a different set of gpio lines.

pelwell, 2018-07-16:
That would make sense. Assuming that is the case, try this [updated vga666 overlay](https://drive.google.com/file/d/157IuzZdv8kZB8bYbXI1bsJ2jtOcX_AnH/view?usp=sharing):
1. Download [the file](https://drive.google.com/file/d/157IuzZdv8kZB8bYbXI1bsJ2jtOcX_AnH/view?usp=sharing) and copy it to /boot/overlays/vga666.dtbo
2. Modify your config.txt to include `dtoverlay=vga666,mode6` and reboot.

6by9, 2018-07-16:
Also in our DPI documentation - https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/README.md
mode 6 would imply GPIOs 4-9, 12-17, and 20-25.

vince87, 2018-07-17:
@popcornmix , yes, it's correct, I would like to use mode 6 because it leaves 2 gpio free for audio (13,18).
@pelwell in the next few days, i try your overlay!

JamesH65, 2019-01-09:
@vince87 Did you try this? Did it work? Can this issue be closed?