raspberrypi/firmware #1020: USBFS issues with DIgilent Analog Discovery 2 devices -- FTDI

rprr, 2018-07-16:
I am not sure if this is the right place to post this issue.

I am having issues using the following device -- Digilent Analog Discovery 2. This uses a FTDI chip and connects over USB to the Pi. 
[Analog Discovery 2](https://reference.digilentinc.com/reference/instrumentation/analog-discovery-2/start?redirect=1)
This uses a software called waveforms. 
[Waveforms Software](https://reference.digilentinc.com/reference/software/waveforms/waveforms-3/start)
There is a version for the Raspberry Pi but it does not seem to work. There is a thread on the support forum for waveforms. 
[Digilent Support Forum Thread](https://forum.digilentinc.com/topic/1713-analog-discovery-2-vs-raspberry-pi-3/)
The suggestion by the moderator/developer is that there are some low level USB/FTDI issues which leads to packet loss and cause this device to not work. Other SOC computers appear to work with this device. Also, I have no problems using it on a Mac. 
The main error that shows up in /var/log/syslog is the following.
`Jul 15 22:09:27 rspi3bp kernel: [  285.039475] usb 1-1.2: usbfs: usb_submit_urb returned -121`

JamesH65, 2018-07-16:
@P33M  Any thoughts?

maxnet, 2018-07-18:
Suggest you also stalk FTDI about it.

http://www.ftdicommunity.com/index.php?topic=40.msg134#msg134

`
There are known issues with FIFO in the Linux d2xx driver, unfortunately there are currently no plans to fix these.
`

("driver" = their closed-source userspace library that uses libusb to talk to device)