raspberrypi/firmware #1023: I2C detection issue since # 894 v.4.4.14

lnjuanj, 2018-07-30:
Hi,
I have a PIC device that connects to a raspi through I2C. Since firmware version # 894, v.4.4.14 the raspi doesn't detect the PIC.
I stayed with the 4.4.13 version till now, but I would like to know what changed on v.4.4.14. Other devices keep being detected correctly, as you know.
Details of the problem:
The connected device is a microchip PIC with I2C through the SSP library.
The I2C works correctly with the firmware version #893. However, it will not work with any newer version up to today, since #894 (not all, buy many firmware versions have been tested; recently tested with 4.14.52).
Other I2C devices work, such as the INA219 ( https://www.parallax.com/product/29130 ).
I have looked into the code at:
https://github.com/raspberrypi/firmware ... f9604c2b26
I read the change log and I couldn't see anything in the code about I2C. However, the binary kernel drivers: i2c/i2c-dev.ko and i2c/busses/i2c-bcm2708.ko changed in that version.
I am reviewing the PIC implementation at the same time, but I would also like to know the origin of the I2C lack of detection, since version #894.

popcornmix, 2018-07-30:
If you are saying https://github.com/Hexxeh/rpi-firmware/commit/2e00dcc98c9415fd7a2b0e909ddd8501fc21e8c6 is the first build with the issue and https://github.com/Hexxeh/rpi-firmware/commit/569e6611ac20c735647eb0e550484a73935a672d is okay, then this should can't be a firmware issue (no firmware changed in that update) but a kernel issue and so better reported here:
https://github.com/raspberrypi/linux/issues

Also there were no downstream changes related to i2c, so it's presumably a change in upstream kernel code between 4.4.13 and 4.4.14. On a quick search I don't see any commit between 4.4.13 and 4.4.14 that mentions i2c or affects files in drivers/i2c.

@pelwell is more familiar with i2c than me so may have a better idea when he returns from holiday.

popcornmix, 2018-07-31:
Continued here: https://github.com/raspberrypi/linux/issues/2632

closed 2018-07-31 by popcornmix