raspberrypi/firmware #1024: PoE HAT driver leaves GPIOs 0&1 mapped to SDA0/SCL0, therefore potentially conflicting with the kernel config

6by9, 2018-08-13:
The HAT EEPROM driver is very careful to put the pin muxing back after it has probed and finished.
The PoE HAT driver appears not to be, and leaves GPIOs 0&1 muxed to I2C0.

If you try to use the i2c0-bcm2708 dtoverlay to remux i2c0 elsewhere (eg 28/29, or 44/45) then you end up with no I2C bus due to the double muxing.

I'd noticed odd muxing when trying to talk to the camera from the kernel, and someone else had partly noticed from https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=219935#p1351627 (although that was the DSI display driver, not PoE HAT.

6by9, 2018-08-13:
@XECDesign @pelwell 
Alternate version of release_i2c_semaphore that does restore the pin muxing, or just more careful timing over when we do it during the boot process?

XECDesign, 2018-08-13:
I added that to i2c_release_i2c_semaphore when adding dual camera support. I didn't know that behaviour has since been changed. It looks like the code there is still trying to do the same thing, so if there's an issue, it should probably be fixed there (IMHO).

6by9, 2018-08-13:
Sadly things moved on. Blame Liam
```
commit 1c68a914cf Tue Aug 16 11:19:50 2016 +0100
 I2C:
    Don't reset I2C back to input on close, only do it if necessary
    when obtaining I2C semaphore for different set of pins.
```
IIRC Dave A was bitbashing I2C on GPIOs from the ARM whilst using the camera. There is no arbitration on the function selection registers so they sometimes got a read/modify/write clash between the two CPUs.
obtain_i2c_semaphore now ensures that it is correctly set, but release no longer restores it on the basis that it is likely to be correct for the next access.

HATs are careful as dt-blob.bin has already been parsed so could have set the HAT pins to some other function, or have i2c0 muxed elsewehere. It stores the state, ensures i2c0 is only muxed to the HAT, makes the accesses, and then restores everything back again.

I'm just wondering how the DSI display is avoiding this issue, as that also polls over i2c0. It may be happening before dt-blob.bin configures the GPIOs, but would have to check with the debugger to say for certain.

XECDesign, 2018-08-13:
Should I steal the pin restore code from hat_lib then? Doesn't seem nice to have that much code duplication. hat_lib could be modified to support multiple hats and alternative device addresses, but that seems like something that could/should be done later if there are more HATs on different addresses and the spec has been updated.

6by9, 2018-08-13:
Discuss with @pelwell when he has 5 mins - he's likely to know the full boot sequence off the top of his head.

HAT and PoE init functions are successive operations in arm_loader, so possibly shift to independent GPIO state save/restore calls from there.
Just looking at hat_lib, it also aborts if you have set the HAT pin funcitonality to anything other than input. That behaviour probably wants to be replicated for the PoE fan control too, so pulling the pinmux backup and basic sanity tests out into a common function has appeal.

pelwell, 2018-08-13:
> That behaviour probably wants to be replicated for the PoE fan control too, so pulling the pinmux backup and basic sanity tests out into a common function has appeal.

That would get my vote.

6by9, 2018-08-13:
> That would get my vote.

OK, I'm on it then.

XECDesign, 2018-08-28:
This can probably be closed now?