raspberrypi/firmware #1028: hdmi-rotate=1 cause issue on Xset dpms function

Joern-HAP, 2018-08-26:
Hello,

I have trouble with using xset since my latest update of the PI.
Even after reinstalling image from 2018-06-27-raspbian-stretch + apt-get update/upgrade and rip-update, problem can be replicated.

When display is not rotated (display_rotate=0) in config
`sudo xset dpms 0 0 30`
works fine & display switches off after 30 sec as it should

When I change in /boot/config.txt
`display_rotate=1`
screen only flicker shortly at 30sec, but no change is done as it should be turned off!

Is there a current issue since last updates. I was always running fine before!

Joern-HAP, 2018-08-26:
Hello,

interesting update:

Using hdmi-rotate=2, screen can still not fully turn off, but goes to black screen if I set after 30sec:
Code: Select all

sudo xset display :0 dpms 0 0 30
Very strange behavior !

valenlovii, 2018-08-28:
Got it, thanks.