raspberrypi/firmware #1030: About raspberry pi 3B+ with Ubuntu core

YunchuZhang, 2018-08-29:
Hi there,
I try to use Ubuntu on raspberry Pi 3B+, and my step is like this: from your website 
https://wiki.ubuntu.com/ARM/RaspberryPi#Updating_the_Pi_GPU_firmware_and_bootloader_files
 
Booting the official Pi 2 image on the Pi 3B/3B+
Change the bootloader

The official Ubuntu images use u-boot as the bootloader. The u-boot binary built for the Pi 2 does not work on the Pi 3 and this is why there are separate images for the Pi 2 and Pi 3. However, the Raspberry Pi has its own built in bootloader. This can be used with a few changes to the config.txt file on the system-boot partition.

Write the image to an SD card as normal, but before you insert it into the Pi mount the two partitions so that you can make changes. Open the config.txt on the first partition (labelled system-boot). Change the kernel line, add an initramfs line, and comment out (#) the device_tree_address line as follows:

kernel=vmlinuz
initramfs initrd.img followkernel
#device_tree_address=0x02000000
The second partition contains the Linux root filesystem. Copy the dtb file for your machine (e.g. bcm2710-rpi-3-b.dtb) plus the overlay folder if needed from /lib/firmware/4.X.X-XXXX-raspi2/device-tree to the system-boot partition.

For the Pi 3B+ you will also have to update the Pi bootloader files (bootcode.bin, fixup.dat and start.elf) on the system-boot partition.

Place the SD card in your Pi and turn on!
[image]
It shows like picture said, but does not work.
The wrong information is:
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
How can I deal with it?
Thanks

6by9, 2018-08-30:
Sorry, Raspberry Pi only support Raspbian.
Any issue with Ubuntu needs to be referred back to those who maintain Ubuntu.

JamesH65, 2018-08-31:
This is not "our" website, https://wiki.ubuntu.com/ARM/RaspberryPi#Updating_the_Pi_GPU_firmware_and_bootloader_files

As 6by9 says, you need to talk to the Ubuntu devs,or you could try the Pi forums, there are Ubuntu users on there.

closed 2018-08-31 by JamesH65