raspberrypi/firmware #1032: Booting RPI 2 B from USB fails

Olma2, 2018-09-04:
I used the new bootcode.bin to boot my rpi 2 B version 1.1 from USB. i.e. SD card has a fat partition with bootcode.bin and the USB stick has a fat partition with the rest of the boot files and an ext4 partition with the rootfs. This works but only as long as I don't have any other USB device plugged in. If I plug in an USB ethernet adapter (I also tried a different manufacturer with different chip) it wont boot or even show the rainbow screen. I also put the "timeout" file in the fat partition but this did not help either. I can boot from USB with other USB devices attached when I put the whole fat partition on the sd card. So I guess it won't be something power related.
I also see something going on on the network, which I guess is due to bootcode.bin trying to netboot.
Any ideas how to fix this?

pelwell, 2018-09-04:
Have you tried booting with a non-network USB device such as a mouse or keyboard attached?

Olma2, 2018-09-04:
Yes. With a keyboard or mouse it also boots. So it is somehow related to the USB device being an usb-ethernet adapter.

ghollingworth, 2018-09-04:
Can you enable UART output from bootcode.bin (when it's the only file on an sd card) and log the output?

Olma2, 2018-09-04:
with usb ethernet adapter:

found sd card, config.txt=0, start.elf=0, recovery.elf=0, timeout=0
trying usb
hub device found at addr 4, enumerating hub
initialising hub
found 5 ports, multi_tt = 1
setting interface 0
enabling port power on port 1
enabling port power on port 2
enabling port power on port 3
enabling port power on port 4
enabling port power on port 5
waiting for devices to respond to reset
found device on port 1
found highspeed device
device found: type = ethernet adapter, addr = 5
found device on port 4
found highspeed device
device found: type = ethernet adapter, addr = 6
trying booting from ethernet device addr 5
initialise ethernet with mac aa:bb:cc:dd:ee:ff
wait for link up
failed to initialise link
trying booting from ethernet device addr 5
initialise ethernet with mac aa:bb:cc:dd:ee:ff
wait for link up
failed to initialise link
...
then it is just repeating the last messages

without usb ethernet device

found sd card, config.txt=0, start.elf=0, recovery.elf=0, timeout=0
trying usb
hub device found at addr 4, enumerating hub
initialising hub
found 5 ports, multi_tt = 1
setting interface 0
enabling port power on port 1
enabling port power on port 2
enabling port power on port 3
enabling port power on port 4
enabling port power on port 5
waiting for devices to respond to reset
found device on port 1
found highspeed device
device found: type = ethernet adapter, addr = 5
found device on port 3
found highspeed device
device found: type = mass storage, addr = 6
trying boot from msd device addr 6

One thing which caught my eye was: "waiting for devices to respond to reset"
A general problem of the USB devices I use is that they are rather slow. At the initrd step I sometimes have the problem that the root is not found fast enough. Maybe in case the usb ethernet adapter is attached and the mass storage is not found quick enough it will try to boot via ethernet. Is just a guess though. I'm going to try it with some other USB sticks but in the end I have to stick to the ones I'm using currently.

Olma2, 2018-09-04:
Ok, I tried it with some other USB sticks. One USB stick was USB2 and did not boot, one with USB3 did not boot, another one with USB3 and which is also the fastest one did boot, however. Is there maybe some way to wait longer for all USB devices?

Olma2, 2018-09-04:
Trying program_usb_boot_timeout=1 did not help.

JamesH65, 2019-01-09:
@ghollingworth Are any of the recent bootcode changes likely to have affected this?

ghollingworth, 2019-01-09:
No, as you say it's related to the slowness of the USB enumeration, if it doesn't detect the USB flash drive inside five seconds then there's nothing we can do

closed 2019-01-09 by ghollingworth