raspberrypi/firmware #1033: Unable to Boot RPI 3B from USB

blacktek, 2018-09-04:
Hi,
  I'm trying (with no success) to let my raspberry pi 3 Model B boot from an external usb SSD device (a SanDisk SDSSDA-120G).

Device info:

Raspbian GNU/Linux 8 (jessie)
Linux raspi 4.14.37-v7+ #1111 SMP Thu Apr 26 13:56:59 BST 2018 armv7l GNU/Linu

pi@raspi:~ $ vcgencmd version
Apr 26 2018 23:13:35

I followed many tutorials, forums and videos, but this easy thing seems to not work for me. I've no other USB devices attached to the raspberry (except the SSD)

First thing I did was to flash the SSD disk with a brand new Raspbian Lite. The SSD disk is seen by the raspberry after I attach it:

pi@raspi:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sde           8:64   0 111.8G  0 disk
├─sde2        8:66   0   1.7G  0 part
└─sde1        8:65   0  43.2M  0 part
mmcblk0     179:0    0  14.9G  0 disk
├─mmcblk0p2 179:2    0     1K  0 part
├─mmcblk0p7 179:7    0  13.7G  0 part /
├─mmcblk0p5 179:5    0    32M  0 part
├─mmcblk0p1 179:1    0   1.1G  0 part
└─mmcblk0p6 179:6    0    63M  0 part /boot

1) echo program_usb_boot_mode=1 | sudo tee -a /boot/config.txt

2) before reboot I had:
pi@raspi:~ $ vcgencmd otp_dump | grep 17:
17:1020000a

3) after reboot I had:
pi@raspi:~ $ vcgencmd otp_dump | grep 17:
17:3020000a

After the reboot I removed “program_usb_boot_mode=1“ from /boot/config.txt and than rebooted again, but the raspi didn't reboot from USB device attached

4) I removed the SD card and than I powered up the device, but nothing was happening, just the red led staying red and nothing else.

5) I tried to create an SD card (formatted as FAT32) with just bootcode.img (https://github.com/raspberrypi/firmware/raw/master/boot/bootcode.bin) copied on it, but no success. I tried to both put this file in the root directory and in the /boot/ directory. When booting the raspberry with this sd card inside I was seeing the green led flashing 4 times. And in some cases the rainbow screen (and nothing else)
Anyway most of tutorials reported that SD card was not needed to boot from external device.

6) I tried to use my working SD card and add rootdelay=5 in the boot settings (/boot/cmdline.txt), but no luck here too.

7) I tried to add "program_usb_boot_timeout=1” in /boot/config.txt and then rebooted again the raspi with the sd card inside, but still no luck. After I shutted down the pi, removed the sd card and tried to boot again, but still no luck (just red led on, and green led off)

pi@raspi:~ $ vcgencmd otp_dump | grep 66
66:01009daa

(so bit 24 seems to be set properly)

I've no more guns, can you help?

Thank you

blacktek, 2018-09-04:
forgetting to say that the SanDisk SSD disk is connected to the RPI with a Sabrent (EC-SSHD) SSD/SATA to USB converter

blacktek, 2018-09-04:
If I copy https://github.com/raspberrypi/firmware/raw/next/boot/bootcode.bin in a formatted FAT32 SD card (copied both in / and in /boot/) I see the green led blinking 7 times (should means missing kernel.img)

ghollingworth, 2019-01-09:
It should work fine with just bootcode.bin if not, then it's likely you need the extended timeout (because your SSD takes a long time to initialise).  You should be able to enable UART debug output and get log messages which would tell you what the problem was.

blacktek, 2019-01-09:
I would avoid using the BOOT in the external SD card because I see it like a dirty solution to just go around the problem without really solving it. Moreover this solution is worse to maintain, manage and backup :/

Could you please provide instructions on how to do UART debug? Sometimes my raspi hangs before the rainbow screen and sometims after. It never hangs after a cold boot, it randomly hangs only after reboots.

I've tried with 2 SSDs (SanDisk SDSSDA-120G and now a crucial MX500) and with 3 sata controllers. I've updated/upgraded the system to latest version of raspbian Stretch. I don't use NOOBS (as some people said it could cause problems with boot)

no more guns to shot :(

tnx

pelwell, 2019-01-09:
To enable UART debugging of the boot process, follow this guide: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/

blacktek, 2019-01-09:
or better this https://www.amazon.it/Sumind-Raspberry-Programmazione-Seriale-Supported/dp/B01N4X3BJB/ ?

pelwell, 2019-01-09:
I can't guarantee anything, but the second link even mentions Raspberry Pi so I'm reasonably certain it will work.

blacktek, 2019-01-09:
ok, ordered; I should receive it tomorrow. One question: at every reboot I should restablish the serial connection or I should open it just one time? should I do something on my host computer (windows) after issuing the reboot command on the raspi?

pelwell, 2019-01-09:
You can leave the terminal window open on the host computer - it will survive unplugging from and rebooting of the Pi. My terminal app is often open for weeks at a time, during which time it will be connected to many different Pis.

blacktek, 2019-01-09:
thank you. I'll update you tomorrow :)

blacktek, 2019-01-10:
Hello,
 UART debug started. First boot worked properly and I read:

Raspberry Pi Bootcode

USB MSD boot
Read File: config.txt, 1685 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 1685
Read File: start.elf, 2857060 (bytes)
Read File: fixup.dat, 6666 (bytes)
MESS:00:00:16.838451:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:16.843203:0: brfs: File read: 1685 bytes
MESS:00:00:17.289041:0: HDMI:EDID version 1.3, 1 extensions, screen size 41x26 cm
MESS:00:00:17.294869:0: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
MESS:00:00:17.312683:0: HDMI:EDID found preferred DMT detail timing format: 1440x900p @ 60 Hz (47)
MESS:00:00:17.321322:0: HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
[...]

Then I issued a reboot and things stopped here:

Raspberry Pi Bootcode

USB MSD boot
Read File: config.txt, 1685 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 1685
Read File: start.elf, 2857060 (bytes)

So never did "Read File: fixup.dat, 6666 (bytes)"

What can we say?

blacktek, 2019-01-10:
2-3 other times stopped there. another time stopped here:

MESS:00:00:19.024008:0: *** Restart logging
MESS:00:00:19.026498:0: brfs: File read: 1685 bytes
MESS:00:00:19.886866:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:19.890864:0: Read command line from file 'cmdline.txt':
MESS:00:00:19.896733:0: 'dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=43be32d2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait rootdelay=5'
MESS:00:00:19.912682:0: brfs: File read: 154 bytes
MESS:00:00:53.477020:0: No kernel trailer - assuming DT-capable
MESS:00:00:54.493675:0: Failed to load Device Tree file '?'
MESS:00:00:54.498057:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:59.307475:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:04.116061:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:08.924643:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:13.733240:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:18.541828:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:23.350418:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:28.159004:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:32.967617:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:37.776198:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:42.584790:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:47.393369:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:52.201965:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:01:57.010547:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined

ideas?

blacktek, 2019-01-10:
somebody said to not do "rpi-update" because this way I would install latest beta firmwares (less tested) too. Is it wrong? should I do rpi-update?

pelwell, 2019-01-10:
We certainly wouldn't advise everybody to rpi-update - in general, "sudo apt-get upgrade" or "sudo apt-get distupgrade" is safer - but in this case you have been asked to do so by a Raspberry Pi employee. I'm not aware of any new problems in the current firmware, so I believe it is safe to update.

blacktek, 2019-01-10:
rpi-update done; after apt-get update and apt-get upgrade. Still no luck.

in one of the reboots the raspi hanged before any output to console too.

At next hard reboot I had this:

Raspberry Pi Bootcode

USB MSD boot
Read File: config.txt, 1685 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 1685
Read File: start.elf, 2869092 (bytes)
Read File: fixup.dat, 6695 (bytes)
MESS:00:00:16.850579:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:16.855347:0: brfs: File read: 1685 bytes
MESS:00:00:17.301913:0: HDMI:EDID version 1.3, 1 extensions, screen size 41x26 cm
MESS:00:00:17.307741:0: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
MESS:00:00:17.325553:0: HDMI:EDID found preferred DMT detail timing format: 1440x900p @ 60 Hz (47)
MESS:00:00:17.334194:0: HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
MESS:00:00:17.343304:0: HDMI:EDID found DMT format: code 5, 640x480p @ 72 Hz in established timing I/II
MESS:00:00:17.352419:0: HDMI:EDID found DMT format: code 6, 640x480p @ 75 Hz in established timing I/II
MESS:00:00:17.361533:0: HDMI:EDID found DMT format: code 8, 800x600p @ 56 Hz in established timing I/II
MESS:00:00:17.370648:0: HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
MESS:00:00:17.379763:0: HDMI:EDID found DMT format: code 10, 800x600p @ 72 Hz in established timing I/II
MESS:00:00:17.388964:0: HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
MESS:00:00:17.398167:0: HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
MESS:00:00:17.407454:0: HDMI:EDID found DMT format: code 17, 1024x768p @ 70 Hz in established timing I/II
MESS:00:00:17.416742:0: HDMI:EDID found DMT format: code 18, 1024x768p @ 75 Hz in established timing I/II
MESS:00:00:17.426033:0: HDMI:EDID found DMT format: code 36, 1280x1024p @ 75 Hz in established timing I/II
MESS:00:00:17.435453:0: HDMI:EDID standard timings block x 8: 0x950F 8180 8140 714F 8100 0101 0101 0101
MESS:00:00:17.444621:0: HDMI:EDID found DMT format: code 48, 1440x900p @ 75 Hz (16:10) in standard timing 0
MESS:00:00:17.454081:0: HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 1
MESS:00:00:17.463455:0: HDMI:EDID found DMT format: code 32, 1280x960p @ 60 Hz (4:3) in standard timing 2
MESS:00:00:17.472743:0: HDMI:EDID found DMT format: code 21, 1152x864p @ 75 Hz (4:3) in standard timing 3
MESS:00:00:17.482032:0: HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 4
MESS:00:00:17.504453:0: HDMI:EDID parsing v1 CEA extension
MESS:00:00:17.508257:0: HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
MESS:00:00:17.515986:0: HDMI:EDID found CEA detail timing format: 1280x720p @ 50 Hz (19)
MESS:00:00:17.523798:0: HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
MESS:00:00:17.531521:0: HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
MESS:00:00:17.539153:0: HDMI:EDID moving support for CEA mode 4 to DMT mode 85 because sink does not support HDMI
MESS:00:00:17.549137:0: HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
MESS:00:00:17.558141:0: HDMI:EDID preferred mode remained as DMT (47) 1440x900p @ 60 Hz with pixel clock 106 MHz
MESS:00:00:18.448794:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:18.453424:0: HDMI:Setting property pixel encoding to Default
MESS:00:00:18.459008:0: HDMI:Setting property pixel clock type to PAL
MESS:00:00:18.465171:0: HDMI:Setting property content type flag to No data
MESS:00:00:18.471769:0: HDMI:Setting property fuzzy format match to enabled
MESS:00:00:18.478548:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:18.680878:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:18.688319:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:18.694324:0: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
MESS:00:00:18.713806:0: hdmi: HDMI:EDID version 1.3, 1 extensions, screen size 41x26 cm
MESS:00:00:18.720163:0: hdmi: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
MESS:00:00:18.738500:0: hdmi: HDMI:EDID found preferred DMT detail timing format: 1440x900p @ 60 Hz (47)
MESS:00:00:18.747656:0: hdmi: HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
MESS:00:00:18.757289:0: hdmi: HDMI:EDID found DMT format: code 5, 640x480p @ 72 Hz in established timing I/II
MESS:00:00:18.766924:0: hdmi: HDMI:EDID found DMT format: code 6, 640x480p @ 75 Hz in established timing I/II
MESS:00:00:18.776560:0: hdmi: HDMI:EDID found DMT format: code 8, 800x600p @ 56 Hz in established timing I/II
MESS:00:00:18.786195:0: hdmi: HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
MESS:00:00:18.795831:0: hdmi: HDMI:EDID found DMT format: code 10, 800x600p @ 72 Hz in established timing I/II
MESS:00:00:18.805553:0: hdmi: HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
MESS:00:00:18.815277:0: hdmi: HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
MESS:00:00:18.825084:0: hdmi: HDMI:EDID found DMT format: code 17, 1024x768p @ 70 Hz in established timing I/II
[... cut ...]
MESS:00:00:19.924943:0: brfs: File read: 154 bytes
MESS:00:00:41.138457:0: No kernel trailer - assuming DT-capable
MESS:00:00:42.155123:0: Failed to load Device Tree file '?'
MESS:00:00:42.159505:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:46.968900:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:51.777378:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:56.585863:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
[... repeated many times...]

any idea?

pelwell, 2019-01-11:
The 21 second gap in the log, and the illegal Device Tree name, are baffling:
```
MESS:00:00:19.924943:0: brfs: File read: 154 bytes
MESS:00:00:41.138457:0: No kernel trailer - assuming DT-capable
MESS:00:00:42.155123:0: Failed to load Device Tree file '?'
```
So, to be clear, this only happens on a reboot? How are you triggering the reboot?

blacktek, 2019-01-11:
Yes, 99% of times happens with reboots or if i quickly remove and reinsert power cord. I issue the reboot with "sudo reboot" from command line.

ghollingworth, 2019-01-15:
Quickly remove / replug seems to suggest the problem is the same as the one I've just fixed, which also effected the crash response (you can check this by hitting Ctrl-Alt-B to crash and reboot, the old bootcode.bin wouldn't reboot)

blacktek, 2019-01-15:
Gordon , how did you fix it? do you have a link to a post where I can see your solution? I would avoid putting the boot code on an external SD card. I wouldn't give this win to raspi :D