raspberrypi/firmware #1036: Latest Kernel - 4.14.69 panics on Pi Zero W/Pi 3B Plus

sfx2000, 2018-09-12:
4.14.62 was fine...

A couple of issues where opened up where rpi-update was crashing devices...

Now with the push to production - many devices may be impacted...

_SW Rev => 4.14.69-v7+ #1141 SMP Mon Sep 10 15:26:29 BST 2018
FW Rev => Sep 10 2018 17:26:38 
Copyright (c) 2012 Broadcom
version 46671264d2e10a76c8071b7b084a5409b1afc2cc (clean) (release)_

@popcornmix - you seem to be the guy to reach out to - @pelwell - I know you've been looking into this.

pelwell, 2018-09-12:
This report is lacking information.

1. What is the Pi doing when the crash happens?

2. Do you have any evidence of the panic - a log file or photo of the display?

3. How is the Pi powered? What does `vcgencmd get_throttled` report?

4. Are there any other devices attached?

sfx2000, 2018-09-12:
1) Booting

2) No, nothing is in the kernel log - might see something on console, have a serial cable on order, since I sent my out to another site.

3) Stock Power Supply - the 2.5 amp official one
3a) vcgencmd get_throttled -- since no console cable, throttled=0x0 post boot...

4) Both are Headless, connected via ethernet on the Pi3 B+, WiFi on the Pi Zero W

Wish I could be more helpful, console cable would definitely be a plug in getting the dump...

As for the confit - get_config returns a fairly stock setup

vcgencmd get_config int
aphy_params_current=819
arm_freq=1400
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=400
desired_osc_freq=0x331df0
desired_osc_freq_boost=0x3c45b0
disable_commandline_tags=2
disable_l2cache=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
dphy_params_current=547
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=300
hdmi_force_cec_address=65535
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=85
init_uart_clock=0x2dc6c00
lcd_framerate=60
over_voltage_avs=0x124f8
over_voltage_avs_boost=0x2ab98
pause_burst_frames=1
program_serial_random=1
sdram_freq=450

GoHod, 2018-09-14:
Could this be related to having Bluetooth switched off?

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=222475

sfx2000, 2018-09-14:
I think this is related to the CMA issue...

https://github.com/raspberrypi/linux/issues/2680

mcollins1290, 2018-09-15:
I don't know if it helps but I noticed that 'Peter Ryan' on the RPi Forum (https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=222475) managed to grab a screen grab of the kernel panic. See below:
[image]