raspberrypi/firmware #1037: what is the mailbox property tag for throttling?

tedder, 2018-09-13:
I expected that I'd find a tag listed in [the wiki](https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface) that matched to `vcgencmd get_throttled`. I don't see it. Can you give me the tag or point me to where it is defined?

popcornmix, 2018-09-13:
It is described here: https://github.com/raspberrypi/linux/issues/2367
(but ideally should be added to wiki)

tedder, 2018-09-13:
ooo, [this is exactly what I needed](https://github.com/raspberrypi/linux/issues/2367#issuecomment-364178453), thanks. Now that I know that I found [raspberrypi-firmware.h](https://github.com/raspberrypi/linux/blob/rpi-4.14.y/include/soc/bcm2835/raspberrypi-firmware.h#L80), which lists the magic numbers, which might be useful to others.

closed 2018-09-13 by tedder