raspberrypi/firmware #1038: rpi 3b+ no wlan device found after rpi-upgrade to   4.14.69-v7+

evangelyul, 2018-09-16:
hi new here

after rpi-upgrade no wlan device was found and no network.
eth0 appear in 'ifconfig' however unusable (burned ?) after a recent storm.

config as follows
boot from micro SD then onto a USB stick (root=/dev/sda2)

how can I fix it ?  downgade firmware ? if I try to start  RPI without the SD it hungs.
can I start rpi with SD jessie/firmware A and proceed to stretch/firmwareB  ?
(each device was updated/upgraded without the presence of the other).

popcornmix, 2018-09-17:
This is probably best asked in the [forum](https://www.raspberrypi.org/forums). This repo is for confirmed gpu firmware bugs and this doesn't sound like that.

JamesH65, 2019-01-09:
Closing, this is a forum question.

closed 2019-01-09 by JamesH65