raspberrypi/firmware #1039: Assertion failure after vc_dispmanx_resource_read_data(...) in firmware after early July releases

tvjon, 2018-09-17:
As an example,

https://github.com/mpromonet/v4l2tools/blob/master/src/v4l2grab_h264.cpp

Lines 48, 49.

ret = vc_dispmanx_resource_read_data(resource, rect, buffer, info.width * 3);
	assert(ret == 0);

take_snapshot: Assertion `ret == 0' failed

July 3 & earlier firmware work as expected. 

The snapshot function is called from

encode_omx.cpp(...)

6by9, 2018-09-17:
Do we need any options to run that test app, or do the defaults all magically work?

6by9, 2018-09-17:
The firmware release on July 9th included an update to video_encode that uses the ISP to do the initial image conversion instead of the VPU. 
The only downside of that was that it required the stride to be a multiple of 32 bytes instead of 16 bytes, and it doesn't fall nicely for 24bpp formats.

16th Aug firmware includes a follow up to drop back to VPU conversion if that can cope with the stride requirements when the ISP can't. Have you tested that firmware, as it should be behaving identically to pre 3rd July.

I suspect that the vc_dispmanx_resource_read_data call is failing due to a buffer size being incorrect, so sanity checks are throwing out the request instead of trampling memory.

Looking at https://github.com/mpromonet/v4l2tools/blob/master/src/encode_omx.cpp#L97, that config for nStride is totally bogus (and this isn't the first time I've seen it incorrectly set like this).
a) nStride is in bytes, nFrameWidth is in pixels. OMX_COLOR_Format24bitBGR888 is 24bits/pixel, therefore there needs to be at least a multiply by 3 there.
b) nBufferAlignment is the alignment on the buffer base address, not per line. From the IL spec section 3.1.2.12.1
> nBufferAlignment is a read-only field that specifies the alignment the port requires for each of its buffers (e.g. a value of 4 denotes that each buffer shall be 4-byte aligned). A value of zero denotes that the port does not have any alignment restrictions

As noted above, the component does require nStride to be a multiple of at least 16bytes, and preferably 32 bytes.
I wouldn't be surprised if ```encode_config_input``` is throwing an error due to the incorrect port format, but the call from ```main``` at https://github.com/mpromonet/v4l2tools/blob/master/src/v4l2grab_h264.cpp#L148 doesn't check the return value and then runs on. Buffers will therefore be allocated with the default settings on video_encode, and therefore the buffers are 160x64 YUV420PackedPlanar.

tvjon, 2018-09-17:
Thank you for that 6by9. Whilst I read through that, here's a C version (I'm much happier in C or asm rather than C++) which I think will highlight the problem for you & verify some of what you posted above. 

Yes, there are several similar examples around the web & all will fail...

[alt-snap-fail.zip](https://github.com/raspberrypi/firmware/files/2388065/alt-snap-fail.zip)

tvjon, 2018-09-17:
" Have you tested that firmware, as it should be behaving identically to pre 3rd July."

Yes, same result.

6by9, 2018-09-17:
Bug found in the firmware that it is accepting the reduced nStride when it shouldn't, but the buffer size is then incorrect and is throwing asserts in the firmware. Another one to fix.

Configure the port definition as
```
	def.format.video.nFrameWidth = width;
	def.format.video.nFrameHeight = height;
	def.format.video.xFramerate = framerate << 16;
   def.format.video.nSliceHeight = ALIGN_UP(def.format.video.nFrameHeight, 16); 
   	//This is valid for BGR888, but not other formats. It really ought to be a switch on colorFormat.
	def.format.video.nStride = ALIGN_UP(def.format.video.nFrameWidth * 3, 32);
	def.format.video.eColorFormat = colorFormat;
```
and it sets things up correctly.

You've also got the issue that OMX_FillThisBuffer does NOT block, so your loop retrieving output:
- collects a potentially filled buffer via ```ilclient_get_output_buffer```
- submits it straight back to the GPU before looking at it (which will likely reset nFilledLen)
- then looks at it to see whether it has any data in. Generally this fails to find anything.

You either need to do
```
				OMX_BUFFERHEADERTYPE *out = ilclient_get_output_buffer(video_encode, 201, 0);
				if (out != NULL)
				{
					if (out->nFilledLen > 0)
					{
						fprintf(stderr, "Got o/p buffer with %u bytes\n", out->nFilledLen);
//						size_t sz = videoOutput->write((char*)out->pBuffer, out->nFilledLen);
					}
					
					out->nFilledLen = 0;

					OMX_ERRORTYPE r = OMX_FillThisBuffer(ILC_GET_HANDLE(video_encode), out);
					if (r != OMX_ErrorNone)
					{
	//					LOG(WARN) << "Error filling buffer:" << r ; 
					}
				}
```
or handle OMX_FillBufferDone callbacks yourself and wait on those.

Combine those and I get encoded data buffers back, although I haven't sanity checked the contents.

tvjon, 2018-09-17:
"//This is valid for BGR888, but not other formats. It really ought to be a switch on colorFormat."

Agreed, that would be very useful.

"Combine those and I get encoded data buffers back, although I haven't sanity checked the contents."

I've made your suggested changes to the code I posted but I get:

encode to idle...
enabling port buffers for 200...
**enabling port buffers for 200 failed!**
looping for buffers...

6by9, 2018-09-17:
```
pi@raspberrypi:~/tmp/fw1039/alt-snap-fail $ vcgencmd version
Sep 10 2018 17:26:38 
Copyright (c) 2012 Broadcom
version 46671264d2e10a76c8071b7b084a5409b1afc2cc (clean) (release)
```
That's the standard start.elf, not start_x or start_debug. I do have gpu_mem set to 192MB.

```
pi@raspberrypi:~/tmp/fw1039/alt-snap-fail $ ./snap 
Open display[0]...
Display is 1920 x 1080
Port 200: in 1/1 15360 16 disabled,not pop.,not cont. 160x64 160x64 @1966080 20
Port 200: in 1/1 15360 16 disabled,not pop.,not cont. 1920x1080 5760x1088 @1966080 12
OMX_SetParameter(OMX_IndexParamVideoPortFormat) for video_encode:201...
OMX_SetParameter(OMX_IndexParamBrcmVideoAVCInlineHeaderEnable) for video_encode:201...
Current Bitrate=1000000
encode to idle...
enabling port buffers for 200...
enabling port buffers for 201...
encode to executing...
looping for buffers...
Got o/p buffer with 29 bytes
Got o/p buffer with 26840 bytes
Got o/p buffer with 432 bytes
Got o/p buffer with 85 bytes
Got o/p buffer with 2488 bytes
Got o/p buffer with 535 bytes
Got o/p buffer with 148 bytes
Got o/p buffer with 78 bytes
Got o/p buffer with 67 bytes
Got o/p buffer with 69 bytes
Got o/p buffer with 4067 bytes
Got o/p buffer with 637 bytes
Got o/p buffer with 243 bytes
...
```
Diffs
```
diff --git a/Makefile b/Makefile
index 6dd3742..e478cdf 100644
--- a/Makefile
+++ b/Makefile
@@ -1,2 +1,2 @@
 all:
-       gcc v4l2grab_h264.c encode_omx.c /opt/vc/src/hello_pi/il/ilclient.c -DOMX_SKIP64BIT -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I /opt/vc/include/ -I /opt/vc/include/interface/vcos/pthreads/ -I /opt/vc/include/interface/vmcs_host/linux/ -I /opt/vc/src/hello_pi/libs/ilclient/ -L /opt/vc/lib -L /opt/vc/src/hello_pi/libs/ilclient -lpthread -lopenmaxil -lbcm_host -lvcos -lvchiq_arm  -o snap
+       gcc v4l2grab_h264.c encode_omx.c /opt/vc/src/hello_pi/libs/ilclient/ilclient.c -DOMX_SKIP64BIT -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I /opt/vc/include/ -I /opt/vc/include/interface/vcos/pthreads/ -I /opt/vc/include/interface/vmcs_host/linux/ -I /opt/vc/src/hello_pi/libs/ilclient/ -L /opt/vc/lib -L /opt/vc/src/hello_pi/libs/ilclient -lpthread -lopenmaxil -lbcm_host -lvcos -lvchiq_arm  -o snap
diff --git a/encode_omx.c b/encode_omx.c
index 1a739b9..862e2b8 100644
--- a/encode_omx.c
+++ b/encode_omx.c
@@ -101,7 +101,8 @@ bool encode_config_input(COMPONENT_T* handle, int32_t width, int32_t height, int
 //     def.format.video.nSliceHeight = def.format.video.nFrameHeight;  //6by9 says wrong
    def.format.video.nSliceHeight = ALIGN_UP(def.format.video.nFrameHeight, 16); //def.format.video.nFrameHeight;
 //     def.format.video.nSliceHeight = height;
-       def.format.video.nStride = (def.format.video.nFrameWidth + def.nBufferAlignment - 1) & (~(def.nBufferAlignment - 1));
+       //This is valid for BGR888, but not other formats. It really ought to be a switch on colorFormat.
+       def.format.video.nStride = ALIGN_UP(def.format.video.nFrameWidth * 3, 32);
        def.format.video.eColorFormat = colorFormat;
 
        print_def(def);
diff --git a/v4l2grab_h264.c b/v4l2grab_h264.c
index c15e298..15220e1 100644
--- a/v4l2grab_h264.c
+++ b/v4l2grab_h264.c
@@ -157,17 +157,19 @@ int main(int argc, char **argv)
                                OMX_BUFFERHEADERTYPE *out = ilclient_get_output_buffer(video_encode, 201, 0);
                                if (out != NULL)
                                {
-                                       OMX_ERRORTYPE r = OMX_FillThisBuffer(ILC_GET_HANDLE(video_encode), out);
-                                       if (r != OMX_ErrorNone)
-                                       {
-       //                                      LOG(WARN) << "Error filling buffer:" << r ; 
-                                       }
                                        if (out->nFilledLen > 0)
                                        {
+                                               fprintf(stderr, "Got o/p buffer with %u bytes\n", out->nFilledLen);
 //                                             size_t sz = videoOutput->write((char*)out->pBuffer, out->nFilledLen);
                                        }
                                        
                                        out->nFilledLen = 0;
+
+                                       OMX_ERRORTYPE r = OMX_FillThisBuffer(ILC_GET_HANDLE(video_encode), out);
+                                       if (r != OMX_ErrorNone)
+                                       {
+       //                                      LOG(WARN) << "Error filling buffer:" << r ;
+                                       }
                                }
 
                        } 
```
Sorry, not much else I can do beyond that.

tvjon, 2018-09-17:
Thank you, that is exactly as I have here, so useful as (my) sanity check :).

Having done several reboots yesterday whilst trying things out, inexplicably today, I managed not to do such, & you've guessed it, that  means it now works as your example.

I must get more sleep...

JamesH65, 2019-01-09:
@tvjon It appears that this be closed?

tvjon, 2019-01-09:
Yes, that's fine.

closed 2019-01-09 by tvjon