raspberrypi/firmware #1045: ACT light not blinking after update

prashantsharan2111, 2018-09-27:
Hi All,

I had updated my Raspberry pi 3b+.

Thereafter, I was toiling with the LED blink example. However,  the ACT light blinks during the bootup process but not after that.

Also, the LED isn't lighting up even though the circuit is complete and correct.

However, the LED was working fine before the update.

Kindly help me out ASAP.This is my humble request.

popcornmix, 2018-09-28:
> I had updated my Raspberry pi 3b+.

Do you mean you have a Pi3b+ that was working and you have updated (apt-get upgrade?) and it now no longer works as before?

Or do you mean you have updated from an older Pi (Pi0/Pi1/Pi2/Pi3?) and the same code doesn't work on a Pi3b+?

Also please link to the "LED blink" example you are referring to.

prashantsharan2111, 2018-10-01:
Hi popcornmix,

Thanks for the prompt reply.and very sorry i couldn't respond promptly.

i have a Pi3b+ . After installing VNC , I had rebooted the Pi and after starting up gain, i got a pop telling me that the PI is looking for updates and after a while it started installing them.Since then, the LED is not working at all even though the circuit is complete.

I have attached the image.

[image]

[image]

pelwell, 2018-10-01:
Are you expecting the external LED (which appears to be attached to GPIO18) to flash rather than the onboard ACT LED? If so, what configuration changes have you made to make that happen?

And, as @popcornmix requested, please link to the LED blink example.

prashantsharan2111, 2018-10-04:
@popcornmix , @pelwell 
Thanks for your reply. I am encountering another problem. The GPIO pins like  GPIO 18,23,24,20,21, etc that were previously working on raspberry pi 3 b+ board have stooped working now.

Kindly explain to me the problem and the solution.

pelwell, 2018-10-04:
We need more information in order to begin to guess what might have gone wrong:

1. How are you trying to use the GPIOs?
2. What did you used to observe happening to the GPIOs?
3. What do you observe now?
4. What did you do between when you saw it working and when it stopped working?

prashantsharan2111, 2018-10-04:
@pelwell 
Hi Phil,

I will answer in order of your questions:

Questions 1,2,3 - I was using to light up my LED. The LED lights up when I connect it to the 3.3V pin and the ground pin. However, when I use the GPIO 18 ,GPIO 20,GPIO 21,etc, the it doesn't light up.

QUestion 4 - It was working fine before I updated the software of Raspberry Pi. After that, GPIO pins stopped working. Then I deleted the NOOBS software from the SD card, reinstalled it on the SD card and then booted the Raspberry pi again but to no avail.

pelwell, 2018-10-04:
But how are you trying to control the GPIOs? You don't say.

Use the `raspi-gpio` command to verify that the GPIO settings are correct, e.g.:
```bash
$ raspi-gpio get 18
```
This should tell you whether the pin is configured as an output, and whether it is driving high or low.

prashantsharan2111, 2018-10-04:
@pelwell 
This is the response I got :
raspi-gpio get 18

pelwell, 2018-10-04:
That's not the output - that's the command you typed.

You've not answered several of my questions. I won't be making any more suggestions until you do.

prashantsharan2111, 2018-10-05:
@pelwell 

Apologies.
 
I typed the response incorrectly. The correct response is the following:
**GPIO 18: level=0 fsel=0 func=INPUT**

Kindly tell me which of your question have not been answered?

pelwell, 2018-10-05:
These questions:

"Are you expecting the external LED (which appears to be attached to GPIO18) to flash rather than the onboard ACT LED? If so, what configuration changes have you made to make that happen?"

"And, as @popcornmix requested, please link to the LED blink example."

"But how are you trying to control the GPIOs?"

prashantsharan2111, 2018-10-07:
@pelwell 

1. "Are you expecting the external LED (which appears to be attached to GPIO18) to flash rather than the onboard ACT LED? If so, what configuration changes have you made to make that happen?"

Answer: I am expecting External LED to flash. I have not made any changes to the configuration of the board. It's   Pi 3 b+.

2. "And, as @popcornmix requested, please link to the LED blink example."

Answer: 
      By the "LED blink example" I mean lighting up a LED fitted in a breadboard and connected to raspberry pi through "jumper wires".
      This is a very basic example,I think, the first one anybody tries with Raspberry Pi. So, the context of the whole situation is that in all the "LED blink example" on the internet, it was told to connect this "external LED" to the Raspberry Pi with "Jumper Wires".One jumper is connected to the GND pin on the GPIO and other to the GPIO 18 pin. Once the 220-ohm resistor is in place and the circuit is complete, the LED will light up.However, that didn't happen.

3. "But how are you trying to control the GPIOs?"

Answer: 
I didn't understand this question.

pelwell, 2018-10-07:
GPIO 18 is not normally used to control an LED, and it is not normally "on" (outputting 3.3V). What have you done that makes you think the LED _should_z be on?

Try this command:
```bash
$ raspi-gpio set 18 op dh
```
It will make GPIO 18 an output, driving high (on).

prashantsharan2111, 2018-10-08:
@pelwell 
Question : What have you done that makes you think the LED should be on?

I just saw the connections in the example and applied that.The LED is on now.

Thanks or your time and help.

closed 2018-10-08 by pelwell