raspberrypi/firmware #1046: What means "score" in edidparser tool?

zervankoru, 2018-09-29:
Hello. I have a problem - if I connect my Raspberry Pi by HDMI-DVI adapter to my full hd monitor, then my Raspberry select 1600x1200 resolution instead of 1920x1080. I made edid.dat file by tvservice -d edid.dat, then opened it by edidparser edid.dat. I can see, that raspberry recognizes full hd resolution, but 1600x1200 have much better score. What is the score in editparser output?

timg236, 2018-10-01:
It's hard to tell without seeing the EDID and the HDMI options in config.txt but the score is increased if

1) The format is in the same group as the preferred format.
2) The format is a native resolution e.g. 1920x1200 might be picked over 1920x1080 on a computer monitor if the native resolution is 1920x1200

popcornmix, 2018-10-01:
The comment from the code reads:
```
   /*
    * Scoring system for choosing a preferred format:
    * 1. Is it listed as one of the detailed formats? If so, use this as a base score.
    *    DT0: 1000000, DT1: 800000, DT2: 600000, DT3: 400000
    *    First detail format is basically the preferred format unless filtered out
    *
    * 2. Is it the same group as the preferred format (if defined)? If so,
    *    multiply the resolution score x 2
    *
    * 3. Is it a native resolution? If so, multiply the resolution score by 1.5
    *
    * 4. The resolution score is horiz active pixels x vert active lines x refresh rate / 1000
    *    If the format is interlaced, divide this by 2.
    *
    * 5. For DMT formats, rank standard timings ahead of established timings
    *
    * 6. Any formats which are filtered out automatically get a score of 0.
    */
```

closed 2018-10-01 by zervankoru