raspberrypi/firmware #1048: Windows 10 IoT leaves GPIO Pin to a different state than Raspbian

alexiordan, 2018-09-29:
Raspberry PI 3 Model B

Given 2 pieces of code that should behave the same, Windows 10 IoT (v. 10.0.17134.1) outcome is different from Raspbian one:

I am not sure which one behaves correclty. Please advise.

Pin under test is GPIO14 that has an internall pull-down resistor.

Code 1 - Python - Raspbian

RESET_WATCHDOG1 =  14
def resetWatchDog():
   GPIO.setup(RESET_WATCHDOG1, GPIO.OUT)
   GPIO.output( RESET_WATCHDOG1, False)
   time.sleep(0.500)
   GPIO.setup(RESET_WATCHDOG1, GPIO.IN)

while True:
    resetWatchDog()
    time.sleep(10)
Read more: http://forum.switchdoc.com/thread/444/watchdog-reboots-system-4-seconds#ixzz5SWcd2Amx

Code 2 - .Net Core - Windows 10 IoT

///called each 10 seconds
private async Task PatTheDog()
{
      m_WatchdogPetPin.SetDriveMode(GpioPinDriveMode.Output);
      m_WatchdogPetPin.Write(GpioPinValue.Low);
      await Task.Delay(500);
      m_WatchdogPetPin.SetDriveMode(GpioPinDriveMode.Input);
}
Folowing happens:
Raspbian

When pin is set LOW Voltage goes to 0
When pin is set as INPUT, Voltage increases slowly up to ~0.7 (then another cycle comes and is set to 0)
When the loop ends Voltage grows slowly up to 3.3V
Windows 10 IoT

When pin is set LOW Voltage goes to 0
When pin is set as INPUT, Voltage increases slowly up to ~0.7 (then another cycle comes and is set to 0)
When the loop ends Voltage grows up to ~0.7 and stays there. It is not growing up to 3.3V...
I tried also setting:
m_WatchdogPetPin.SetDriveMode(GpioPinDriveMode.InputPullUp);
but then

When pin is set as INPUT, Voltage increases slowly up to 3.3V

I have also posted here: https://social.msdn.microsoft.com/Forums/en-US/1e51a5ec-96a1-4540-8efe-53960e471015/raspberry-pi-3-model-b-windows-10-iot-leaves-gpio-pin-to-a-different-state-than-raspbian?forum=WindowsIoT
and here: https://github.com/ms-iot/bsp/issues/44
 looking for the people who better know what is under the hood.

pelwell, 2018-09-30:
Your Raspbian loop is infinite, so how are you causing it to exit? Do you get the same result with a finite loop? Do you get the same result without the loop altogether?

alexiordan, 2018-09-30:
I have pdate initial post with the actual behavior when voltage grows. I added this part where appropriate:  _(actually it only reaches ~2.2 V because it is pulled down to ~1.1 from where it start growing again...)_

Yes it is infinite. I just Ctrl+C ed the app. I tried with 2 code versions with same result:
- Having an exception handling with `GPIO.cleanup()`
```
except KeyboardInterrupt:
        GPIO.cleanup() # clean up GPIO on CTRL+C exit
GPIO.cleanup() # clean up GPIO on normal exit
```
- Without such an exception handling, but closing the app during the period where pin was set as input.

From python docs here: https://sourceforge.net/p/raspberry-gpio-python/wiki/BasicUsage/
_By returning all channels you have used back to inputs with no pull up/down, you can avoid accidental damage to your RPi by shorting out the pins._ 
doesn't seem to be too much of a difference between the 2 approaches.

@pelwell , Hopefully I have answer your question.

pelwell, 2018-09-30:
What is the shortest sequence that shows the difference between the two platforms?

alexiordan, 2018-09-30:
After the first cycle.
I can record a movie if you think is useful.

pelwell, 2018-09-30:
No, that's not necessary. Instead, write the shortest two pieces of straight line code - one Raspbian, one Windows IoT - that appear to do the same but produce different results.

alexiordan, 2018-09-30:
I have placed the two pieces of code here: https://github.com/alexiordan/WindowsIoTGPIO
- I measure voltage between GPIO18 and GND while test is running.

However I have seen that Windows on fresh boot, before I run my .NetCore test has the same behavior as Raspbian.
After running the application the pin remains in that "unusual" state.

pelwell, 2018-09-30:
Is that really the simplest example you can manage? Anything involving classes, delays and exception handling is already too complex. I suspect you can reduce the code down to something like "initialise the pin; do something with the pin; exit".

alexiordan, 2018-09-30:
Sorry for that. I will update to something simplier.

alexiordan, 2018-09-30:
@pelwell , I have updated the .net core example. Is it OK now?

pelwell, 2018-09-30:
Not really. I asked for the minimal, straight-line (i.e. no loops) code that demonstrates the difference - are you telling me that there has to be a while loop in the Python?

1. Replace your Python with this:
```python
import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)

# Patting line
RESET_WATCHDOG1 = 18

GPIO.setup(RESET_WATCHDOG1, GPIO.OUT)
GPIO.output( RESET_WATCHDOG1, False)
time.sleep(0.500)
GPIO.setup(RESET_WATCHDOG1, GPIO.IN)
```
How does it behave?

2. Then add `GPIO.cleanup` on the end - does that make any difference?

3. If you finish the code with:
```python
while True:
        continue
```
and exit with Ctrl-C, what difference does that make?

4. Does adding the exception catching change anything?

alexiordan, 2018-10-01:
>    1. Does adding the exception catching change anything?

Oh, I thought you were talking about the. Net sample.

I'll make the test as you proposed and come back with results, but it will
take some time.

JamesH65, 2019-01-09:
@alexiordan Anything to report on this?

alexiordan, 2019-01-09:
No @JamesH65 , unfortunately I did not make any further progress on this. 
I would have expected that someone could have tried my samples and see the results.
Since this was not the case, we can close this because I'm not sure when I'll be able to post something new. We can reopen it later if that will be the case.

JamesH65, 2019-01-10:
We simply have nothing to do with WinIoT here, no-one uses it and no-one has any idea how it works. All our dev team are Linux based, so we are not able to help. You'll need to approach MS. Sorry.

closed 2019-01-10 by JamesH65